Hamlib/Hamlib #1134: String data type in XML-RPC incorrectly defined in Hamlib 4.5

AG7GN, 2022-10-30:
Hamlib 4.5 does not recognize string data type returned via XML-RPC. The reason is that this line in `rigs/dummy/flrig.c`:

`if (streq(p, "i4") || streq(p, "double") || streq(p, "int") || streq(p, " string"))`

has an extraneous space in `" string"`. It should be:

`if (streq(p, "i4") || streq(p, "double") || streq(p, "int") || streq(p, "string"))`

Thanks.

Steve AG7GN

N0NB, 2022-10-31:
Thanks, Steve.  Mike, will this require a 4.5.1 bug release soon?

mdblack98, 2022-10-31:
Where is this causing an error?
It does need fixed but FLRig is the only known client for this and I don't know of any errors with that (I use it here)

AG7GN, 2022-10-31:
Thanks for fixing this. I've written my own flrig-like app for certain Kenwood radios using Python's `xmlrpc.server` module. That module does not support the `nil` or `i4` data types, but it does support `int` and `string`, hence my need for `hamlib` XML-RPC to support strings.

Steve AG7GN

closed 2022-11-08 by mdblack98