Hamlib/Hamlib #1140: Timeout errors communicating with Icom IC-746

saharbin, 2022-11-07:
**Attempting command**: rigctl -m 3023 -r COM1 -vvvvv

**Responds**:
rigctl.c(444) Startup: rigctl -m 3023 -r COM1 -vvvvv
rigctl Hamlib 4.6~git Sat Nov 05 17:58:40 2022 +0000 SHA=107441
Report bugs to <hamlib-developer@lists.sourceforge.net>

rig_check_rig_caps: p1=000000006A299CE0, p2=000000006A2A0258, rig_model=000000006A299CE0, macro_name=000000006A2A0258
initrigs4_icom: _init called
rig_init(558): 00000000001A9FEC rs->comm_state==0?=0
rig_init: rig has VFO_A
rig_init: rig has VFO_B
rig_init: rig has VFO_MEM
 1:icom.c(623):icom_init entered
icom_init: done
 1:icom.c(723):icom_init returning(0)
 1:rig.c(807):rig_open entered
rig_settings_get_path: path=.hamlib_settings
rig_settings_load_all: settings_file (.hamlib_settings): No such file or directory
rig_open: cwd=C:\Users\steve
rig_open: C:\Users\steve/hamlib_settings does not exist
rig_open: async_data_enable=0, async_data_supported=0
serial_open: COM1
serial_setup: tcgetattr
serial_setup: cfsetispeed=19200,0x000e
serial_setup: cfsetospeed=19200,0x000e
serial_setup: data_bits=8
serial_setup: parity=0
serial_setup: Handshake=None
serial_setup: tcsetattr TCSANOW
read_string_generic called, rxmax=4095 direct=1, expected_len=1
  2:rig.c(7217):async_data_handler_start entered
async_data_handler_start: async data support disabled since asynd_data_enabled=0
  2:rig.c(7224):async_data_handler_start returning(0)
rig.c(254):add_opened_rig returning2(0)
rig_open: 00000000001A9FEC rs->comm_state==1?=1
  2:icom.c(986):icom_rig_open entered
icom_rig_open: IC-746 v20221104.3
   3:icom.c(776):icom_get_usb_echo_off entered
    4:frame.c(425):icom_transaction entered
icom_transaction: cmd=0x03, subcmd=0xffffffff, payload_len=0
     5:frame.c(138):icom_one_transaction entered
rig_flush: called for serial device
read_string_generic called, rxmax=4095 direct=1, expected_len=1
write_block(): TX 6 bytes, method=2
0000    fe fe 56 e0 03 fd                                   ..V...
read_string_generic called, rxmax=200 direct=1, expected_len=1
read_string_generic(): Timed out 1.671 seconds after 0 chars, direct=1
     5:frame.c(294):icom_one_transaction returning(-5) Communication timed out

icom_transaction: retry=0: async_data_handler_start: async data support disabled since asynd_data_enabled=0
  2:rig.c(7224):async_data_handler_start returning(0)
rig.c(254):add_opened_rig returning2(0)
rig_open: 00000000001A9FEC rs->comm_state==1?=1
  2:icom.c(986):icom_rig_open entered
icom_rig_open: IC-746 v20221104.3
   3:icom.c(776):icom_get_usb_echo_off entered
    4:frame.c(425):icom_transaction entered
icom_transaction: cmd=0x03, subcmd=0xffffffff, payload_len=0
     5:frame.c(138):icom_one_transaction entered
rig_flush: called for serial device
read_string_generic called, rxmax=4095 direct=1, expected_len=1
write_block(): TX 6 bytes, method=2
0000    fe fe 56 e0 03 fd                                   ..V...
read_string_generic called, rxmax=200 direct=1, expected_len=1
read_string_generic(): Timed out 1.671 seconds after 0 chars, direct=1
     5:frame.c(294):icom_one_transaction returning(-5) Communication timed out

Communication timed out
Communication timed out
 
icom_transaction: failed: rig.c(254):add_opened_rig returning2(0)
rig_open: 00000000001A9FEC rs->comm_state==1?=1
  2:icom.c(986):icom_rig_open entered
icom_rig_open: IC-746 v20221104.3
   3:icom.c(776):icom_get_usb_echo_off entered
    4:frame.c(425):icom_transaction entered
icom_transaction: cmd=0x03, subcmd=0xffffffff, payload_len=0
     5:frame.c(138):icom_one_transaction entered
rig_flush: called for serial device
read_string_generic called, rxmax=4095 direct=1, expected_len=1
write_block(): TX 6 bytes, method=2
0000    fe fe 56 e0 03 fd                                   ..V...
read_string_generic called, rxmax=200 direct=1, expected_len=1
read_string_generic(): Timed out 1.671 seconds after 0 chars, direct=1
     5:frame.c(294):icom_one_transaction returning(-5) Communication timed out

Communication timed out

Communication timed out
Communication timed out

.
.
.
Communication timed out

   3:frame.c(138):icom_one_transaction entered
rig_flush: called for serial device
read_string_generic called, rxmax=4095 direct=1, expected_len=1
write_block(): TX 7 bytes, method=2
0000    fe fe 56 e0 07 00 fd                                ..V....
read_string_generic called, rxmax=200 direct=1, expected_len=1
read_string_generic(): Timed out 1.692 seconds after 0 chars, direct=1
   3:frame.c(294):icom_one_transaction returning(-5) Communication timed out

Communication timed out

Communication timed out
Communication timed out

  2:frame.c(455):icom_transaction returning(-5) Communication timed out

icom_set_vfo: line#3046
icom.c(3050):icom_set_vfo returning2(-5) Communication timed out

rig.c(2098):rig_get_freq returning2(-5) Communication timed out

 1:rig.c(1361):rig_open returning(0)
Opened rig model 3023, 'IC-746'
Backend version: 20221104.3, Status: Stable
rigctl_parse: called, interactive=1

Rig command: scanfc: ret=-1
rigctl_parse: nothing to scan#1? retcode=-1, last_cmd=

mdblack98, 2022-11-07:
Something is wrong with your COM port.
#1 Ensure cable is plugged in correctly
#2 Ensure correct COM port is selected
#3 Double check baud rate on rig
#4 Ensure correct baud rate is selected to match the rig with the -s switch -- e.g. rigctl -m 3023 -r COM1 -s 19200

saharbin, 2022-11-08:
Thanks mdblack98.  That info is helpful.

It turns out that I needed to add "-C  rts_state=ON" as the serial interface is powered by the RTS pin.  So far, everything seems to be working.

closed 2022-11-08 by saharbin