Hamlib/Hamlib #1149: Building tests/memsave.c with clang-15 and -Werror=incompatible-function-pointer-types failes

dl1jbe, 2022-11-14:
Coming clang-16 will force the above -Werror check to fail. During tests here at gentoo linux we observed the following (see also https://bugs.gentoo.org/880961)

`
memsave.c:65:53: error: incompatible function pointer types passing 'int (RIG *, vfo_t, channel_t
      **, int, const chan_t *, void *)' (aka 'int (struct s_rig *, unsigned int, struct channel **, int, const struct
      chan_list *, void *)') to parameter of type 'chan_cb_t' (aka 'int (*)(struct s_rig *, struct channel **, int,
      const struct chan_list *, void *)') [-Werror,-Wincompatible-function-pointer-types]
    retval = rig_get_chan_all_cb(rig, RIG_VFO_NONE, dump_xml_chan, root);
                                                    ^~~~~~~~~~~~~
../include/hamlib/rig.h:2953:46: note: passing argument to parameter 'chan_cb' here
                                   chan_cb_t chan_cb,
                                             ^
`

Seems declaration and definition of callback function `static int dump_xml_chan` on lines 32 and 87 should drop  second parameter (`vfo_t vfo`).

mdblack98, 2022-11-14:
Try changing
typedef int (* chan_cb_t)(RIG *, channel_t **, int, const chan_t *, rig_ptr_t);
To
typedef int (* chan_cb_t)(RIG *, vfo_t vfo, channel_t **, int, const chan_t *, rig_ptr_t);

See if that clears up the error.  If it does I'll post a patch.

mdblack98, 2022-11-14:
I see that.  Needs changes in multiple files.
I'm finishing up another patch (hopefully today) and I'll get this fixed next either today or tomorrow.
Mike
 

    On Monday, November 14, 2022 at 12:17:00 PM CST, Thomas Beierlein ***@***.***> wrote:  
 
 

It seems not to be so easy.

Changing it in include/hamlib/rig.h resuts in aor.c no longer compiling.

`
aor.c:1371:59: error: too few arguments to function call, expected 6, have 5
retval = chan_cb(rig, &chan, chan_next, chan_list, arg);
~~~~~~~ ^
'

Same on line 1421. Adding the vfo parameter helps.

But a lot of similar errors turns up in mem.c too. 3 times not enough parameters and 6 times incompatible pointer types.

Maybe there are more problems in further files.

—
Reply to this email directly, view it on GitHub, or unsubscribe.
You are receiving this because you commented.Message ID: ***@***.***>

dl1jbe, 2022-11-17:
Looks good so far. No runtime tests yet, but compiles cleanly (even if backported to Hamlib-4.4).
Thanks.

closed 2022-11-17 by mdblack98