Hamlib/Hamlib #1152: X6100 Rig Split does not work

mdblack98, 2022-11-17:
WSJTx direct to X6100: WSJTx set for Fake It & Data/Pkt. Works and how I use it without FLrig, controlled X6100 and showed correct readouts, except for filters (same as test 1) and VFO A did not reset to rx after tx so fake it offset added each tx, WSJTx Split set for Rig does not work due to flakey comm.

Tyrbiter, 2022-11-17:
Wonder if this still happen with the latest v1.1.6 firmware released about 2nd November? Quite a lot of CAT/CI-V changes and additional commands in it.

mdblack98, 2022-11-17:
Are you running an X6100?

The problem was the old firmware split command wasn't working but they may have fixed that.  I sent a test version to find out.

Mike W9MDB

Tyrbiter, 2022-11-17:
Not got an X6100 yet, but thinking about it.

closed 2022-11-17 by mdblack98