Hamlib/Hamlib #1155: Add AGC levels in NET rig dumpcaps

PianetaRadio, 2022-11-18:
Reflect AGC levels of the rig backend in dumpcaps of NET rigctl https://github.com/Hamlib/Hamlib/blob/master/rigs/dummy/netrigctl.c

Actually NET rigctl dumpcaps prints out all the supported AGC levels, ignoring the real rig backend.

PianetaRadio, 2023-01-05:
Still not working properly, for example with Yaesu FT-991A

```
rigctl -m 1035
Rig command: \dump_caps
Caps dump for model: 1035
Model name:	FT-991
...
AGC levels: 0=OFF 1=FAST 2=MEDIUM 3=SLOW 4=AUTO
...
```

```
rigctl -m 2
Rig command: \dump_caps
Caps dump for model: 2
Model name:	NET rigctl
...
AGC levels: 0=OFF 1=OFF 2=SUPERFAST 3=FAST 4=SLOW 5=USER
...
```

PianetaRadio, 2023-01-14:
Sorry, still not working, now:

```
Rig command: \dump_caps
Caps dump for model: 2
Model name:	NET rigctl
Mfg name:	Hamlib
Backend version:	20230106.0
...
AGC levels: 0=OFF 1=SUPERFAST 2=FAST 3=SLOW 4=USER 0=OFF

```

PianetaRadio, 2023-01-17:
Strange that when I open the communication there are some info referring to the AGC and they are ok:

```
netrigctl_open: rigctld_version=Hamlib 4.6~git Jan 16 15:24:00Z 2023 SHA=cb1c98
read_string_generic called, rxmax=1024 direct=1, expected_len=1
read_string_generic(): RX 47 characters, direct=1
0000    61 67 63 5f 6c 65 76 65 6c 73 3d 30 3d 4f 46 46     agc_levels=0=OFF
0010    20 31 3d 46 41 53 54 20 32 3d 4d 45 44 49 55 4d      1=FAST 2=MEDIUM
0020    20 33 3d 53 4c 4f 57 20 34 3d 41 55 54 4f 0a         3=SLOW 4=AUTO. 
netrigctl_open: rig has agc code=0, level=OFF
0=OFF
netrigctl_open: rig has agc code=1, level=FAST
1=FAST
netrigctl_open: rig has agc code=2, level=MEDIUM
2=MEDIUM
netrigctl_open: rig has agc code=3, level=SLOW
3=SLOW
netrigctl_open: rig has agc code=4, level=AUTO
4=AUTO

```

But later, if I call \dump_caps, it reports a wrong AGC list

mdblack98, 2023-01-18:
Try again....I think it's fixed now.

Mike W9MDB

PianetaRadio, 2023-01-19:
Ok

`AGC levels: 0=OFF 2=FAST 5=MEDIUM 3=SLOW 6=AUTO`

closed 2023-01-19 by PianetaRadio