Hamlib/Hamlib #1157: rig_send_raw not working

PianetaRadio, 2022-11-22:
Code example:
```
RIG *my_rig;
(my_rig open... etc)
unsigned char sendCmd[] = "xyz;";
int sendCmdSize = 4;
unsigned char term[] = ";";
unsigned char rcvdCmd[100];
int rcvdCmdLen = 0;

rig_send_raw(my_rig, sendCmd, sendCmdLen, rcvdCmd, rcvdCmdLen, term);
```

The program crashes after calling rig_send_raw, the last line in debug is "1:rig.c(7401):rig_send_raw entered", nothing else.
Tested with 991A (serial or netrigctl) and dummy.

PianetaRadio, 2022-11-22:
int rig_send_raw(rig, const unsigned char *send, int send_len, unsigned char *reply, int reply_len, unsigned char *term)

Also, looking into the code, int reply_len is not used and unsigned char *reply not returned.

PianetaRadio, 2022-11-24:
I don't know, still not working for me...

mdblack98, 2022-11-24:
Debug logs please?
Mike W9MDB

PianetaRadio, 2022-11-24:
Mike, let me setup a clean code in the next days to test it and I will be back.
Anyway, the error is the same, when the function is called, immediately crashes, in the log I find only `1:rig.c(7401):rig_send_raw entered` and nothing else.

I have a question, what happens if rig_send_raw is called for dummy rig or net rictl rig?

mdblack98, 2022-11-24:
Dummy knows nothing of raw commands.
if rigctld is hooked up to a real radio rigctld should work with the raw command.
Mike

PianetaRadio, 2022-11-25:
Ok, now if I use with dummy, doesn't crash and report the right error message.
Still problem with real rig, maybe some issue with char, unsigned char, pointer, array, conversion, etc. It's a nightmare!

mdblack98, 2022-11-25:
What rig? Sent from my iPhone

PianetaRadio, 2022-11-25:
Yaesu FT-991

PianetaRadio, 2022-11-25:
I suggest to temporary insert some verbose messages for each step, reporting also variables content to better understand the issue.

PianetaRadio, 2022-11-26:
Tested on my Yaesu FT-991 using rigctl

```
Rig command: \send_raw FA;
rigctl_parse: cmd=send_raw handle=0
rigctl_parse: rig_powerstat is not on = 0
  2:rig.c(5856):rig_get_powerstat entered
   3:newcat.c(3470):newcat_get_powerstat entered
newcat.c(7427):newcat_valid_command returning2(1)
    4:newcat.c(10251):newcat_get_cmd entered
    4:newcat.c(10473):newcat_get_cmd returning(0)
   3:newcat.c(3503):newcat_get_powerstat returning(0)
  2:rig.c(5878):rig_get_powerstat returning(0)
  2:rig.c(7415):rig_send_raw entered
rig_send_raw: writing 3 bytes
```

Then it returns to command prompt!
Something is strange, "rig_powerstat is not on = 0", but the radio is on.

mdblack98, 2022-11-26:
Please try the latest

send_raw from rigctl now requires a terminator

e.g.

send_raw ; FA;

       send_raw 'Terminator' 'String'
              Can send ASCII string or 0xnn values -- there can be no spaces in the command string.
              Possible terminator values are CR, LF, ;, ICOM, 0-100 (bytes to read), or -1 meaning unknown (will timeout on read)
              Examples:
                send_raw ; FA;MD;
                send_raw icom 0xFE;0xFE;0x94;0x03;0xFD
                send_raw -1 0xFE;0xFE;0x94;0x03;0xFD
                send_raw 14 0xFE;0xFE;0x94;0x03;0xFD

Mike W9MDB

PianetaRadio, 2022-11-26:
Ok. I will test tomorrow with the last daily snapshot.

Concerning rig_powerstat is not on = 0, is just a misunderstanding, = 0 means false, so "is not on = false" means radio is on.

General question out of theme: how to use the simulators? Can you provide a socat command example to link them to a virtual serial port.

mdblack98, 2022-11-26:
rigctl -m 1035 -r com1 -s 19200 -vvvvv -Z --vfo >log.txt 2>&1
You won't see a prompt.
Then do this
\sendraw ; FA;q
Then send me the log.txt file
Mike

PianetaRadio, 2022-11-27:
Mike,

find attached the log file.

73

Johnny

.\rigctl.exe : 2022-11-27T23:50:52.804436+0100: rigctl.c(445) Startup: 
C:\Users\Johnny\Documents\QT\CatRadio\hamlib\rigctl.exe -m 1035 -r COM5 -s 38400 -vvvvv -Z --vfo
In riga:1 car:1
+ .\rigctl.exe -m 1035 -r COM5 -s 38400 -vvvvv -Z --vfo >log.txt 2>&1
+ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : NotSpecified: (2022-11-27T23:5...-vvvvv -Z --vfo:String) [], RemoteException
    + FullyQualifiedErrorId : NativeCommandError
 
2022-11-27T23:50:52.804436+0100: rigctl Hamlib 4.6~git Sun Nov 27 04:10:19 2022 +0000 SHA=09e73b
2022-11-27T23:50:52.804436+0100: Report bugs to ***@***.***>

2022-11-27T23:50:52.804436+0100: rig_check_rig_caps: p1=000000005FE2ACE0, p2=000000005FE31258, 
rig_model=000000005FE2ACE0, macro_name=000000005FE31258
2022-11-27T23:50:52.804436+0100: yaesu: initrigs4_yaesu called
2022-11-27T23:50:52.804436+0100: rig_init: rig_model=Yaesu FT-991
2022-11-27T23:50:52.804436+0100: rig_init: rig has VFO_A
2022-11-27T23:50:52.804436+0100: rig_init: rig has VFO_B
2022-11-27T23:50:52.804436+0100: rig_init: rig has VFO_MEM
2022-11-27T23:50:52.804436+0100: ft991_init called, version 20221104.14
2022-11-27T23:50:52.804436+0100:  1:newcat.c(468):newcat_init entered
2022-11-27T23:50:52.804436+0100:  1:newcat.c(508):newcat_init returning(0) 
2022-11-27T23:50:52.804436+0100:  1:rig.c(812):rig_open entered
2022-11-27T23:50:52.804436+0100: rig_settings_get_path: path=.hamlib_settings
2022-11-27T23:50:52.804436+0100: rig_settings_load_all: settings_file (.hamlib_settings): No such file or directory
2022-11-27T23:50:52.804436+0100: rig_open: cwd=C:\Users\Johnny\Documents\QT\CatRadio\hamlib
2022-11-27T23:50:52.804436+0100: rig_open: C:\Users\Johnny\Documents\QT\CatRadio\hamlib/hamlib_settings does not exist
2022-11-27T23:50:52.804436+0100: rig_open: async_data_enable=0, async_data_supported=0
2022-11-27T23:50:52.804436+0100: serial_open: COM5
2022-11-27T23:50:52.804436+0100: ser_set_rts: RTS=0
2022-11-27T23:50:52.804436+0100: ser_set_dtr: DTR=0
2022-11-27T23:50:52.804436+0100: serial_setup: tcgetattr
2022-11-27T23:50:52.804436+0100: serial_setup: cfsetispeed=38400,0x000f
2022-11-27T23:50:52.804436+0100: serial_setup: cfsetospeed=38400,0x000f
2022-11-27T23:50:52.804436+0100: serial_setup: data_bits=8
2022-11-27T23:50:52.804436+0100: serial_setup: parity=0
2022-11-27T23:50:52.804436+0100: serial_setup: Handshake=Hardware
2022-11-27T23:50:52.804436+0100: serial_setup: tcsetattr TCSANOW
2022-11-27T23:50:52.804436+0100: read_string_generic called, rxmax=4095 direct=1, expected_len=1
2022-11-27T23:50:52.886488+0100:   2:rig.c(7227):async_data_handler_start entered
2022-11-27T23:50:52.886488+0100: async_data_handler_start: async data support disabled since asynd_data_enabled=0
2022-11-27T23:50:52.886488+0100:   2:rig.c(7234):async_data_handler_start returning(0) 
2022-11-27T23:50:52.886488+0100: rig.c(254):add_opened_rig returning2(0) 
2022-11-27T23:50:52.886488+0100: rig_open: 0000000000D59F4C rs->comm_state==1?=1
2022-11-27T23:50:52.886488+0100:   2:newcat.c(548):newcat_open entered
2022-11-27T23:50:52.886488+0100: newcat_open: Rig=FT-991, version=20221104.14
2022-11-27T23:50:52.886488+0100: newcat_open: write_delay = 0 msec
2022-11-27T23:50:52.886488+0100: newcat_open: post_write_delay = 25 msec
2022-11-27T23:50:52.886488+0100: newcat_open: serial_handshake = Hardware 
2022-11-27T23:50:52.886488+0100:    3:newcat.c(6960):newcat_get_trn entered
2022-11-27T23:50:52.886488+0100: newcat_valid_command AI
2022-11-27T23:50:52.886488+0100: newcat.c(7427):newcat_valid_command returning2(1) 
2022-11-27T23:50:52.886488+0100:     4:newcat.c(10251):newcat_get_cmd entered
2022-11-27T23:50:52.886488+0100: rig_flush: called for serial device
2022-11-27T23:50:52.886488+0100: read_string_generic called, rxmax=4095 direct=1, expected_len=1
2022-11-27T23:50:53.085429+0100: 0000    ff fb 03 ff fb 01 ff fe 01 ff fd 00                 ............    
2022-11-27T23:50:53.085429+0100: cmd_str = AI;
2022-11-27T23:50:53.115974+0100: write_block(): TX 3 bytes, method=2
2022-11-27T23:50:53.115974+0100: 0000    41 49 3b                                            AI;             
2022-11-27T23:50:53.144530+0100: read_string_generic called, rxmax=129 direct=1, expected_len=1
2022-11-27T23:50:53.328648+0100: read_string_generic(): RX 4 characters, direct=1
2022-11-27T23:50:53.328648+0100: 0000    41 49 30 3b                                         AI0;            
2022-11-27T23:50:53.328648+0100: newcat_get_cmd: read count = 4, ret_data = AI0;
[... 216 lines cut ...]
2022-11-27T23:51:05.762440+0100: rigctl_parse: cmd=\(5c) handle=0
2022-11-27T23:51:05.762440+0100: rigctl_parse: cmd=send_raw handle=0
2022-11-27T23:51:05.762440+0100: rigctl_parse: vfo_opt=1
2022-11-27T23:51:05.762440+0100:   2:rig.c(5856):rig_get_powerstat entered
2022-11-27T23:51:05.762440+0100: rig.c(5876) trace
2022-11-27T23:51:05.762440+0100:    3:newcat.c(3470):newcat_get_powerstat entered
2022-11-27T23:51:05.762440+0100: newcat_valid_command PS
2022-11-27T23:51:05.762440+0100: newcat.c(7427):newcat_valid_command returning2(1) 
2022-11-27T23:51:05.762440+0100:     4:newcat.c(10251):newcat_get_cmd entered
2022-11-27T23:51:05.762440+0100: rig_flush: called for serial device
2022-11-27T23:51:05.762440+0100: read_string_generic called, rxmax=4095 direct=1, expected_len=1
2022-11-27T23:51:05.787733+0100: cmd_str = PS;
[... cut ...]
2022-11-27T23:51:06.063725+0100: newcat_get_cmd: read count = 4, ret_data = PS1;
2022-11-27T23:51:06.063725+0100:     4:newcat.c(10473):newcat_get_cmd returning(0) 
2022-11-27T23:51:06.063725+0100:    3:newcat.c(3503):newcat_get_powerstat returning(0) 
2022-11-27T23:51:06.063725+0100:   2:rig.c(5878):rig_get_powerstat returning(0) 
2022-11-27T23:51:06.063725+0100: rigctl_send_raw:
2022-11-27T23:51:06.063725+0100:   2:rig.c(7415):rig_send_raw entered
2022-11-27T23:51:06.063725+0100: rig_send_raw: writing 3 bytes
Rig command:

mdblack98, 2022-11-27:
What is this?

In riga:1 car:1
+ .\rigctl.exe -m 1035 -r COM5 -s 38400 -vvvvv -Z --vfo >log.txt 2>&1
+ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
+ CategoryInfo : NotSpecified: (2022-11-27T23:5...-vvvvv -Z --vfo:String) [], RemoteException
+ FullyQualifiedErrorId : NativeCommandError

Are you running this on a cmd shell?

Can we hook up using AnyDesk -- this runs just fine here.

Mike W9MDB

.\rigctl.exe : 2022-11-27T23:50:52.804436+0100: rigctl.c(445) Startup: 
C:\Users\Johnny\Documents\QT\CatRadio\hamlib\rigctl.exe -m 1035 -r COM5 -s 38400 -vvvvv -Z --vfo 
In riga:1 car:1 
+ .\rigctl.exe -m 1035 -r COM5 -s 38400 -vvvvv -Z --vfo >log.txt 2>&1 
+ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ 
+ CategoryInfo : NotSpecified: (2022-11-27T23:5...-vvvvv -Z --vfo:String) [], RemoteException 
+ FullyQualifiedErrorId : NativeCommandError 

2022-11-27T23:50:52.804436+0100: rigctl Hamlib 4.6~git Sun Nov 27 04:10:19 2022 +0000 SHA=09e73b 
2022-11-27T23:50:52.804436+0100: Report bugs to ***@***.***> 

2022-11-27T23:50:52.804436+0100: rig_check_rig_caps: p1=000000005FE2ACE0, p2=000000005FE31258, 
rig_model=000000005FE2ACE0, macro_name=000000005FE31258 
2022-11-27T23:50:52.804436+0100: yaesu: initrigs4_yaesu called 
2022-11-27T23:50:52.804436+0100: rig_init: rig_model=Yaesu FT-991 
2022-11-27T23:50:52.804436+0100: rig_init: rig has VFO_A 
2022-11-27T23:50:52.804436+0100: rig_init: rig has VFO_B 
2022-11-27T23:50:52.804436+0100: rig_init: rig has VFO_MEM 
2022-11-27T23:50:52.804436+0100: ft991_init called, version 20221104.14 
2022-11-27T23:50:52.804436+0100: 1:newcat.c(468):newcat_init entered 
2022-11-27T23:50:52.804436+0100: 1:newcat.c(508):newcat_init returning(0) 
2022-11-27T23:50:52.804436+0100: 1:rig.c(812):rig_open entered 
2022-11-27T23:50:52.804436+0100: rig_settings_get_path: path=.hamlib_settings 
2022-11-27T23:50:52.804436+0100: rig_settings_load_all: settings_file (.hamlib_settings): No such file or directory 
2022-11-27T23:50:52.804436+0100: rig_open: cwd=C:\Users\Johnny\Documents\QT\CatRadio\hamlib 
2022-11-27T23:50:52.804436+0100: rig_open: C:\Users\Johnny\Documents\QT\CatRadio\hamlib/hamlib_settings does not exist 
2022-11-27T23:50:52.804436+0100: rig_open: async_data_enable=0, async_data_supported=0 
2022-11-27T23:50:52.804436+0100: serial_open: COM5 
2022-11-27T23:50:52.804436+0100: ser_set_rts: RTS=0 
2022-11-27T23:50:52.804436+0100: ser_set_dtr: DTR=0 
2022-11-27T23:50:52.804436+0100: serial_setup: tcgetattr 
2022-11-27T23:50:52.804436+0100: serial_setup: cfsetispeed=38400,0x000f 
2022-11-27T23:50:52.804436+0100: serial_setup: cfsetospeed=38400,0x000f 
2022-11-27T23:50:52.804436+0100: serial_setup: data_bits=8 
2022-11-27T23:50:52.804436+0100: serial_setup: parity=0 
2022-11-27T23:50:52.804436+0100: serial_setup: Handshake=Hardware 
2022-11-27T23:50:52.804436+0100: serial_setup: tcsetattr TCSANOW 
2022-11-27T23:50:52.804436+0100: read_string_generic called, rxmax=4095 direct=1, expected_len=1 
2022-11-27T23:50:52.886488+0100: 2:rig.c(7227):async_data_handler_start entered 
2022-11-27T23:50:52.886488+0100: async_data_handler_start: async data support disabled since asynd_data_enabled=0 
2022-11-27T23:50:52.886488+0100: 2:rig.c(7234):async_data_handler_start returning(0) 
2022-11-27T23:50:52.886488+0100: rig.c(254):add_opened_rig returning2(0) 
2022-11-27T23:50:52.886488+0100: rig_open: 0000000000D59F4C rs->comm_state==1?=1 
2022-11-27T23:50:52.886488+0100: 2:newcat.c(548):newcat_open entered 
2022-11-27T23:50:52.886488+0100: newcat_open: Rig=FT-991, version=20221104.14 
2022-11-27T23:50:52.886488+0100: newcat_open: write_delay = 0 msec 
2022-11-27T23:50:52.886488+0100: newcat_open: post_write_delay = 25 msec 
2022-11-27T23:50:52.886488+0100: newcat_open: serial_handshake = Hardware 
2022-11-27T23:50:52.886488+0100: 3:newcat.c(6960):newcat_get_trn entered 
2022-11-27T23:50:52.886488+0100: newcat_valid_command AI 
2022-11-27T23:50:52.886488+0100: newcat.c(7427):newcat_valid_command returning2(1) 
2022-11-27T23:50:52.886488+0100: 4:newcat.c(10251):newcat_get_cmd entered 
2022-11-27T23:50:52.886488+0100: rig_flush: called for serial device 
2022-11-27T23:50:52.886488+0100: read_string_generic called, rxmax=4095 direct=1, expected_len=1 
2022-11-27T23:50:53.085429+0100: 0000 ff fb 03 ff fb 01 ff fe 01 ff fd 00 ............ 
2022-11-27T23:50:53.085429+0100: cmd_str = AI; 
[... 222 lines cut ...]
2022-11-27T23:51:05.762440+0100: rigctl_parse: cmd=\(5c) handle=0 
2022-11-27T23:51:05.762440+0100: rigctl_parse: cmd=send_raw handle=0 
2022-11-27T23:51:05.762440+0100: rigctl_parse: vfo_opt=1 
2022-11-27T23:51:05.762440+0100: 2:rig.c(5856):rig_get_powerstat entered 
2022-11-27T23:51:05.762440+0100: rig.c(5876) trace 
2022-11-27T23:51:05.762440+0100: 3:newcat.c(3470):newcat_get_powerstat entered 
2022-11-27T23:51:05.762440+0100: newcat_valid_command PS 
2022-11-27T23:51:05.762440+0100: newcat.c(7427):newcat_valid_command returning2(1) 
2022-11-27T23:51:05.762440+0100: 4:newcat.c(10251):newcat_get_cmd entered 
2022-11-27T23:51:05.762440+0100: rig_flush: called for serial device 
2022-11-27T23:51:05.762440+0100: read_string_generic called, rxmax=4095 direct=1, expected_len=1 
2022-11-27T23:51:05.787733+0100: cmd_str = PS; 
2022-11-27T23:51:05.862704+0100: write_block(): TX 3 bytes, method=2 
2022-11-27T23:51:05.862704+0100: 0000 50 53 3b PS; 
2022-11-27T23:51:05.888377+0100: read_string_generic called, rxmax=129 direct=1, expected_len=1 
[... cut ...]
2022-11-27T23:51:06.063725+0100: newcat_get_cmd: read count = 4, ret_data = PS1; 
2022-11-27T23:51:06.063725+0100: 4:newcat.c(10473):newcat_get_cmd returning(0) 
2022-11-27T23:51:06.063725+0100: 3:newcat.c(3503):newcat_get_powerstat returning(0) 
2022-11-27T23:51:06.063725+0100: 2:rig.c(5878):rig_get_powerstat returning(0) 
2022-11-27T23:51:06.063725+0100: rigctl_send_raw: 
2022-11-27T23:51:06.063725+0100: 2:rig.c(7415):rig_send_raw entered 
2022-11-27T23:51:06.063725+0100: rig_send_raw: writing 3 bytes 
Rig command:

PianetaRadio, 2022-11-28:
Yes, is the Windows 10 command shell.

PianetaRadio, 2022-11-29:
Progress, send works well. Get takes some extra bytes in the answer.

```
Rig command: \send_raw ; FA;
rigctl_parse: cmd=\(5c) handle=0
rigctl_parse: cmd=send_raw handle=0
rigctl_parse: vfo_opt=0
  2:rig.c(5856):rig_get_powerstat entered
rig.c(5876) trace
   3:newcat.c(3470):newcat_get_powerstat entered
newcat_valid_command PS
newcat.c(7427):newcat_valid_command returning2(1)
    4:newcat.c(10251):newcat_get_cmd entered
rig_flush: called for serial device
read_string_generic called, rxmax=4095 direct=1, expected_len=1
0000    ff fb 03 ff fb 01 ff fe 01 ff fd 00 ff fb 03 ff     ................
0010    fb 01 ff fe 01 ff fd 00                             ........
cmd_str = PS;
write_block(): TX 3 bytes, method=2
0000    50 53 3b                                            PS;
read_string_generic called, rxmax=129 direct=1, expected_len=1
read_string_generic(): RX 4 characters, direct=1
0000    50 53 31 3b                                         PS1;
newcat_get_cmd: read count = 4, ret_data = PS1;
    4:newcat.c(10473):newcat_get_cmd returning(0)
   3:newcat.c(3503):newcat_get_powerstat returning(0)
  2:rig.c(5878):rig_get_powerstat returning(0)
rigctl_send_raw:
  2:rig.c(7415):rig_send_raw entered
rig_send_raw: writing 3 bytes
write_block(): TX 3 bytes, method=2
0000    46 41 3b                                            FA;
rig_send_raw: reading frame terminated by ';'
read_string_generic called, rxmax=200 direct=1, expected_len=1
read_string_generic(): RX 12 characters, direct=1
0000    46 41 30 30 37 31 35 36 30 30 30 3b                 FA007156000;
  2:rig_send_raw: elapsed=534ms
  2:rig.c(7490):rig_send_raw returning(12)
Send raw answer: FA007156000;
rigctl_parse: called, interactive=1
```

```
Rig command: \send_raw ; ID;
rigctl_parse: cmd=\(5c) handle=0
rigctl_parse: cmd=send_raw handle=0
rigctl_parse: vfo_opt=0
rigctl_send_raw:
  2:rig.c(7415):rig_send_raw entered
rig_send_raw: writing 3 bytes
write_block(): TX 3 bytes, method=2
0000    49 44 3b                                            ID;
rig_send_raw: reading frame terminated by ';'
read_string_generic called, rxmax=200 direct=1, expected_len=1
read_string_generic(): RX 79 characters, direct=1
0000    ff fb 03 ff fb 01 ff fe 01 ff fd 00 ff fb 03 ff     ................
0010    fb 01 ff fe 01 ff fd 00 ff fb 03 ff fb 01 ff fe     ................
0020    01 ff fd 00 ff fb 03 ff fb 01 ff fe 01 ff fd 00     ................
0030    ff fb 03 ff fb 01 ff fe 01 ff fd 00 ff fb 03 ff     ................
0040    fb 01 ff fe 01 ff fd 00 49 44 30 36 37 30 3b        ........ID0670;
  2:rig_send_raw: elapsed=1476ms
  2:rig.c(7490):rig_send_raw returning(79)
Send raw answer:  ¹♥ ¹☺ ■☺ ²
rigctl_parse: called, interactive=1
```

```
Rig command: \send_raw ; FA007155000;
rigctl_parse: cmd=\(5c) handle=0
rigctl_parse: cmd=send_raw handle=0
rigctl_parse: vfo_opt=0
rigctl_send_raw:
  2:rig.c(7415):rig_send_raw entered
rig_send_raw: writing 12 bytes
write_block(): TX 12 bytes, method=2
0000    46 41 30 30 37 31 35 35 30 30 30 3b                 FA007155000;
rig_send_raw: reading frame terminated by ';'
read_string_generic called, rxmax=200 direct=1, expected_len=1
0000    ff fb 03 ff fb 01 ff fe 01 ff fd 00 ff fb 03 ff     ................
0010    fb 01 ff fe 01 ff fd 00 ff fb 03 ff fb 01 ff fe     ................
0020    01 ff fd 00 ff fb 03 ff fb 01 ff fe 01 ff fd 00     ................
0030    ff fb 03 ff fb 01 ff fe 01 ff fd 00 ff fb 03 ff     ................
0040    fb 01 ff fe 01 ff fd 00 ff fb 03 ff fb 01 ff fe     ................
0050    01 ff fd 00 ff fb 03 ff fb 01 ff fe 01 ff fd 00     ................
0060    ff fb 03 ff fb 01 ff fe 01 ff fd 00 ff fb 03 ff     ................
0070    fb 01 ff fe 01 ff fd 00 ff fb 03 ff fb 01 ff fe     ................
0080    01 ff fd 00 ff fb 03 ff fb 01 ff fe 01 ff fd 00     ................
0090    ff fb 03 ff fb 01 ff fe 01 ff fd 00                 ............
read_string_generic(): Timed out 5.964 seconds after 156 chars, direct=1
  2:rig_send_raw: elapsed=6049ms
  2:rig.c(7490):rig_send_raw returning(-8) Protocol error

Send raw answer:  ¹♥ ¹☺ ■☺ ²
rigctl_parse: called, interactive=1
```

```
Rig command: \send_raw ; FA;
rigctl_parse: cmd=\(5c) handle=0
rigctl_parse: cmd=send_raw handle=0
rigctl_parse: vfo_opt=0
rigctl_send_raw:
  2:rig.c(7415):rig_send_raw entered
rig_send_raw: writing 3 bytes
write_block(): TX 3 bytes, method=2
0000    46 41 3b                                            FA;
rig_send_raw: reading frame terminated by ';'
read_string_generic called, rxmax=200 direct=1, expected_len=1
read_string_generic(): RX 96 characters, direct=1
0000    ff fb 03 ff fb 01 ff fe 01 ff fd 00 ff fb 03 ff     ................
0010    fb 01 ff fe 01 ff fd 00 ff fb 03 ff fb 01 ff fe     ................
0020    01 ff fd 00 ff fb 03 ff fb 01 ff fe 01 ff fd 00     ................
0030    ff fb 03 ff fb 01 ff fe 01 ff fd 00 ff fb 03 ff     ................
0040    fb 01 ff fe 01 ff fd 00 ff fb 03 ff fb 01 ff fe     ................
0050    01 ff fd 00 46 41 30 30 37 31 35 35 30 30 30 3b     ....FA007155000;
  2:rig_send_raw: elapsed=1657ms
  2:rig.c(7490):rig_send_raw returning(96)
Send raw answer:  ¹♥ ¹☺ ■☺ ²
rigctl_parse: called, interactive=1
```

mdblack98, 2022-11-29:
You are getting random characters on your serial I/O and it's messing things up.

What OS is this on?  What serial cable?

Mike W9MDB

PianetaRadio, 2022-11-29:
Sorry, I'm using virtual serial port, configured in Telnet from server side, raw in the client side, so service commands weren't filtered. Now I put it in full raw mode and "bingo" is ok, let me try with my software CatRadio if I manage the same  good results.

Instead using NET rigctl, no answer:

```
Rig command: \send_raw ; FA;
rigctl_parse: cmd=\(5c) handle=0
rigctl_parse: cmd=send_raw handle=0
rigctl_parse: vfo_opt=1
rigctl_send_raw:
 1:rig.c(7415):rig_send_raw entered
rig_send_raw: writing 3 bytes
write_block(): TX 3 bytes, method=2
0000    46 41 3b                                            FA;
rig_send_raw: reading frame terminated by ';'
read_string_generic called, rxmax=200 direct=1, expected_len=1
read_string_generic(): Timed out 10.010 seconds after 0 chars, direct=1
 1:rig_send_raw: elapsed=10011ms
 1:rig.c(7490):rig_send_raw returning(-8) Protocol error

Send raw answer:
rigctl_parse: called, interactive=1
```

mdblack98, 2022-11-29:
Works from rigctl but not rigctld

PianetaRadio, 2022-11-30:
I confirm that is also ok from my application, using rig_send_raw function, tested with Yaesu.

Now the only remaining issue is with rigctld.