Hamlib/Hamlib #1165: Broken pkg-config test in hamlib-4.5.tar.gz

atsampson, 2022-11-30:
Something to bear in mind for the next release...

The `hamlib-4.5.tar.gz` tarball was built on a system that didn't have `pkg.m4` available. Because of [how the pkg-config test is written](https://github.com/Hamlib/Hamlib/blob/master/configure.ac#L459) in `configure.ac`, this means that the shipped `configure` script will never detect `pkg-config`, so it's not possible to enable XML support. (It works fine if you re-bootstrap before configuring.)

It might be better to remove the fallback code, or explicitly check that XML support works in release tarballs?

N0NB, 2022-12-01:
I will have to double check my Debian 10 VM to see if it is lacking `pkg.m4` which I presume should be supplied by Debian?  @atsampson If it is present, do you have time to help resolve this as I am sure the `pkg-config` section was written by Stephane many years ago, well before I started beating on the build system!

N0NB, 2022-12-01:
Well, whaddayaknow!

The `pkg-config` package was most certainly not installed in my VM.  It is now.  Sigh...

The daily source snapshot that is generated tomorrow morning should contain it if you would like to test:  https://n0nb.users.sourceforge.net/

N0NB, 2022-12-01:
The `pkg.m4` file is included in today's source snapshot.  Hopefully @atsampson can report back that it works.

atsampson, 2022-12-01:
Yep, that works correctly now:

```
$ ./configure --with-xml-support
[...]
checking for pkg-config... /gar/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for LIBXML2... yes
```

Thanks very much!

closed 2022-12-01 by mdblack98