Hamlib/Hamlib #1184: Having troubles including hamlib into CMake project

porkiedev, 2022-12-11:
I've been trying for the last few days to get CMake to include hamlib. I am using the prebuilt windows binaries for hamlib ver 4.5 and I've been copying the prebuilt binaries into my project source directory (though I do also have the same copy of hamlib actually installed on my PC with my environment variables setup)

Anyways, I'm new to C++ and CMake, so perhaps I am making a newbie mistake. I've had previous issues trying to incorporate libraries but managed to figure them out eventually. Hamlib has unfortunately not been so kind to me :(
To clarify: I did get CMake to import the hamlib headers. My main cpp file (hamlib_test.cpp) can indeed include the rig.h header file, and I can write code that my IDE (vs2019) recognizes and can auto-complete. Still, once I compile and run the project, it fails due to linker errors (LNK2019 unresolved external symbol rig_init reference in function main) <- and others similar. I figured this meant that I was including the header files but not the library itself. I tried everything I could think of and find online. I tried using CMake to import the prebuilt binaries' libhamlib-4.dll file, as well as the libhamlib-4.lib file under hamlib/lib/msvc, and even the libhamlib.dll.a file under hamlib/lib/gcc.
Perhaps this question is more fitting for the CMake support page but as hamlib doesn't support CMake by default (as far as I can tell), I am not so sure.
I'm using Windows 10 and Visual Studio 2019 as my development workspace. This is the farthest I have gotten:
In my root CMakeLists.txt file, I include the hamlib headers, and then I try to use the find_library command in CMake to include the libhamlib-4.lib file located under hamlib/lib/msvc. My program will compile just fine but when I run it, I get an unusual error (_The code execution cannot proceed because (null).DLL was not found. Reinstalling the program may fix this problem._)
This is my file structure as well as my CMakeLists file.
[image]
[image]

Any help would be greatly appreciated. Thanks for reading!

mdblack98, 2022-12-11:
This is really a cmake question and not Hamlib.

Here's a simple CMakeLists.txt example to compile a C or C++ Hamlib program.

https://www.dropbox.com/s/el1xc03c2p3vnc7/cmakehamlib.zip?dl=0

Extract it and do this

cd cmakehamlib
mkdir build
cd build
cmake ..
make

Mike W9DMB

porkiedev, 2022-12-13:
Unfortunately, this didn't help me figure out my include issue. I just don't know how to tell CMake to include hamlib in my project as a library. That said, you're right. This is indeed more appropriate for a CMake forum. Thanks for the help anyways!

closed 2022-12-13 by porkiedev

mdblack98, 2022-12-13:
Just this line replace testcpp with your file name.  But you need hamlib++ for C++.
target_link_libraries(testcpp LINK_PUBLIC hamlib++)