Hamlib/Hamlib #1194: Add RX2 control for PowerSDR/Thetis

mdblack98, 2022-12-28:
ZZTV00014074000;
ZZTV;
Sets or reads the transmit VFO frequency when RX2 is enabled.
To set/get status
ZZRS1;
ZZRS;
Will implement ZZRS to  check and use ZZTV to set RX2 frequency.

rsaue, 2022-12-28:
Hi @mdblack98,
as an ANAN-200D owner I am a bit disappointed on what i thought was Thetis inability to support independent control of RX2. Making a comment about this on the Apache Labs reflector today, I was told I was flat wrong, as Thetis today was said to have all the abilities to let 3d party programs control the RX2 (like the Flex and the Sunsdr-radios). 

Looking at Hamlib github site, the first thing I found was your newly opened issue #1194 !

I am no programmer but can compile from github og tar-balls on linux, and guess I can learn to do the same on windows. 

If you need an ANAN-user to test the code, please do not hesitate to contact me.

73 de LB2EG Richard

mdblack98, 2022-12-28:
It's really quite different than Flex or SunSDR.

Can you confirm the expected behavior?

RX2 is receive only, right?  Transmit still happens on VFOB.

So you need your 2nd instance of JTDX to read the RX2 VFO so it can set VFOB to the appropriate RX2 band.

Mike W9MDB

rsaue, 2022-12-28:
Hi, 
Yes my intention is to let JTDX (now on 2.59) read the Thetis VFOB QRG as RX2, but cannot get to that using CAT or TCP/IP CAT server. 

When RX2 is not activated, using split puts TX into VFO B.  

When RX2 is activated, VFO B shows the QRG on RX2. Now invoking split on RX1, does not put TX into VFOB. Instead there is a small window under VFO A that shows av split QRG with a offset, 

[image]

Using the CAT Command Tester window, I can now  read and write the TX QRG using the ZZTV; command. 

On JTDX i usually use fake split. Invoking rig split, activates split in Thetis VFOA as described above, but does not set the TX split QRG. Changing band using JTDX does not change the split TX QRG 

But still no show on RX2 using JTDX. 

73 de LB2EG Richard