Hamlib/Hamlib #1205: KX2 power calculation fails

jeffpc, 2023-01-06:
I have C code that uses rig_get_level & rig_power2mW to obtain the current power level setting.  This code works fine with FT-991A (model 1035) but fails with KX2 (model 2044).

Specifically, the call to rig_power2mW returns -1.  The values I pass in are 0.266667 for the power (but all the values I obtain from rig_get_level fail the same way; the range appears to be [0,0.6666667]), 7230000 for the frequency, and 2 for the mode (LSB).  The same appears to happen when I try to use \power2mW in rigctl:

```
$ rigctl -m 2044 -r /dev/cuaU1

Rig command: \get_level RFPOWER
Level Value: 0.266667

Rig command: \power2mW 0.266667 7230000 2
power2mW: error = Invalid parameter
rig.c(5698):rig_power2mW entered
rig.c(5717):rig_power2mW return(-1)
rigctl_parse.c(3038):rigctl_power2mW return(-1)

Rig command: \power2mW 0.266667 7230000 LSB
power2mW: error = Invalid parameter
rig.c(5698):rig_power2mW entered
rig.c(5717):rig_power2mW return(-1)
rigctl_parse.c(3038):rigctl_power2mW return(-1)
```

The host is running FreeBSD with hamlib 4.4.

I traced the "conversation" between hamlib and the KX2, and saw (H = hamlib, R = radio):

```
H: ID;
R: ID017;
H: OM;
R: OM A---------01;
H: K2;
R: K20;
H: K3;
R: K30;
H: RVM;
R: RVM03.02;
H: AI;
R: AI0;
H: AI0;
H: ID;
R: ID017;
H: IF;
R: IF00007230000     -000000 0001000001 ;
H: FA;
R: FA00007230000;
H: MD;
R: MD1;
H: BW;
R: BW0190;
H: PC;
R: PC001;
H: BN;
R: BN03;
```

Based on my reading of the K & KX programmer's reference, the requested power output level ("PC") uses a format in 1W increments unless the K2 extended mode is used at which point it uses 0.1W increments.  The KX2 allows power setting in 0.1W increments.

jeffpc, 2023-01-06:
Oh, and based on the hamlib debug output, it looks like power2mW just gives up:

```
kx3_get_level called
k3_get_level called
kenwood_safe_transaction called
kenwood_transaction called
kenwood_transaction: cmdstr = PC
rig_flush: called for serial device
serial.c(645):serial_flush entered
read_string called, rxmax=4095
serial.c(713):serial_flush return(0)
write_block(): TX 3 bytes
0000    50 43 3b                                            PC;             
read_string called, rxmax=17
read_string(): RX 6 characters
0000    50 43 30 30 34 3b                                   PC004;          
kenwood_transaction: read_string(len=6)='PC004;'
kenwood_transaction: returning RIG_OK, retval=0
kenwood.c(586):kenwood_transaction return(0)
kenwood.c(656):kenwood_safe_transaction return(0)
kenwood_safe_transaction called
kenwood_transaction called
kenwood_transaction: cmdstr = BN
rig_flush: called for serial device
serial.c(645):serial_flush entered
read_string called, rxmax=4095
serial.c(713):serial_flush return(0)
write_block(): TX 3 bytes
0000    42 4e 3b                                            BN;             
read_string called, rxmax=128
read_string(): RX 5 characters
0000    42 4e 30 33 3b                                      BN03;           
kenwood_transaction: read_string(len=5)='BN03;'
kenwood_transaction: returning RIG_OK, retval=0
kenwood.c(586):kenwood_transaction return(0)
kenwood.c(656):kenwood_safe_transaction return(0)
k3_get_maxpower: maxpower=15
rig.c(5698):rig_power2mW entered
rig.c(5717):rig_power2mW return(-1)
```

jeffpc, 2023-01-07:
Thanks for the prompt fix, but It doesn't appear to be enough.  With the KX2 set to 7.230 MHz CW and 4.8W:

```
$ ./rigctl -m 2044 -r /dev/cuaU1 
Rig command: \get_level RFPOWER
Level Value: 0.266667
Rig command: \power2mW 0.266667 7230000 CW
power2mW: error = read_string_generic called, rxmax=128 direct=1, expected_len=1
read_string_generic(): RX 5 characters, direct=1
kenwood_transaction: read_string(expected=128, len=5)='BN03;'
kenwood_transaction: returning RIG_OK, retval=0
kenwood.c(624):kenwood_transaction returning2(0) 
kenwood.c(695):kenwood_safe_transaction returning2(0) 
k3_get_maxpower: maxpower=15
 1:rigctl_parse.c(3384):rigctl_get_level returning(0) 
rigctl_parse: called, interactive=1
rigctl_parse: input_line: \power2mW 0.266667 7230000 CW
rigctl_parse: vfo_opt=0
 1:rigctl_parse.c(3139):rigctl_power2mW entered
rig_parse_mode called
  2:rig.c(5836):rig_power2mW entered
rig_power2mW: freq not in tx range
  2:rig.c(5867):rig_power2mW returning(-1) Invalid parameter

 1:rigctl_parse.c(3149):rigctl_power2mW returning(-1) Invalid parameter

Invalid parameter
```

Changing the mode doesn't help.

Switching to 14.07615 MHz CW 4.8W yields a number, but it is wrong:

```
Rig command: \get_level RFPOWER
Level Value: 0.266667
Rig command: \power2mW 0.266667 14076150 CW
Power mW: 2666
```

This is with the current 4.5.4 branch hamlib (df8c625cc9723a5200bf8e4cd27fc993687ab656).

In case it helps, here's the range related output from '1':

```
TX ranges #1 for ITU1:
	1810000 Hz - 2000000 Hz
		VFO list: VFOA VFOB 
		Mode list: AM CW USB LSB RTTY FM CWR RTTYR PKTLSB PKTUSB 
		Antenna list: ANT1 ANT2 
		Low power: 0.01 W, High power: 10 W
	3500000 Hz - 3800000 Hz
		VFO list: VFOA VFOB 
		Mode list: AM CW USB LSB RTTY FM CWR RTTYR PKTLSB PKTUSB 
		Antenna list: ANT1 ANT2 
		Low power: 0.01 W, High power: 10 W
	7000000 Hz - 7200000 Hz
		VFO list: VFOA VFOB 
		Mode list: AM CW USB LSB RTTY FM CWR RTTYR PKTLSB PKTUSB 
		Antenna list: ANT1 ANT2 
		Low power: 0.01 W, High power: 10 W
	10100000 Hz - 10150000 Hz
		VFO list: VFOA VFOB 
		Mode list: AM CW USB LSB RTTY FM CWR RTTYR PKTLSB PKTUSB 
		Antenna list: ANT1 ANT2 
		Low power: 0.01 W, High power: 10 W
	14000000 Hz - 14350000 Hz
		VFO list: VFOA VFOB 
		Mode list: AM CW USB LSB RTTY FM CWR RTTYR PKTLSB PKTUSB 
		Antenna list: ANT1 ANT2 
		Low power: 0.01 W, High power: 10 W
	18068000 Hz - 18168000 Hz
		VFO list: VFOA VFOB 
		Mode list: AM CW USB LSB RTTY FM CWR RTTYR PKTLSB PKTUSB 
		Antenna list: ANT1 ANT2 
		Low power: 0.01 W, High power: 10 W
	21000000 Hz - 21450000 Hz
		VFO list: VFOA VFOB 
		Mode list: AM CW USB LSB RTTY FM CWR RTTYR PKTLSB PKTUSB 
		Antenna list: ANT1 ANT2 
		Low power: 0.01 W, High power: 10 W
	24890000 Hz - 24990000 Hz
		VFO list: VFOA VFOB 
		Mode list: AM CW USB LSB RTTY FM CWR RTTYR PKTLSB PKTUSB 
		Antenna list: ANT1 ANT2 
		Low power: 0.01 W, High power: 10 W
	28000000 Hz - 29700000 Hz
		VFO list: VFOA VFOB 
		Mode list: AM CW USB LSB RTTY FM CWR RTTYR PKTLSB PKTUSB 
		Antenna list: ANT1 ANT2 
		Low power: 0.01 W, High power: 10 W
	50000000 Hz - 54000000 Hz
		VFO list: VFOA VFOB 
		Mode list: AM CW USB LSB RTTY FM CWR RTTYR PKTLSB PKTUSB 
		Antenna list: ANT1 ANT2 
		Low power: 0.01 W, High power: 10 W
RX ranges #1 for TBD:
	500000 Hz - 30000000 Hz
		VFO list: VFOA VFOB 
		Mode list: AM CW USB LSB RTTY FM CWR RTTYR PKTLSB PKTUSB 
		Antenna list: ANT1 ANT2 
	48000000 Hz - 54000000 Hz
		VFO list: VFOA VFOB 
		Mode list: AM CW USB LSB RTTY FM CWR RTTYR PKTLSB PKTUSB 
		Antenna list: ANT1 ANT2 
TX ranges #2 for TBD:
	1800000 Hz - 2000000 Hz
		VFO list: VFOA VFOB 
		Mode list: AM CW USB LSB RTTY FM CWR RTTYR PKTLSB PKTUSB 
		Antenna list: ANT1 ANT2 
		Low power: 0.01 W, High power: 10 W
	3500000 Hz - 4000000 Hz
		VFO list: VFOA VFOB 
		Mode list: AM CW USB LSB RTTY FM CWR RTTYR PKTLSB PKTUSB 
		Antenna list: ANT1 ANT2 
		Low power: 0.01 W, High power: 10 W
	7000000 Hz - 7300000 Hz
		VFO list: VFOA VFOB 
		Mode list: AM CW USB LSB RTTY FM CWR RTTYR PKTLSB PKTUSB 
		Antenna list: ANT1 ANT2 
		Low power: 0.01 W, High power: 10 W
	10100000 Hz - 10150000 Hz
		VFO list: VFOA VFOB 
		Mode list: AM CW USB LSB RTTY FM CWR RTTYR PKTLSB PKTUSB 
		Antenna list: ANT1 ANT2 
		Low power: 0.01 W, High power: 10 W
	14000000 Hz - 14350000 Hz
		VFO list: VFOA VFOB 
		Mode list: AM CW USB LSB RTTY FM CWR RTTYR PKTLSB PKTUSB 
		Antenna list: ANT1 ANT2 
		Low power: 0.01 W, High power: 10 W
	18068000 Hz - 18168000 Hz
		VFO list: VFOA VFOB 
		Mode list: AM CW USB LSB RTTY FM CWR RTTYR PKTLSB PKTUSB 
		Antenna list: ANT1 ANT2 
		Low power: 0.01 W, High power: 10 W
	21000000 Hz - 21450000 Hz
		VFO list: VFOA VFOB 
		Mode list: AM CW USB LSB RTTY FM CWR RTTYR PKTLSB PKTUSB 
		Antenna list: ANT1 ANT2 
		Low power: 0.01 W, High power: 10 W
	24890000 Hz - 24990000 Hz
		VFO list: VFOA VFOB 
		Mode list: AM CW USB LSB RTTY FM CWR RTTYR PKTLSB PKTUSB 
		Antenna list: ANT1 ANT2 
		Low power: 0.01 W, High power: 10 W
	28000000 Hz - 29700000 Hz
		VFO list: VFOA VFOB 
		Mode list: AM CW USB LSB RTTY FM CWR RTTYR PKTLSB PKTUSB 
		Antenna list: ANT1 ANT2 
		Low power: 0.01 W, High power: 10 W
	50000000 Hz - 54000000 Hz
		VFO list: VFOA VFOB 
		Mode list: AM CW USB LSB RTTY FM CWR RTTYR PKTLSB PKTUSB 
		Antenna list: ANT1 ANT2 
		Low power: 0.01 W, High power: 10 W
RX ranges #2 for TBD:
	500000 Hz - 30000000 Hz
		VFO list: VFOA VFOB 
		Mode list: AM CW USB LSB RTTY FM CWR RTTYR PKTLSB PKTUSB 
		Antenna list: ANT1 ANT2 
	48000000 Hz - 54000000 Hz
		VFO list: VFOA VFOB 
		Mode list: AM CW USB LSB RTTY FM CWR RTTYR PKTLSB PKTUSB 
		Antenna list: ANT1 ANT2 
```

(The remaining ranges are empty.)

jeffpc, 2023-01-07:
Thanks again.

Trying latest master branch (eb904148b5a35dc1b13d70076370125658671778) things work but aren't quite as precise as one would like.

1. \power2mW seems to work ok now
2. \get_level still reports very coarse power information

It is using "PC" to get the current power level which gives the KX2 responds with anything between "PC000;" (near zero/zero power) and "PC010;" (10W).  This is not enough precision to be able to calculate the power accurately in \power2mW.

Here's a sampling of what happens.  KX2 power setting -> \get_level value -> \power2mW result:

* 10W -> 0.666667 -> 10W
* 9.8W -> 0.600000 -> 9W
* 9.0W -> 0.600000 -> 9W
* 6.6W -> 0.400000 -> 6W
* 6.0W -> 0.400000 -> 6W
* 1.0W -> 0.066667 -> 1W
* 0.9W -> 0.000000 -> 0W

This is all the precision that the standard PC format offers, but the Elecraft doc suggests that the "K2 extended mode" gives 0.1W resolution instead of just 1W resolution.  The doc says:

> K2 Extended RSP format (K22): PCnnnx; where nnn is power, and x is 0 (low range) or 1 (high range). Notes: (1) A lower power limit may be in effect on trasnsverter bands (menu entries X V TR 1-9). (3) The MN and DS command can be used to select other settings of the KPA3 MD menu parameter.

Earlier, the manual describes the K22 mode as:

> K22, K2 Extended mode: Enables all K2 command extensions.

At least skimming through the manual, it looks like it should be safe to enable the extended mode on rig open and then interpret the PC value appropriately.  However, for all I know there are assumptions in some of the other Kenwood/Elecraft rig code that'd trip up on possible changes in FW, ID, and KY behavior.

jeffpc, 2023-01-07:
Seems to work much better now (most of the values are spot-on).  There is a tiny bit of a rounding error for some of the power settings, but when it is off, it is off by only 1mW.  As before, KX2 power setting -> \get_level value -> \power2mW output:

0.3W -> 0.02 -> 0.299W
0.5W -> 0.033333 -> 0.499W
9.0W -> 0.600000 -> 9.000W
9.8W -> 0.653333 -> 9.799W

For example, during the last one (9.8W) the KX2 responds to PC with: PC0980;

My (unsubstantiated) guess is that it is the float to unsigned int conversion that isn't getting the rounding just right.

Thanks.

mdblack98, 2023-01-07:
If you print those values with "%g" you should get the right answer.

Mike W9MDB

jeffpc, 2023-01-07:
I'm not sure how %g would help.  The power (in mW) returned by rig_power2mW is an unsigned int.  rigctl prints it as %i and that value is already wrong.  IOW, it is *not* a printing issue:

```
Rig command: \get_level RFPOWER
Level Value: 0.653333

Rig command: \power2mW 0.653333 7204210 LSB
Power mW: 9799
```

mdblack98, 2023-01-10:
Should be reflecting correct values now

closed 2023-01-10 by mdblack98

jeffpc, 2023-01-10:
Confirmed, \power2mW seems to show the correct number now.  Thanks.