Hamlib/Hamlib #1207: RPI OS update seems to break hamlib

murtrah57, 2023-01-07:
I just did the latest updates to the RPI OS and packages. Now rigctld and rotctld will not run.

mdblack98, 2023-01-07:
I need a whole lot more information than "will not run".  I assume there's an error of some sort?

Mike W9MDB

murtrah57, 2023-01-07:
when I try to run either rigctl or rotctl here is what I get:

pi@hampi:~ $ rigctl
rigctl: symbol lookup error: rigctl: undefined symbol: lock_mode
pi@hampi:~ $ rotctl
rotctl: symbol lookup error: rotctl: undefined symbol: debugmsgsave2
pi@hampi:~ $

murtrah57, 2023-01-07:
I have a crontab entry that autostarts both rigctld and rotctld on startup. It also checks once per minute to see if they are running. If not it restarts them. My logs show both rigctld and rotctld are both trying to be restarted every minute. 

Everything was working fine before I did a package update this morning.

mdblack98, 2023-01-07:
Please run rigctld manually with "-vvvvv" added and send me a capture of the log.

Mike

mdblack98, 2023-01-07:
Then you have an old libhamlib sitting around.

When you install from source it defaults to /usr/local/bin and you probably have libhamlib.so in /usr/lib/ too.

If you have a hamlib package installed then you need to uninstall it when compiling it yourself.

Mike W9MDB

murtrah57, 2023-01-07:
I did a complete search of my file system from root level down. I have only one copy of libhamlib.so
I do not have another copy or version in /usr/lib

See screen shot attached.

[image]

murtrah57, 2023-01-07:
Not sure why but now it seems to be running again.  ??????
Sorry to have taken up your time.

closed 2023-01-07 by murtrah57