Hamlib/Hamlib #1208: Building hamlib while hamlib is installed fails the build

jeffpc, 2023-01-07:
I have the OS's hamlib package installed but I tried to build the latest git code to try out a recent bugfix.  The hamlib build tried to use rig.h and libhamlib.so.4 from the system directories instead of forcing the use of the ones in the source tree.  Needless to say, the mismatch caused various unexpected failures.

I saw this happen on FreeBSD, but I don't see any reason it shouldn't fail on other OSes as well.

Probably the easiest way to reproduce it reliably is to install the OS's package and hand-edit the installed rig.h to contain something like:
```
#error "should not be used"
```

Building hamlib from source should not pick up the system rig.h and therefore shouldn't error out.

Ultimately, it is an include directory ordering issue.  `-I../include` should be ahead of any `-I` directives for installed header dirs.  (On FreeBSD, external software ends up in /usr/local which may be why autotools inserts `-I/usr/local/include` into CPPFLAGS.)

N0NB, 2023-01-07:
> Ultimately, it is an include directory ordering issue. -I../include should be ahead of any -I directives for installed header dirs. 

On a BSD that is probably true, but on GNU based systems, as are most Linux distributions, I have found it to be the opposite.  One would likely need to take this up with GNU coding standards and some such.  My understanding is that locally installed headers, i.e. in `/usr/local/incude` are considered to be preferred over system installed ones.

Other than a collision of philosophies, I don't think anything is "wrong" with either BSD or Hamlib in this regard.

jeffpc, 2023-01-07:
I'm not sure if I misunderstood you, or you misunderstood me.  So, just to make sure, let me try to rephrase.

On FreeBSD, OS-provided external software ("ports") are installed in /usr/local.  The base system (kernel, basic userspace) is in /usr.  (On Linux, both are in /usr.)

I run into *build errors* because CPPFLAGS includes -I/usr/local/include *before* -I../include.  This makes the compiler find rig.h from the installed binary package instead of find rig.h in the source tree I'm trying to compile.

Any software's build system should use its own headers in the source tree instead of using same-named headers in /usr/include or /usr/local/include.  In other words, this is *not* about /usr/local/include vs. /usr/include, but rather about ${path-to-hamlib-source}/include vs. /usr/local/include.

mdblack98, 2023-01-07:
We should be able to put in a test for /usr/include/hamlib/rig.h and die with a message like:

"Do not install a 2nd hamlib version.  You should uninstall the system hamlib package first".

This is actually a major problem with Linux versions (or any Unix really) as any apps that are compiled with a prior version will cease to work when libhamlib.so is replaced.  Windows can put the DLL in the directory so upgrading Hamlib won't break Windows applications.  Linux versions are also prohibited from static linking so that can't be done either.

I'm thinking we need to migrate to a different include path using the major version (compatibility indicator) and also include the major/minor version on the libraries too.

Seems the shared library versions should also show the full version number which they don't now.

lrwxrwxrwx 1 root root       20 Jan  7 11:42 /usr/local/lib/libhamlib++.so -> libhamlib++.so.4.0.6
lrwxrwxrwx 1 root root       18 Jan  5 23:10 /usr/local/lib/libhamlib.so -> libhamlib.so.4.0.6
lrwxrwxrwx 1 root root       20 Jan  7 11:42 /usr/local/lib/libhamlib++.so.4 -> libhamlib++.so.4.0.6
lrwxrwxrwx 1 root root       18 Jan  5 23:10 /usr/local/lib/libhamlib.so.4 -> libhamlib.so.4.0.6
-rwxr-xr-x 1 root root   276336 Oct 29 23:04 /usr/local/lib/libhamlib++.so.4.0.5
-rwxr-xr-x 1 root root 18538088 Jan  7 11:42 /usr/local/lib/libhamlib++.so.4.0.6
-rwxr-xr-x 1 root root 23494648 Jan  5 23:10 /usr/local/lib/libhamlib.so.4.0.6

Then one could compile against major versions and link against incremental versions if desired.  This would maximize backwards capability.

Mike W9MDB

On a BSD that is probably true, but on GNU based systems, as are most Linux distributions, I have found it to be the opposite. One would likely need to take this up with GNU coding standards and some such. My understanding is that locally installed headers, i.e. in /usr/local/incude are considered to be preferred over system installed ones.

Other than a collision of philosophies, I don't think anything is "wrong" with either BSD or Hamlib in this regard.

mdblack98, 2023-01-07:
What are you compiling that is showing that path problem?

Mike

jeffpc, 2023-01-07:
On Sat, Jan 07, 2023 at 10:51:17 -0800, Michael Black wrote:
> We should be able to put in a test for /usr/include/hamlib/rig.h and die
> with a message like:
> 
> "Do not install a 2nd hamlib version.  You should uninstall the system
> hamlib package first".

While that would catch the scenario and provide an obvious reason for the
build failure, why not instead fix the order of include paths and make the
build less fragile?

The suggested check would be really annoying in at least two scenarios:

(1) A user has hamlib installed on the system and wants to do a quick build
    of latest changes to verify that things are better - the user doesn't
    want to install the new hamlib version, to build it and run it from
    the source directory.
(2) A user has hamlib installed on the system and wants to upgrade by
    building from source and then running 'make install'.

The first scenario is what I ran into.  FreeBSD updates its packages rather
promptly, so I don't feel the need to build software myself except to verify
that upstream has a correct fix.

The suggested check would mean that one must remove installed hamlib before
building a new one.  This is rather disruptive and not the expected
behavior.  (One expects ./configure && make to build no matter if the
software is already installed.)  Requiring removal prior to building also
increases the chance that the system will be left unsable if the new version
doesn't (for any reason) build.

> This is actually a major problem with Linux versions (or any Unix really)
> as any apps that are compiled with a prior version will cease to work when
> libhamlib.so is replaced.
...
> Then one could compile against major versions and link against incremental
> versions if desired.  This would maximize backwards capability.

Yes.

I'd argue that this is a related but a separate problem - this is a problem
for *consumers* of hamlib.  Hamlib itself shouldn't use headers of a
different hamlib version no matter where they are - it has all it needs
right in the source directory.

As an example, this is what it looks like when I have
/usr/local/include/hamlib/rig.h:

```
Making all in rigs/adat
depbase=`echo adt_200a.lo | sed 's|[^/]*$|.deps/&|;s|\.lo$||'`; /bin/sh ../../libtool  --tag=CC    --mode=compile cc -DHAVE_CONFIG_H  -I. -I../../include/hamlib  -I/usr/local/include  -DIN_HAMLIB -I../../include -I../../src -I../../lib -I../../include   -Wall -g -O2 -MT adt_200a.lo -MD -MP -MF $depbase.Tpo -c -o adt_200a.lo adt_200a.c && mv -f $depbase.Tpo $depbase.Plo
libtool: compile:  cc -DHAVE_CONFIG_H -I. -I../../include/hamlib -I/usr/local/include -DIN_HAMLIB -I../../include -I../../src -I../../lib -I../../include -Wall -g -O2 -MT adt_200a.lo -MD -MP -MF .deps/adt_200a.Tpo -c adt_200a.c  -fPIC -DPIC -o .libs/adt_200a.o
adt_200a.c:209:30: error: use of undeclared identifier 'HAMLIB_CHECK_RIG_CAPS'
    .hamlib_check_rig_caps = HAMLIB_CHECK_RIG_CAPS
                             ^
```

Note that /usr/local/include is ahead of ../../include.  Therefore, the
hamlib/rig.h #include on line 40 includes /usr/local/include/hamlib/rig.h.
Since I have hamlib 4.4 installed, it doesn't have HAMLIB_CHECK_RIG_CAPS and
the build errors out.

Hmm...  I'm not sure if this is the right fix, but the following change to
configure.ac seems to make the build finish even with the installed hamlib
version.

```
diff --git a/configure.ac b/configure.ac
index fcaffc79..0af90a70 100644
--- a/configure.ac
+++ b/configure.ac
@@ -273,7 +273,7 @@ dnl The host_os variable is set by the AC_CANONICAL_HOST macro above.
 AM_CONDITIONAL([VERSIONDLL], false)
 AS_CASE(["$host_os"],
     [freebsd*], [
-	AM_CPPFLAGS="-I/usr/local/include ${AM_CPPFLAGS}"
+	AM_CPPFLAGS="${AM_CPPFLAGS} -I/usr/local/include"
 	AM_LDFLAGS="${AM_LDFLAGS} -L/usr/local/lib"
 	AC_SUBST([AM_LDFLAGS])],
 
```

closed 2023-01-11 by mdblack98