Hamlib/Hamlib #1219: FT710 and FLDIGI

wx4cb, 2023-01-15:
So i have a shortcut that starts hamlib with the following command line.

````
D:\radio\hamlib\bin\rigctld.exe -vv --model=1049 --rig-file=COM4 --serial-speed=38400 -Z --vfo
````

Log4OM, WSJTX works fine with hamlib and the FT710, however when i try and run FLDIGI I get the following error:

````
2023-01-14T21:07:05.065847-0500: rig_parse_vfo: 'f' not found so vfo='None'
````

FL digi is set to use the "hamlib NET" radio type. Here is a -vvvv dump of the start up and the startup of fldigi

````
2023-01-14T21:09:19.864464-0500: rigctld.c(664) Startup: D:\radio\hamlib\bin\rigctld.exe -vvvv --model=1049 --rig-file=COM4 --serial-speed=38400 -Z --vfo
2023-01-14T21:09:19.864464-0500: rigctld Hamlib 4.5.4 Jan 10 01:31:41Z 2023 SHA=921cc5
2023-01-14T21:09:19.865467-0500: Report bugs to <hamlib-developer@lists.sourceforge.net>

2023-01-14T21:09:19.865467-0500: Max# of rigctld client services=32
2023-01-14T21:09:19.865467-0500: yaesu: initrigs4_yaesu called
2023-01-14T21:09:19.865467-0500: rig_init: rig_model=Yaesu FT-710
2023-01-14T21:09:19.865467-0500: rig_init: rig has VFO_A
2023-01-14T21:09:19.865467-0500: rig_init: rig has VFO_B
2023-01-14T21:09:19.866469-0500: rig_init: rig has VFO_MEM
2023-01-14T21:09:19.866469-0500:  1:newcat.c(469):newcat_init entered
2023-01-14T21:09:19.866469-0500:  1:newcat.c(509):newcat_init returning(0)
2023-01-14T21:09:19.866469-0500:  1:rig.c(817):rig_open entered
2023-01-14T21:09:19.867471-0500: rig_settings_load_all: settings_file (.hamlib_settings): No such file or directory
2023-01-14T21:09:19.867471-0500: rig_open: cwd=d:\radio\hamlib
2023-01-14T21:09:19.867471-0500: rig_open: d:\radio\hamlib/hamlib_settings does not exist
2023-01-14T21:09:19.867471-0500: serial_open: COM4
2023-01-14T21:09:19.955466-0500:   2:rig.c(7434):async_data_handler_start entered
2023-01-14T21:09:19.955466-0500:   2:rig.c(7441):async_data_handler_start returning(0)
2023-01-14T21:09:19.955466-0500: rig.c(254):add_opened_rig returning2(0)
2023-01-14T21:09:19.955466-0500: rig_open: 0000000001019D9C rs->comm_state==1?=1
2023-01-14T21:09:19.955466-0500:   2:newcat.c(551):newcat_open entered
2023-01-14T21:09:19.955466-0500:    3:newcat.c(7179):newcat_get_trn entered
2023-01-14T21:09:19.955466-0500: newcat.c(7686):newcat_valid_command returning2(1)
2023-01-14T21:09:19.955466-0500:     4:newcat.c(10470):newcat_get_cmd entered
2023-01-14T21:09:19.987465-0500:     4:newcat.c(10692):newcat_get_cmd returning(0)
2023-01-14T21:09:19.987465-0500:    3:newcat.c(7211):newcat_get_trn returning(0)
2023-01-14T21:09:19.987465-0500:    3:newcat.c(10324):newcat_get_rigid entered
2023-01-14T21:09:19.987465-0500:     4:newcat.c(10470):newcat_get_cmd entered
2023-01-14T21:09:20.017926-0500:     4:newcat.c(10692):newcat_get_cmd returning(0)
2023-01-14T21:09:20.018925-0500:    3:newcat.c(10347):newcat_get_rigid returning(800)
2023-01-14T21:09:20.019924-0500: newcat_open: rig_id=800
2023-01-14T21:09:20.019924-0500:   2:newcat.c(667):newcat_open returning(0)
2023-01-14T21:09:20.019924-0500:   2:rig.c(2884):rig_get_vfo entered
2023-01-14T21:09:20.020926-0500:    3:newcat.c(1785):newcat_get_vfo entered
2023-01-14T21:09:20.020926-0500: newcat.c(7686):newcat_valid_command returning2(1)
2023-01-14T21:09:20.020926-0500:     4:newcat.c(10470):newcat_get_cmd entered
2023-01-14T21:09:20.049186-0500:     4:newcat.c(10692):newcat_get_cmd returning(0)
2023-01-14T21:09:20.049186-0500:     4:newcat.c(10364):newcat_get_vfo_mode entered
2023-01-14T21:09:20.052189-0500: newcat.c(7686):newcat_valid_command returning2(1)
2023-01-14T21:09:20.053185-0500:      5:newcat.c(10470):newcat_get_cmd entered
2023-01-14T21:09:20.082243-0500:      5:newcat.c(10692):newcat_get_cmd returning(0)
2023-01-14T21:09:20.082243-0500:     4:newcat.c(10419):newcat_get_vfo_mode returning(0)
2023-01-14T21:09:20.083247-0500:    3:newcat.c(1847):newcat_get_vfo returning(0)
2023-01-14T21:09:20.084283-0500:   2:rig.c(2947):rig_get_vfo returning(0)
2023-01-14T21:09:20.086245-0500: rig_get_freq(1990) called vfo=VFOA
2023-01-14T21:09:20.087243-0500: rig.c(1999) vfo=VFOA, curr_vfo=VFOA
2023-01-14T21:09:20.087243-0500: rig_get_freq(2086): vfo_opt=0, model=1049
2023-01-14T21:09:20.087243-0500:   2:newcat.c(1362):newcat_get_freq entered
2023-01-14T21:09:20.088242-0500: newcat.c(7686):newcat_valid_command returning2(1)
2023-01-14T21:09:20.088242-0500: newcat.c(7686):newcat_valid_command returning2(1)
2023-01-14T21:09:20.089242-0500:    3:newcat.c(7885):newcat_set_vfo_from_alias entered
2023-01-14T21:09:20.089242-0500:    3:newcat.c(7943):newcat_set_vfo_from_alias returning(0)
2023-01-14T21:09:20.090241-0500: newcat.c(7686):newcat_valid_command returning2(1)
2023-01-14T21:09:20.090241-0500:    3:newcat.c(10470):newcat_get_cmd entered
[... 236 lines cut ...]
2023-01-14T21:10:20.195398-0500:       6:newcat.c(10692):newcat_get_cmd returning(0)
2023-01-14T21:10:20.195398-0500:       6:newcat.c(11120):newcat_rmode_width entered
2023-01-14T21:10:20.198399-0500:        7:rig.c(2563):rig_passband_normal entered
2023-01-14T21:10:20.199400-0500:        7:rig_passband_normal: return filter#0, width=600
2023-01-14T21:10:20.199400-0500:        7:rig.c(2584):rig_passband_normal returning(600)
2023-01-14T21:10:20.200398-0500:        7:newcat.c(9134):newcat_get_rx_bandwidth entered
2023-01-14T21:10:20.201401-0500: newcat.c(7686):newcat_valid_command returning2(1)
2023-01-14T21:10:20.203410-0500:         8:newcat.c(7885):newcat_set_vfo_from_alias entered
2023-01-14T21:10:20.204397-0500:         8:newcat.c(7943):newcat_set_vfo_from_alias returning(0)
2023-01-14T21:10:20.205396-0500:         8:newcat.c(10470):newcat_get_cmd entered
2023-01-14T21:10:20.243396-0500:         8:newcat.c(10692):newcat_get_cmd returning(0)
2023-01-14T21:10:20.243396-0500:         8:rig.c(2667):rig_passband_wide entered
2023-01-14T21:10:20.245402-0500:         8:rig.c(2687):rig_passband_wide returning(2400)
2023-01-14T21:10:20.245402-0500:        7:newcat.c(10249):newcat_get_rx_bandwidth returning(0)
2023-01-14T21:10:20.246401-0500:       6:newcat.c(1609):newcat_get_mode returning(0)
2023-01-14T21:10:20.247399-0500:       6:cache.c(35):rig_set_cache_mode entered
2023-01-14T21:10:20.248407-0500:       6:cache.c(108):rig_set_cache_mode returning(0)
2023-01-14T21:10:20.249397-0500:      5:rig.c(2542):rig_get_mode returning(0)
2023-01-14T21:10:20.249397-0500:     4:rigctl_parse.c(2260):rigctl_get_mode returning(0)
2023-01-14T21:10:20.250406-0500: mutex_rigctld: client lock disengaged
2023-01-14T21:10:20.250406-0500: mutex_rigctld: client lock engaged
2023-01-14T21:10:20.252396-0500: mutex_rigctld: client lock disengaged
2023-01-14T21:10:20.552738-0500: handle_socket: rigctl_parse retcode=1
2023-01-14T21:10:20.553214-0500: Connection closed from ::ffff:127.0.0.1:63863
````

mdblack98, 2023-01-15:
FLDigi does not handle the --vfo switch correctly.

Mike W9MDB

mdblack98, 2023-01-15:
What version of FLDigi?

I'm testing 4.1.23.24 and it is working OK.

http://www.w1hkj.com/alpha/fldigi/

Mike W9MDB

wx4cb, 2023-01-15:
hi michael....

it's 4.1.23 - whatever the latest version is downloadable. I downloaded it
last night... I am getting assertion errors when i start it, but from what
i read that's normal at the minute.

charlie

mdblack98, 2023-01-15:
Try the alpha version please.  I believe FLDigi updated the hamlib version recently.

Mike W9MDB

wx4cb, 2023-01-17:
Hi mike

i just installed that and it's getting better I can at least get the frequency now lol, but now i get the following error if i try to jump to a frequency from log4oms bandmap/cluster thing or even just clicking a value down in the frequency window i get this error

now please bear in mind, i'm not saying it's hamlibs fault, in the slightest, but I want to be sure it's not anything im doing wrong.
````
Opened rig model 1049, 'FT-710'

*here i connected log4om*

2023-01-16T20:06:20.500998-0500: newcat_get_cmd: Rig busy - retrying 1 of 3: 'SH1;'
2023-01-16T20:06:21.141587-0500: newcat_get_cmd: Rig busy - retrying 2 of 3: 'SH1;'
2023-01-16T20:06:21.773817-0500: newcat_get_cmd: Rig busy - retrying 3 of 3: 'SH1;'
2023-01-16T20:06:22.414251-0500: newcat_get_cmd: Rig busy - retrying 4 of 3: 'SH1;'
2023-01-16T20:06:23.019678-0500: rigctl_get_vfo_info: vfo=VFOB
2023-01-16T20:06:23.645975-0500: newcat_get_cmd: Rig busy - retrying 1 of 3: 'SH1;'
2023-01-16T20:06:24.292978-0500: newcat_get_cmd: Rig busy - retrying 2 of 3: 'SH1;'
2023-01-16T20:06:24.928748-0500: newcat_get_cmd: Rig busy - retrying 3 of 3: 'SH1;'
2023-01-16T20:06:25.570111-0500: newcat_get_cmd: Rig busy - retrying 4 of 3: 'SH1;'
2023-01-16T20:06:26.174634-0500: rigctl_get_vfo_info: vfo=VFOB
2023-01-16T20:06:26.805861-0500: newcat_get_cmd: Rig busy - retrying 1 of 3: 'SH1;'
2023-01-16T20:06:27.440290-0500: newcat_get_cmd: Rig busy - retrying 2 of 3: 'SH1;'
2023-01-16T20:06:28.073777-0500: newcat_get_cmd: Rig busy - retrying 3 of 3: 'SH1;'
2023-01-16T20:06:28.717853-0500: newcat_get_cmd: Rig busy - retrying 4 of 3: 'SH1;'
2023-01-16T20:06:29.327237-0500: rigctl_get_vfo_info: vfo=VFOB

*HERE I clicked on a call in the band map to change the VFO frequency*

2023-01-16T20:06:29.389736-0500: rigctl_set_freq: chkarg err
rig_pathname: "Path name to the device file of the rig"
        Default: /dev/rig, Value: COM4
        String.
write_delay: "Delay in ms between each byte sent out"
        Default: 0, Value: 0
        Range: 0.0..1000.0, step 1.0
post_write_delay: "Delay in ms between each command sent out"
        Default: 0, Value: 0
        Range: 0.0..1000.0, step 1.0
timeout: "Timeout in ms"
        Default: 0, Value: 1000
        Range: 0.0..10000.0, step 1.0
retry: "Max number of retry"
        Default: 0, Value: 3
        Range: 0.0..10.0, step 1.0
Selected range list: "The tx/rx range list in use"
        Default: 0, Value:
        Range: 1.0..5.0, step 1.0
Selected range list: "The tx/rx range list name"
        Default: Default, Value:
        String.
vfo_comp: "VFO compensation in ppm"
        Default: 0, Value: 0.000000
        Range: 0.0..1000.0, step 0.0
poll_interval: "Polling interval in milliseconds for transceive emulation, value of 0 disables polling"
        Default: 0, Value: 0
        Range: 0.0..1000000.0, step 1.0
ptt_type: "Push-To-Talk interface type override"
        Default: RIG, Value: RIG
        Combo: RIG, RIGMICDATA, DTR, RTS, Parallel, CM108, GPIO, GPION, None
ptt_pathname: "Path name to the device file of the Push-To-Talk"
        Default: /dev/rig, Value:
        String.
ptt_bitnum: "Push-To-Talk GPIO bit number"
        Default: 2, Value: 0
        Range: 0.0..7.0, step 1.0
dcd_type: "Data Carrier Detect (or squelch) interface type override"
        Default: RIG, Value: None
        Combo: RIG, DSR, CTS, CD, Parallel, CM108, GPIO, GPION, None
dcd_pathname: "Path name to the device file of the Data Carrier Detect (or squelch)"
        Default: /dev/rig, Value:
[... 448 lines cut ...]
        Default: (null), Value:
        Check button.
CONTOUR_FREQ: "Contour frequency"
        Default: (null), Value:
        Range: 10.0..3200.0, step 1.0
CONTOUR_LEVEL: "Contour level (dB)"
        Default: (null), Value:
        Range: -40.0..20.0, step 1.0
CONTOUR_WIDTH: "Contour width"
        Default: (null), Value:
        Range: 1.0..11.0, step 1.0
Command '5' not found!
Command '7' not found!
Command '0' not found!
Command '0' not found!
Command '0' not found!
2023-01-16T20:06:29.927717-0500: newcat_get_cmd: Rig busy - retrying 1 of 3: 'SH1;'
2023-01-16T20:06:30.563091-0500: newcat_get_cmd: Rig busy - retrying 2 of 3: 'SH1;'
2023-01-16T20:06:31.213094-0500: newcat_get_cmd: Rig busy - retrying 3 of 3: 'SH1;'
2023-01-16T20:06:31.863772-0500: newcat_get_cmd: Rig busy - retrying 4 of 3: 'SH1;'
2023-01-16T20:06:32.477116-0500: rigctl_get_vfo_info: vfo=VFOB
2023-01-16T20:06:32.477116-0500: fscanf: parsing '' with '%c'
2023-01-16T20:06:32.477116-0500: rigctl_parse: nothing to scan#1? retcode=-1, last_cmd=

````

closed 2023-09-17 by mdblack98