Hamlib/Hamlib #1220: get_powerstat

PianetaRadio, 2023-01-15:
Hamlib 4.5.4, Yaesu FT-991A with rigctld net connection, get_powerstat return RIG_POWER_ON when the radio is OFF

```
2023-01-15T18:24:09.690042+0100:  1:rig.c(1375):rig_open returning(0) 
2023-01-15T18:24:09.690042+0100:  1:rig.c(6040):rig_get_powerstat entered
2023-01-15T18:24:09.690042+0100: netrigctl_get_powerstat called
2023-01-15T18:24:09.690042+0100: netrigctl_transaction: called len=15
2023-01-15T18:24:09.690042+0100: network_flush called
2023-01-15T18:24:19.690321+0100: read_string_generic(): Timed out 10.000 seconds after 0 chars, direct=1
2023-01-15T18:24:19.690321+0100: netrigctl_get_powerstat: PS command failed (ret=-5) so returning RIG_POWER_ON
2023-01-15T18:24:19.690321+0100:  1:rig.c(6062):rig_get_powerstat returning(0) 
```

mdblack98, 2023-01-15:
That's because the rig cannot return PS status when it is turned off so we have to assume power is on.  A timeout does not necessarily mean power is off.  Some rigs can return power status when turned off.

Does this cause a problem with some software package?

PianetaRadio, 2023-01-16:
I think that assuming RIG_POWER_ON when there is a timeout of the get_powerstat command, it's not logic.
It generates an indetermination and will be not possible to understand if the rig is really on, off or there is a communication delay or error.

mdblack98, 2023-01-16:
Did you try the latest master?  It might be getting power status now when power is off.

PianetaRadio, 2023-01-17:
Sorry, not working with the latest snapshot [hamlib-4.6~git-20230117-cb1c98fa.tar.gz](https://n0nb.users.sourceforge.net/hamlib-4.6~git-20230117-cb1c98fa.tar.gz)

```
Rig command: \get_powerstat
rigctl_parse: input_line: \get_powerstat
rigctl_parse: vfo_opt=1
 1:rig.c(6053):rig_get_powerstat entered
rig.c(6073) trace
netrigctl_get_powerstat called
netrigctl_transaction: called len=15
rig_flush: called for network device
network_flush called
write_block(): TX 15 bytes, method=2
0000    5c 67 65 74 5f 70 6f 77 65 72 73 74 61 74 0a        \get_powerstat. 
read_string_generic called, rxmax=1024 direct=1, expected_len=1
read_string_generic(): Timed out 10.010 seconds after 0 chars, direct=1
netrigctl_get_powerstat: PS command failed (ret=-5) so returning RIG_POWER_ON
 1:rig.c(6075):rig_get_powerstat returning(0) 
 1:rigctl_parse.c(4745):rigctl_get_powerstat entered
  2:rig.c(6053):rig_get_powerstat entered
rig.c(6073) trace
netrigctl_get_powerstat called
netrigctl_transaction: called len=15
rig_flush: called for network device
network_flush called
write_block(): TX 15 bytes, method=2
0000    5c 67 65 74 5f 70 6f 77 65 72 73 74 61 74 0a        \get_powerstat. 
read_string_generic called, rxmax=1024 direct=1, expected_len=1
read_string_generic(): Timed out 10.010 seconds after 0 chars, direct=1
netrigctl_get_powerstat: PS command failed (ret=-5) so returning RIG_POWER_ON
  2:rig.c(6075):rig_get_powerstat returning(0) 
Power Status: 1
 1:rigctl_parse.c(4762):rigctl_get_powerstat returning(0) 
rigctl_parse: called, interactive=1
```

mdblack98, 2023-01-18:
Can you please use rigctl instead of rigctld and test it and show the log?

PianetaRadio, 2023-01-19:
Ok with rigctl

```
Rig command: \get_powerstat
rigctl_parse: input_line: \get_powerstat
rigctl_parse: vfo_opt=0
  2:rig.c(6054):rig_get_powerstat entered
rig.c(6074) trace
   3:newcat.c(3610):newcat_get_powerstat entered
newcat_valid_command PS
newcat.c(7655):newcat_valid_command returning2(1) 
write_block(): TX 3 bytes, method=2
0000    50 53 3b                                            PS;             
    4:newcat.c(10479):newcat_get_cmd entered
rig_flush: called for serial device
read_string_generic called, rxmax=4095 direct=1, expected_len=1
0000    00                                                  .               
cmd_str = PS;
write_block(): TX 3 bytes, method=2
0000    50 53 3b                                            PS;             
read_string_generic called, rxmax=129 direct=1, expected_len=1
read_string_generic(): RX 4 characters, direct=1
0000    50 53 30 3b                                         PS0;            
newcat_get_cmd: read count = 4, ret_data = PS0;
    4:newcat.c(10701):newcat_get_cmd returning(0) 
   3:newcat.c(3648):newcat_get_powerstat returning(0) 
  2:rig.c(6076):rig_get_powerstat returning(0) 
rigctl_parse: rig_powerstat is not on = 0
rigctl_parse: command get_powerstat not allowed when rig is powered off
rigctl_parse: called, interactive=1
```

mdblack98, 2023-01-19:
That's what I needed...try now....

Also add the -Z switch next time.  Need to see timing too.

Mike W9MDB

PianetaRadio, 2023-01-21:
Rigctl

```
Rig command: \get_powerstat
2023-01-21T19:23:32.945321+0100: rigctl_parse: input_line: \get_powerstat
2023-01-21T19:23:32.945529+0100: rigctl_parse: vfo_opt=0
2023-01-21T19:23:32.945682+0100:   2:rig.c(6054):rig_get_powerstat entered
2023-01-21T19:23:32.945752+0100: rig.c(6074) trace
2023-01-21T19:23:32.945817+0100:    3:newcat.c(3610):newcat_get_powerstat entered
2023-01-21T19:23:32.945881+0100: newcat_valid_command PS
2023-01-21T19:23:32.945950+0100: newcat.c(7655):newcat_valid_command returning2(1) 
2023-01-21T19:23:32.946069+0100: write_block(): TX 3 bytes, method=2
2023-01-21T19:23:32.946139+0100: 0000    50 53 3b                                            PS;             
2023-01-21T19:23:34.148497+0100:     4:newcat.c(10479):newcat_get_cmd entered
2023-01-21T19:23:34.148608+0100: rig_flush: called for serial device
2023-01-21T19:23:34.148683+0100: read_string_generic called, rxmax=4095 direct=1, expected_len=1
2023-01-21T19:23:34.149967+0100: 0000    00                                                  .               
2023-01-21T19:23:34.150038+0100: cmd_str = PS;
2023-01-21T19:23:34.150151+0100: write_block(): TX 3 bytes, method=2
2023-01-21T19:23:34.150219+0100: 0000    50 53 3b                                            PS;             
2023-01-21T19:23:34.152373+0100: read_string_generic called, rxmax=129 direct=1, expected_len=1
2023-01-21T19:23:34.152875+0100: read_string_generic(): RX 4 characters, direct=1
2023-01-21T19:23:34.152949+0100: 0000    50 53 30 3b                                         PS0;            
2023-01-21T19:23:34.153013+0100: newcat_get_cmd: read count = 4, ret_data = PS0;
2023-01-21T19:23:34.153082+0100:     4:newcat.c(10701):newcat_get_cmd returning(0) 
2023-01-21T19:23:34.153153+0100:    3:newcat.c(3648):newcat_get_powerstat returning(0) 
2023-01-21T19:23:34.153223+0100:   2:rig.c(6076):rig_get_powerstat returning(0) 
2023-01-21T19:23:34.153310+0100: rigctl_parse: rig_powerstat is not on = 0
2023-01-21T19:23:34.153393+0100:   2:rigctl_parse.c(4755):rigctl_get_powerstat entered
2023-01-21T19:23:34.153461+0100:    3:rig.c(6054):rig_get_powerstat entered
2023-01-21T19:23:34.153526+0100: rig.c(6074) trace
2023-01-21T19:23:34.153591+0100:     4:newcat.c(3610):newcat_get_powerstat entered
2023-01-21T19:23:34.153655+0100: newcat_valid_command PS
2023-01-21T19:23:34.153726+0100: newcat.c(7655):newcat_valid_command returning2(1) 
2023-01-21T19:23:34.153827+0100: write_block(): TX 3 bytes, method=2
2023-01-21T19:23:34.153894+0100: 0000    50 53 3b                                            PS;             
2023-01-21T19:23:35.356267+0100:      5:newcat.c(10479):newcat_get_cmd entered
2023-01-21T19:23:35.356380+0100: rig_flush: called for serial device
2023-01-21T19:23:35.356454+0100: read_string_generic called, rxmax=4095 direct=1, expected_len=1
2023-01-21T19:23:35.357763+0100: 0000    50 53 30 3b                                         PS0;            
2023-01-21T19:23:35.357841+0100: cmd_str = PS;
2023-01-21T19:23:35.357949+0100: write_block(): TX 3 bytes, method=2
2023-01-21T19:23:35.358014+0100: 0000    50 53 3b                                            PS;             
2023-01-21T19:23:35.360170+0100: read_string_generic called, rxmax=129 direct=1, expected_len=1
2023-01-21T19:23:35.360678+0100: read_string_generic(): RX 4 characters, direct=1
2023-01-21T19:23:35.360752+0100: 0000    50 53 30 3b                                         PS0;            
2023-01-21T19:23:35.360816+0100: newcat_get_cmd: read count = 4, ret_data = PS0;
2023-01-21T19:23:35.360886+0100:      5:newcat.c(10701):newcat_get_cmd returning(0) 
2023-01-21T19:23:35.360955+0100:     4:newcat.c(3648):newcat_get_powerstat returning(0) 
2023-01-21T19:23:35.361024+0100:    3:rig.c(6076):rig_get_powerstat returning(0) 
Power Status: 0
2023-01-21T19:23:35.361142+0100:   2:rigctl_parse.c(4772):rigctl_get_powerstat returning(0) 
2023-01-21T19:23:35.361223+0100: rigctl_parse: called, interactive=1
```

PianetaRadio, 2023-01-21:
Rigctld failed:

```
Rig command: \get_powerstat
2023-01-21T19:26:49.795820+0100: rigctl_parse: input_line: \get_powerstat
2023-01-21T19:26:49.796034+0100: rigctl_parse: vfo_opt=1
2023-01-21T19:26:49.796190+0100:  1:rig.c(6054):rig_get_powerstat entered
2023-01-21T19:26:49.796270+0100: rig.c(6074) trace
2023-01-21T19:26:49.796339+0100: netrigctl_get_powerstat called
2023-01-21T19:26:49.796406+0100: netrigctl_transaction: called len=15
2023-01-21T19:26:49.796471+0100: rig_flush: called for network device
2023-01-21T19:26:49.796544+0100: network_flush called
2023-01-21T19:26:49.796813+0100: write_block(): TX 15 bytes, method=2
2023-01-21T19:26:49.796900+0100: 0000    5c 67 65 74 5f 70 6f 77 65 72 73 74 61 74 0a        \get_powerstat. 
2023-01-21T19:26:49.796963+0100: read_string_generic called, rxmax=1024 direct=1, expected_len=1
2023-01-21T19:26:52.210872+0100: read_string_generic(): RX 2 characters, direct=1
2023-01-21T19:26:52.210986+0100: 0000    30 0a                                               0.              
2023-01-21T19:26:52.211060+0100: netrigctl_get_powerstat: PS command failed (ret=2) so returning RIG_POWER_ON
2023-01-21T19:26:52.211136+0100:  1:rig.c(6076):rig_get_powerstat returning(0) 
2023-01-21T19:26:52.211225+0100:  1:rigctl_parse.c(4755):rigctl_get_powerstat entered
2023-01-21T19:26:52.211297+0100:   2:rig.c(6054):rig_get_powerstat entered
2023-01-21T19:26:52.211374+0100: rig.c(6074) trace
2023-01-21T19:26:52.211443+0100: netrigctl_get_powerstat called
2023-01-21T19:26:52.211509+0100: netrigctl_transaction: called len=15
2023-01-21T19:26:52.211576+0100: rig_flush: called for network device
2023-01-21T19:26:52.211645+0100: network_flush called
2023-01-21T19:26:52.211851+0100: write_block(): TX 15 bytes, method=2
2023-01-21T19:26:52.211935+0100: 0000    5c 67 65 74 5f 70 6f 77 65 72 73 74 61 74 0a        \get_powerstat. 
2023-01-21T19:26:52.212003+0100: read_string_generic called, rxmax=1024 direct=1, expected_len=1
2023-01-21T19:26:58.636778+0100: read_string_generic(): RX 2 characters, direct=1
2023-01-21T19:26:58.636829+0100: 0000    30 0a                                               0.              
2023-01-21T19:26:58.636861+0100: netrigctl_get_powerstat: PS command failed (ret=2) so returning RIG_POWER_ON
2023-01-21T19:26:58.636892+0100:   2:rig.c(6076):rig_get_powerstat returning(0) 
Power Status: 1
2023-01-21T19:26:58.636948+0100:  1:rigctl_parse.c(4772):rigctl_get_powerstat returning(0) 
2023-01-21T19:26:58.636984+0100: rigctl_parse: called, interactive=1
```

PianetaRadio, 2023-01-22:
Yet wrong reply from rigctld with FT-991

```
Rig command: \get_powerstat
2023-01-22T20:01:08.120260+0100: rigctl_parse: input_line: \get_powerstat
2023-01-22T20:01:08.120477+0100: rigctl_parse: vfo_opt=1
2023-01-22T20:01:08.120571+0100:  1:rigctl_parse.c(4738):rigctl_get_powerstat entered
2023-01-22T20:01:08.120652+0100:   2:rig.c(6054):rig_get_powerstat entered
2023-01-22T20:01:08.120727+0100: rig.c(6074) trace
2023-01-22T20:01:08.120794+0100: netrigctl_get_powerstat called
2023-01-22T20:01:08.120862+0100: netrigctl_transaction: called len=15
2023-01-22T20:01:08.120927+0100: rig_flush: called for network device
2023-01-22T20:01:08.120998+0100: network_flush called
2023-01-22T20:01:08.121073+0100: network_flush: network data clear d: ret=0, len=8, ''
2023-01-22T20:01:08.121174+0100: network_flush: network data cleared: ret=0, len_read=8/0x8
2023-01-22T20:01:08.121253+0100: 0000    52 50 52 54 20 2d 35 0a                             RPRT -5.        
2023-01-22T20:01:08.121496+0100: write_block(): TX 15 bytes, method=2
2023-01-22T20:01:08.121584+0100: 0000    5c 67 65 74 5f 70 6f 77 65 72 73 74 61 74 0a        \get_powerstat. 
2023-01-22T20:01:08.121655+0100: read_string_generic called, rxmax=1024 direct=1, expected_len=1
2023-01-22T20:01:09.328649+0100: read_string_generic(): RX 2 characters, direct=1
2023-01-22T20:01:09.328760+0100: 0000    30 0a                                               0.              
2023-01-22T20:01:09.328837+0100: netrigctl_get_powerstat: PS command failed (ret=2) so returning RIG_POWER_ON
2023-01-22T20:01:09.328912+0100:   2:rig.c(6076):rig_get_powerstat returning(0) 
Power Status: 1
2023-01-22T20:01:09.329036+0100:  1:rigctl_parse.c(4755):rigctl_get_powerstat returning(0) 
2023-01-22T20:01:09.329120+0100: rigctl_parse: called, interactive=1

Rig command: 
```

mdblack98, 2023-01-22:
I need the rigctl debug side -- rigctld doesn't show what's happening with the rig.

Mike W9MDB

PianetaRadio, 2023-01-23:
Mike, rigctl works well reporting the right power state when off, see my previous post.
The problem is with rigctld.

mdblack98, 2023-01-23:
I mean I need to see the rigctl debug output when running it against rigctld.

PianetaRadio, 2023-01-24:
Ah, ok.
Sorry, what I already reported is the output of rigctl -m 2

> Yet wrong reply from rigctld with FT-991
>
> ```
> Rig command: \get_powerstat
> 2023-01-22T20:01:08.120260+0100: rigctl_parse: input_line: \get_powerstat
> 2023-01-22T20:01:08.120477+0100: rigctl_parse: [...]

closed 2023-04-10 by mdblack98

PianetaRadio, 2023-04-23:
Still problem with this issue, hamlib 4.5.5.

Radio (Yaesu FT-991A), power supply off then on, radio off:
```
rigctl -m 2 -vvvvv -Z --set-conf=auto_power_on=1
2023-04-23T20:40:50.842966+0100: rigctl.c(445) Startup: rigctl -m 2 -vvvvv -Z --set-conf=auto_power_on=1
2023-04-23T20:40:50.843235+0100: rigctl Hamlib 4.5.5 Apr 05 11:43:08Z 2023 SHA=6eecd3
2023-04-23T20:40:50.843293+0100: Report bugs to <hamlib-developer@lists.sourceforge.net>

2023-04-23T20:40:50.843367+0100: rig_check_rig_caps: p1=0xb6f1a158, p2=0xb6f1f8a4, rig_model=0xb6f1a158, macro_name=0xb6f1f8a4
2023-04-23T20:40:50.843558+0100: initrigs4_dummy: _init called
2023-04-23T20:40:50.843675+0100: rig_init: rig_model=Hamlib NET rigctl
2023-04-23T20:40:50.843831+0100: rig_init: rx_range_list1 is empty, using rx_range_list2
2023-04-23T20:40:50.843892+0100: rig_init: rig does not have tx_range!!
2023-04-23T20:40:50.844016+0100: netrigctl_init version 20230328.0
2023-04-23T20:40:50.844137+0100: rig_token_lookup called for auto_power_on
2023-04-23T20:40:50.844203+0100: rig_confparam_lookup called for auto_power_on
2023-04-23T20:40:50.844279+0100: rig_set_conf called
2023-04-23T20:40:50.844338+0100: rig_confparam_lookup called for 1073741948
2023-04-23T20:40:50.844398+0100: rig_set_conf: auto_power_on='1'
2023-04-23T20:40:50.844570+0100:  1:rig.c(817):rig_open entered
2023-04-23T20:40:50.844661+0100: rig_settings_get_path: path=/home/pi/.config//home/pi/hamlib_settings
2023-04-23T20:40:50.844848+0100: rig_settings_load_all: settings_file (/home/pi/.config//home/pi/hamlib_settings): No such file or directory
2023-04-23T20:40:50.844962+0100: rig_open: cwd=/home/pi
2023-04-23T20:40:50.845107+0100: rig_open: /home/pi/.config//home/pi/hamlib_settings does not exist
2023-04-23T20:40:50.845178+0100: rig_open: async_data_enable=1, async_data_supported=0
2023-04-23T20:40:50.845286+0100: rig_open: using network address 127.0.0.1:4532
2023-04-23T20:40:50.845370+0100: network_open: hoststr=127.0.0.1, portstr=4532
2023-04-23T20:40:50.845927+0100: network_open: client port=44165
2023-04-23T20:40:50.846008+0100:   2:rig.c(7481):async_data_handler_start entered
2023-04-23T20:40:50.846078+0100: async_data_handler_start: async data support disabled since async_data_enabled=0
2023-04-23T20:40:50.846181+0100:   2:rig.c(7488):async_data_handler_start returning(0) 
2023-04-23T20:40:50.846260+0100: rig.c(254):add_opened_rig returning2(0) 
2023-04-23T20:40:50.846409+0100: rig_open: 0x1ce1134 rs->comm_state==1?=1
2023-04-23T20:40:50.846545+0100:   2:rig.c(6084):rig_get_powerstat entered
2023-04-23T20:40:50.846668+0100: rig.c(6104) trace
2023-04-23T20:40:50.846788+0100: netrigctl_get_powerstat called
2023-04-23T20:40:50.846908+0100: netrigctl_transaction: called len=15
2023-04-23T20:40:50.847032+0100: rig_flush: called for network device
2023-04-23T20:40:50.847159+0100: network_flush called
2023-04-23T20:40:50.847478+0100: write_block(): TX 15 bytes, method=2
2023-04-23T20:40:50.847629+0100: 0000    5c 67 65 74 5f 70 6f 77 65 72 73 74 61 74 0a        \get_powerstat. 
2023-04-23T20:40:50.847758+0100: read_string_generic called, rxmax=1024 direct=1, expected_len=1
2023-04-23T20:41:00.857967+0100: read_string_generic(): Timed out 10.010 seconds after 0 chars, direct=1
2023-04-23T20:41:00.858072+0100: netrigctl_get_powerstat: PS command failed (ret=-5) so returning RIG_POWER_ON
2023-04-23T20:41:00.858144+0100:   2:rig.c(6109):rig_get_powerstat returning(0) 
2023-04-23T20:41:00.858214+0100:   2:netrigctl.c(261):netrigctl_open entered
2023-04-23T20:41:00.858278+0100: netrigctl_transaction: called len=9
2023-04-23T20:41:00.858340+0100: rig_flush: called for network device
2023-04-23T20:41:00.858406+0100: network_flush called
2023-04-23T20:41:00.858612+0100: write_block(): TX 9 bytes, method=2
2023-04-23T20:41:00.858685+0100: 0000    5c 63 68 6b 5f 76 66 6f 0a                          \chk_vfo.       
2023-04-23T20:41:00.858747+0100: read_string_generic called, rxmax=1024 direct=1, expected_len=1
2023-04-23T20:41:10.868889+0100: read_string_generic(): Timed out 10.010 seconds after 0 chars, direct=1
2023-04-23T20:41:10.869090+0100: netrigctl_open: chk_vfo error: netrigctl_get_powerstat called
netrigctl_transaction: called len=15
rig_flush: called for network device
network_flush called
write_block(): TX 15 bytes, method=2
0000    5c 67 65 74 5f 70 6f 77 65 72 73 74 61 74 0a        \get_powerstat. 
read_string_generic called, rxmax=1024 direct=1, expected_len=1
read_string_generic(): Timed out 10.010 seconds after 0 chars, direct=1
netrigctl_get_powerstat: PS command failed (ret=-5) so returning RIG_POWER_ON
  2:rig.c(6109):rig_get_powerstat returning(0) 
  2:netrigctl.c(261):netrigctl_open entered
netrigctl_transaction: called len=9
rig_flush: called for network device
network_flush called
write_block(): TX 9 bytes, method=2
0000    5c 63 68 6b 5f 76 66 6f 0a                          \chk_vfo.       
[... 2 lines cut ...]
Communication timed out
Communication timed out

2023-04-23T20:41:10.869411+0100: netrigctl_open: vfo_mode=0
2023-04-23T20:41:10.869533+0100: netrigctl_transaction: called len=12
2023-04-23T20:41:10.869601+0100: rig_flush: called for network device
2023-04-23T20:41:10.869673+0100: network_flush called
2023-04-23T20:41:10.869916+0100: write_block(): TX 12 bytes, method=2
2023-04-23T20:41:10.869999+0100: 0000    5c 64 75 6d 70 5f 73 74 61 74 65 0a                 \dump_state.    
[... cut ...]
2023-04-23T20:41:20.876836+0100: network_close: close socket ret=0
2023-04-23T20:41:20.876917+0100:  1:rig.c(1301):rig_open returning(-5) Communication timed out

rig_open: error = (null) No such file or directory 
```

Remarks:
1) Using rigctl with serial connection no problem, it reports power status off and I can turn on the radio on with set_powerstat.
2) Then if I connect trough NET rig, it works.
3) If I power off the radio, exit from rigctl and run again the previous command, now it works well reporting radio power status off.

PianetaRadio, 2023-04-27:
The same without option auto_power_on

```
rigctl -m 2 -vvvvv -Z
2023-04-27T20:32:34.269000+0100: rigctl.c(445) Startup: rigctl -m 2 -vvvvv -Z
2023-04-27T20:32:34.269268+0100: rigctl Hamlib 4.5.5 Apr 05 11:43:08Z 2023 SHA=6eecd3
2023-04-27T20:32:34.269363+0100: Report bugs to <hamlib-developer@lists.sourceforge.net>

2023-04-27T20:32:34.269445+0100: rig_check_rig_caps: p1=0xb6eed158, p2=0xb6ef28a4, rig_model=0xb6eed158, macro_name=0xb6ef28a4
2023-04-27T20:32:34.269637+0100: initrigs4_dummy: _init called
2023-04-27T20:32:34.269754+0100: rig_init: rig_model=Hamlib NET rigctl
2023-04-27T20:32:34.269907+0100: rig_init: rx_range_list1 is empty, using rx_range_list2
2023-04-27T20:32:34.269967+0100: rig_init: rig does not have tx_range!!
2023-04-27T20:32:34.270032+0100: netrigctl_init version 20230328.0
2023-04-27T20:32:34.270147+0100:  1:rig.c(817):rig_open entered
2023-04-27T20:32:34.270283+0100: rig_settings_get_path: path=/home/pi/.config//home/pi/hamlib_settings
2023-04-27T20:32:34.270473+0100: rig_settings_load_all: settings_file (/home/pi/.config//home/pi/hamlib_settings): No such file or directory
2023-04-27T20:32:34.270566+0100: rig_open: cwd=/home/pi
2023-04-27T20:32:34.270703+0100: rig_open: /home/pi/.config//home/pi/hamlib_settings does not exist
2023-04-27T20:32:34.270770+0100: rig_open: async_data_enable=1, async_data_supported=0
2023-04-27T20:32:34.270873+0100: rig_open: using network address 127.0.0.1:4532
2023-04-27T20:32:34.270949+0100: network_open: hoststr=127.0.0.1, portstr=4532
2023-04-27T20:32:34.271471+0100: network_open: client port=41605
2023-04-27T20:32:34.271549+0100:   2:rig.c(7481):async_data_handler_start entered
2023-04-27T20:32:34.271612+0100: async_data_handler_start: async data support disabled since async_data_enabled=0
2023-04-27T20:32:34.271724+0100:   2:rig.c(7488):async_data_handler_start returning(0) 
2023-04-27T20:32:34.271799+0100: rig.c(254):add_opened_rig returning2(0) 
2023-04-27T20:32:34.271864+0100: rig_open: 0x10eb134 rs->comm_state==1?=1
2023-04-27T20:32:34.271932+0100:   2:rig.c(6084):rig_get_powerstat entered
2023-04-27T20:32:34.271995+0100: rig.c(6104) trace
2023-04-27T20:32:34.272056+0100: netrigctl_get_powerstat called
2023-04-27T20:32:34.272117+0100: netrigctl_transaction: called len=15
2023-04-27T20:32:34.272181+0100: rig_flush: called for network device
2023-04-27T20:32:34.272248+0100: network_flush called
2023-04-27T20:32:34.272464+0100: write_block(): TX 15 bytes, method=2
2023-04-27T20:32:34.272547+0100: 0000    5c 67 65 74 5f 70 6f 77 65 72 73 74 61 74 0a        \get_powerstat. 
2023-04-27T20:32:34.272614+0100: read_string_generic called, rxmax=1024 direct=1, expected_len=1
2023-04-27T20:32:44.282775+0100: read_string_generic(): Timed out 10.010 seconds after 0 chars, direct=1
2023-04-27T20:32:44.282886+0100: netrigctl_get_powerstat: PS command failed (ret=-5) so returning RIG_POWER_ON
2023-04-27T20:32:44.282967+0100:   2:rig.c(6109):rig_get_powerstat returning(0) 
2023-04-27T20:32:44.283044+0100:   2:netrigctl.c(261):netrigctl_open entered
2023-04-27T20:32:44.283115+0100: netrigctl_transaction: called len=9
2023-04-27T20:32:44.283181+0100: rig_flush: called for network device
2023-04-27T20:32:44.283251+0100: network_flush called
2023-04-27T20:32:44.283487+0100: write_block(): TX 9 bytes, method=2
2023-04-27T20:32:44.283565+0100: 0000    5c 63 68 6b 5f 76 66 6f 0a                          \chk_vfo.       
2023-04-27T20:32:44.283630+0100: read_string_generic called, rxmax=1024 direct=1, expected_len=1
2023-04-27T20:32:54.293739+0100: read_string_generic(): Timed out 10.010 seconds after 0 chars, direct=1
2023-04-27T20:32:54.293831+0100: netrigctl_open: chk_vfo error: netrigctl_get_powerstat called
netrigctl_transaction: called len=15
rig_flush: called for network device
network_flush called
write_block(): TX 15 bytes, method=2
0000    5c 67 65 74 5f 70 6f 77 65 72 73 74 61 74 0a        \get_powerstat. 
read_string_generic called, rxmax=1024 direct=1, expected_len=1
read_string_generic(): Timed out 10.010 seconds after 0 chars, direct=1
netrigctl_get_powerstat: PS command failed (ret=-5) so returning RIG_POWER_ON
  2:rig.c(6109):rig_get_powerstat returning(0) 
  2:netrigctl.c(261):netrigctl_open entered
netrigctl_transaction: called len=9
rig_flush: called for network device
network_flush called
write_block(): TX 9 bytes, method=2
0000    5c 63 68 6b 5f 76 66 6f 0a                          \chk_vfo.       
read_string_generic called, rxmax=1024 direct=1, expected_len=1
read_string_generic(): Timed out 10.010 seconds after 0 chars, direct=1
Communication timed out
Communication timed out

2023-04-27T20:32:54.293980+0100: netrigctl_open: vfo_mode=0
2023-04-27T20:32:54.294031+0100: netrigctl_transaction: called len=12
2023-04-27T20:32:54.294058+0100: rig_flush: called for network device
2023-04-27T20:32:54.294088+0100: network_flush called
2023-04-27T20:32:54.294194+0100: write_block(): TX 12 bytes, method=2
2023-04-27T20:32:54.294227+0100: 0000    5c 64 75 6d 70 5f 73 74 61 74 65 0a                 \dump_state.    
2023-04-27T20:32:54.294254+0100: read_string_generic called, rxmax=1024 direct=1, expected_len=1
2023-04-27T20:33:04.296635+0100: read_string_generic(): Timed out 10.002 seconds after 0 chars, direct=1
2023-04-27T20:33:04.296811+0100:   2:netrigctl.c(299):netrigctl_open returning(-5) Communication timed out

2023-04-27T20:33:04.296898+0100:   2:rig.c(7525):async_data_handler_stop entered
2023-04-27T20:33:04.296972+0100:   2:rig.c(7555):async_data_handler_stop returning(0) 
2023-04-27T20:33:04.297227+0100: network_close: close socket ret=0
2023-04-27T20:33:04.297309+0100:  1:rig.c(1301):rig_open returning(-5) Communication timed out

rig_open: error = (null) No such file or directory 

```

PianetaRadio, 2023-04-27:
Please reopen the issue

PianetaRadio, 2023-04-27:
Instead rigctl with serial works as expected:

```
rigctl -m 1035 -r /dev/ttyUSB1 -s 38400 -vvvvv -Z
2023-04-27T20:51:27.484503+0100: rigctl.c(445) Startup: rigctl -m 1035 -r /dev/ttyUSB1 -s 38400 -vvvvv -Z
2023-04-27T20:51:27.484966+0100: rigctl Hamlib 4.5.5 Apr 05 11:43:08Z 2023 SHA=6eecd3
2023-04-27T20:51:27.485098+0100: Report bugs to <hamlib-developer@lists.sourceforge.net>

2023-04-27T20:51:27.485236+0100: rig_check_rig_caps: p1=0xb6ec0158, p2=0xb6ec58a4, rig_model=0xb6ec0158, macro_name=0xb6ec58a4
2023-04-27T20:51:27.485436+0100: yaesu: initrigs4_yaesu called
2023-04-27T20:51:27.485767+0100: rig_init: rig_model=Yaesu FT-991
2023-04-27T20:51:27.485925+0100: rig_init: rig has VFO_A
2023-04-27T20:51:27.485982+0100: rig_init: rig has VFO_B
2023-04-27T20:51:27.486075+0100: rig_init: rig has VFO_MEM
2023-04-27T20:51:27.486262+0100: ft991_init called, version 20230328.15
2023-04-27T20:51:27.486451+0100:  1:newcat.c(469):newcat_init entered
2023-04-27T20:51:27.486618+0100:  1:newcat.c(510):newcat_init returning(0) 
2023-04-27T20:51:27.486855+0100:  1:rig.c(817):rig_open entered
2023-04-27T20:51:27.487009+0100: rig_settings_get_path: path=/home/pi/.config//home/pi/hamlib_settings
2023-04-27T20:51:27.487286+0100: rig_settings_load_all: settings_file (/home/pi/.config//home/pi/hamlib_settings): No such file or directory
2023-04-27T20:51:27.487447+0100: rig_open: cwd=/home/pi
2023-04-27T20:51:27.487670+0100: rig_open: /home/pi/.config//home/pi/hamlib_settings does not exist
2023-04-27T20:51:27.487805+0100: rig_open: async_data_enable=1, async_data_supported=0
2023-04-27T20:51:27.487951+0100: serial_open: /dev/ttyUSB1
2023-04-27T20:51:27.493382+0100: serial_setup: tcgetattr
2023-04-27T20:51:27.493581+0100: serial_setup: cfmakeraw
2023-04-27T20:51:27.493725+0100: serial_setup: cfsetispeed=38400,0x000f
2023-04-27T20:51:27.493861+0100: serial_setup: cfsetospeed=38400,0x000f
2023-04-27T20:51:27.493994+0100: serial_setup: data_bits=8
2023-04-27T20:51:27.494126+0100: serial_setup: parity=0
2023-04-27T20:51:27.494258+0100: serial_setup: Handshake=Hardware
2023-04-27T20:51:27.494405+0100: serial_setup: tcsetattr TCSANOW
2023-04-27T20:51:27.494609+0100: read_string_generic called, rxmax=4095 direct=1, expected_len=1
2023-04-27T20:51:27.495960+0100: tcflush
2023-04-27T20:51:27.546248+0100:   2:rig.c(7481):async_data_handler_start entered
2023-04-27T20:51:27.546341+0100: async_data_handler_start: async data support disabled since async_data_enabled=0
2023-04-27T20:51:27.546384+0100:   2:rig.c(7488):async_data_handler_start returning(0) 
2023-04-27T20:51:27.546427+0100: rig.c(254):add_opened_rig returning2(0) 
2023-04-27T20:51:27.546468+0100: rig_open: 0x1dda134 rs->comm_state==1?=1
2023-04-27T20:51:27.546519+0100:   2:rig.c(6084):rig_get_powerstat entered
2023-04-27T20:51:27.546559+0100: rig.c(6104) trace
2023-04-27T20:51:27.546598+0100:    3:newcat.c(3625):newcat_get_powerstat entered
2023-04-27T20:51:27.546732+0100: newcat_valid_command PS
2023-04-27T20:51:27.546780+0100: newcat.c(7665):newcat_valid_command returning2(1) 
2023-04-27T20:51:27.546826+0100:     4:newcat.c(10489):newcat_get_cmd entered
2023-04-27T20:51:27.546876+0100: rig_flush: called for serial device
2023-04-27T20:51:27.546917+0100: read_string_generic called, rxmax=4095 direct=1, expected_len=1
2023-04-27T20:51:27.548049+0100: tcflush
2023-04-27T20:51:27.548104+0100: cmd_str = PS;
2023-04-27T20:51:27.548180+0100: write_block(): TX 3 bytes, method=2
2023-04-27T20:51:27.548220+0100: 0000    50 53 3b                                            PS;             
2023-04-27T20:51:27.550335+0100: read_string_generic called, rxmax=129 direct=1, expected_len=1
2023-04-27T20:51:27.550829+0100: read_string_generic(): RX 4 characters, direct=1
2023-04-27T20:51:27.550875+0100: 0000    50 53 30 3b                                         PS0;            
2023-04-27T20:51:27.550912+0100: newcat_get_cmd: read count = 4, ret_data = PS0;
2023-04-27T20:51:27.550951+0100:     4:newcat.c(10711):newcat_get_cmd returning(0) 
2023-04-27T20:51:27.550991+0100:    3:newcat.c(3658):newcat_get_powerstat returning(0) 
2023-04-27T20:51:27.551032+0100:   2:rig.c(6109):rig_get_powerstat returning(0) 
2023-04-27T20:51:27.551069+0100: rig_open: rig power is off, use --set-conf=auto_power_on if power on is wanted
rig_open: error = /dev/ttyUSB1 No such file or directory 

```