Hamlib/Hamlib #1221: IC-7300 stops tracking the frequency after some time

ftl, 2023-01-18:
I see the following behaviour on the IC-7300 with version 4.6-git (aa1cf567): after some time I cannot get the current frequency using get_freq anymore. Instead I get the value that is in the cache. The cached value is not updated, so I keep getting the same value, no matter what frequency is selected on the radio.

I tried to disable the cache using `-C cache_timeout=0`, but in the trace I see that `use_cached_freq` changes from 0 to 1 at some point.

I found no way to reproduce the problem immediately, but I found out that changing the band or doing fast frequency changes makes the problem occur sooner.

This issue may be related to #1108.

[dump_frequency_tracking.txt](https://github.com/Hamlib/Hamlib/files/10449683/dump_frequency_tracking.txt)

ftl, 2023-01-22:
Looks like I need to turn of "CI-V Transceive" on the IC-7300 to fix this behavior.

mdblack98, 2023-01-22:
Reopening this -- we should be able to run with transceive on.

ftl, 2023-01-22:
Ok. I can give a litte bit more context to help find out what the problem could be.

1. I found the behaviour originally using `rigctld -m 3073 -r /dev/ttyUSB0 -s 115200 -c 148`
2. To disable the cache I added `-C cache_timeout=0` to the command in 1.
3. I also tried rigctl instead of rigctld and interactively "polled" the frequency just to find out if it has something to do with rigctld. I was also able to reproduce the problem in this setup, so it is nothing specific to rigctld.

I hope this helps.

mdblack98, 2023-07-23:
Turn this off for Icom rigs.  Need some testing for async=1 on Kenwood/Yaesu rigs before closing this.

closed 2023-07-31 by mdblack98