Hamlib/Hamlib #1223: Hamlib control program

mdblack98, 2023-01-21:
Create a portable GUI control program for Hamlib to control most functions.  Depends on multicast capability being created.
In particular...
Freq
Mode
PTT
AutoTune
RF Power
AF Level
Front/Rear select for USB and USBD
and others....

N0NB, 2023-01-22:
I suspect you have a language and toolkit in mind.

mdblack98, 2023-01-22:
C# and GTK at the moment.  Can use "dotnet build" on Windows, Linux, and MAC.

N0NB, 2023-01-22:
Interesting.  I doubt you'll get much help from the Linux side and I was unaware that there is any way to build a C# app for Linux or that there is any .NET support for Linux.  GTK seems to be poorly supported on Windows, at least that is what I read elsewhere and, IMO, the choices made for GTK3 will be foreign on Windows.  As I see it, GTK4 is hopeless for any software outside of GNOME.

I don't have a recommendation.  I've dabbled with other GUI projects over the years and I generally like the look and feel of Qt apps, but it is a huge beast and depends on C++ (mostly) which to me is completely beyond the realm of hobbyist friendly.

It's getting to the point that maybe only an AI can write a GUI program!

mdblack98, 2023-01-22:
On Windows
dotnet new install GtkSharp.Template.CSharp
dotnet new gtkapp
dotnet build

On Ubuntu I don't see the Gtk package -- but you can build the code generated on the Windows side.
apt install dotnet-sdk-6.0
dotnet build

PianetaRadio, 2023-01-22:
I'm with the same opinion of Nate, QT is a very good development platform.
See my GUI https://github.com/PianetaRadio/CatRadio

markjfine, 2023-01-22:
Qt is really the best option if you're looking cross-platform, however, just be aware that everything falls apart with every major new version making it hard to maintain. Things quickly get deprecated and the GUI practically needs a major rewrite every time, such as recently when it went from 5.15 to 6.0. Have seen this very thing on DReaM over the years.

mdblack98, 2023-01-22:
Qt is way too C++ dependent and I don't want to add that in.
Plus their recent breaking of the API has really turned me off to Qt.
It's not completely out of the question though depending on the roadblocks I run into with portability and capability.

Mike

N0NB, 2023-01-22:
I was completely unaware of there being Dot NET stuff in the current Debian Stable repository but there is:  https://packages.debian.org/bullseye/libgtk3.0-cil

This should also be in Ubuntu flavors as well.

N0NB, 2023-01-22:
There is also: https://packages.debian.org/bullseye/libgtk-dotnet3.0-cil

which might be more of what you're looking for, Mike.  I'm not sure as this is entirely new territory for me.

N0NB, 2023-01-22:
Those two packages may be the only ones in Debian proper as I find nothing for the dotnet-sdk package in the Debian Unstable repository.  There is this link from Microsoft: https://learn.microsoft.com/en-us/dotnet/core/install/linux-debian

mdblack98, 2023-01-22:
Question is can you install dotnet on your build system?

I've got some info in the README.txt that needs to be fleshed out.

It will do a cross build for Windows

dotnet publish -c release -r win7-x64

And for different OS's.

https://www.hanselman.com/blog/selfcontained-net-core-applications

Mike

N0NB, 2023-01-22:
* On 2023 22 Jan 13:16 -0600, Michael Black wrote:
> Question is can you install dotnet on your build system?

I'll admit to being queasy about installing Microsoft packages!

73, Nate

-- 
"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."
Web: https://www.n0nb.us
Projects: https://github.com/N0NB
GPG fingerprint: 82D6 4F6B 0E67 CD41 F689 BBA6 FB2C 5130 D55A 8819

N0NB, 2023-01-22:
> Qt is really the best option if you're looking cross-platform, however, just be aware that everything falls apart with every major new version making it hard to maintain. Things quickly get [...]

I guess they have their reasons for making changes so rapidly.  I suppose I'm getting to the point in life where I value stability.  In that vein there is always Tk.

;)

markjfine, 2023-01-22:
> Qt is way too C++ dependent and I don't want to add that in. Plus their recent breaking of the API has really turned me off to Qt. It's not completely out of the question though depending on the roadblocks I run into with portability and capability.

Yup. I will kinda agree. And the API breaking, as I intimated, is a semi-regular thing. The only thing that could be worse wrt constant deprecation is Java - learned that many years ago, developing an ADRG map server / application way before ESRI was even a thing.

N0NB, 2023-01-22:
Aside from the things already discussed, after thinking about this for a while, I think that anything that depends on the proprietary .NET SDK should be in its own repository and not a part of Hamlib proper.  If the bindings for C# do not rely on the SDK then they should remain in the Hamlib tree.

mdblack98, 2023-09-17:
See https://github.com/Hamlib/Hamlib/issues/695

closed 2023-09-17 by mdblack98