Hamlib/Hamlib #1224: K3 does not set VFOB correctly

mdblack98, 2023-01-22:
K3 does not set VFOB mode/width independently from VFOA

N0NB, 2023-01-22:
* On 2023 22 Jan 12:57 -0600, Michael Black wrote:
> K3 does not set VFOB mode/width independently from VFOA

I installed WSJT-X on Thursday and started using it with my K3 on 10m
and it seemed to work okay.  I went down to 160 and 8-m and I saw the
VFO B on the P3 was violet instead of red (red indicates split).  Any
time WSJT-X tried to set VFO B the radio showed that split was not
available.  In my case VFO B was still in LSB and was not switched to
DATA.

73, Nate

-- 
"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."
Web: https://www.n0nb.us
Projects: https://github.com/N0NB
GPG fingerprint: 82D6 4F6B 0E67 CD41 F689 BBA6 FB2C 5130 D55A 8819

N0NB, 2023-01-25:
I realized I still had old libs laying around since late '20/early '21.  Oops!  Oh. well, WSJT-X doesn't use those anyway so even though I've compiled and installed the latest Git master into /usr/local/lib, is there a way to get WSJT-X to use it?  Or, do I have to tell it to us `rigctld`?

mdblack98, 2023-01-25:
What version of WSJT-X?  The VFOB set fix in WSJT-X was Aug 2022.

If you do "ldd wsjtx" it will show you what Hamlib is being referenced.

Mike W9MDB

N0NB, 2023-01-25:
* On 2023 25 Jan 07:19 -0600, Michael Black wrote:
> What version of WSJT-X?  The VFOB set fix in WSJT-X was Aug 2022.

2.6.1

> If you do "ldd wsjtx" it will show you what Hamlib is being referenced.

libhamlib is not listed.  This version was installed from the WSJT-X Web
site, not a Debian packaged version nor compiled locally.

73, Nate

-- 
"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."
Web: https://www.n0nb.us
Projects: https://github.com/N0NB
GPG fingerprint: 82D6 4F6B 0E67 CD41 F689 BBA6 FB2C 5130 D55A 8819

mdblack98, 2023-01-25:
Which package did you install?  If libhamlib is not shown that means it's a static build which surprises me.

Mike W9MDB

> What version of WSJT-X?  The VFOB set fix in WSJT-X was Aug 2022.

2.6.1

> If you do "ldd wsjtx" it will show you what Hamlib is being referenced.

libhamlib is not listed. This version was installed from the WSJT-X Web
site, not a Debian packaged version nor compiled locally.

73, Nate

mdblack98, 2023-01-25:
Hamlib version should also be here....but 2.6.1 won't have the patch I just put in to fix the K3 VFOB set_mode.
.local/share/WSJT-X/wsjtx_syslog.log 

That was this patch
commit 9626894029686803544cae7860f34b362dd7a19b
Author: Mike Black W9MDB ***@***.***>
Date:   Sun Jan 22 16:45:22 2023 -0600

Mike W9MDB

> What version of WSJT-X?  The VFOB set fix in WSJT-X was Aug 2022.

2.6.1

> If you do "ldd wsjtx" it will show you what Hamlib is being referenced.

libhamlib is not listed. This version was installed from the WSJT-X Web
site, not a Debian packaged version nor compiled locally.

73, Nate

N0NB, 2023-01-25:
This one:  https://sourceforge.net/projects/wsjt/files/wsjtx-2.6.1/wsjtx_2.6.1_amd64.deb

Well,
```
$ ldd $(which wsjtx) | grep hamlib
```
returns nothing, so....

The prebuilt packages have always had Hamlib statically linked as I recall.  I guess I need to do the build myself in order to use shared libhamlib.

With the improvements you've made to `rigctld`, I've not tried to run two programs concurrently such as `wsjtx` and `cqrlog`.