Hamlib/Hamlib #1229: IC-9700 & WSJT-X V2.6.1 hamlib error

K0EKL, 2023-02-02:
I'm trying to set up WSJT-X for my IC-9700. The radio is running firmware V1.32.

Other apps can communicate with the 9700 on COM 10 at any speed I set it to from 4800 baud to 115200 but WSJT-X times out and throws a Hamlib when I try to test CAT at any baud rate. I'm uploading a screenshot of the error. 

[image]

mdblack98, 2023-02-02:
Try the latest DLL here....
https://n0nb.users.sourceforge.net/

If you still see an error please place this file as described below
https://www.dropbox.com/s/t52ngcalsgnpm8m/wsjtx_log_config.ini?dl=0

    C:\Users\[username]\AppData\Local\WSJT-X
    The WSJT-X_Rigcontrol.log file will be in the same location

For Linux put it in  
    ~/.config
    The WSJT-X_Rigcontrol.log file will be here:
    ~/.local/share/WSJT-X

For MacOS put it in
    /Users/[username]/Library/Preferences
 
Restart WSJT-X and duplicate the problem.
Shut down WSJT-X

Then send me the WSJT-X_RigControl.log file
Mike W9MDB

K0EKL, 2023-02-02:
Thanks for your advice. I replaced the libhamlid.dll file with the file you linked to. It made no difference and WSJT-X threw the same Hamlib error. Then I dropped your WSJT-X_RigControl.ini file in C:\Users\KØEKL\AppData\Local\WSJT-X and tried again. Same error message. 

Attached is the wsjt-x_RigControl.log file you requested. 
[WSJT-X_RigControl.log](https://github.com/Hamlib/Hamlib/files/10570500/WSJT-X_RigControl.log)
[image]

K0EKL, 2023-02-02:
I did the test again, this time just COM10. The results are below.

Alos of note, I installed WFView, the freeware Icom rig control package and it found my IC-9700 on COM 10 @ 57,600 baud and worked perfectly. I did this as a test to prove or disprove that CAT was possible. 
[image]
[WSJT-X_RigControl.log](https://github.com/Hamlib/Hamlib/files/10570768/WSJT-X_RigControl.log)

mdblack98, 2023-02-02:
Can we connect up?  

217-960-0233

Mike W9MDB

mdblack98, 2023-02-03:
Wrong CI-V address

K0EKL, 2023-02-04:
With help from Mike, W9MDB, my IC-9700 problem has been resolved. I had changed the radio's CI-V address to A3. WSJT-X utilizes Hamlib for it's cat control middleware and Hamlib expects the 9700 to be at the default address of A2.  Hamlib can be configured to use a different address, but the simplest solution was to switch the radio back to A2.

closed 2023-02-04 by mdblack98