Hamlib/Hamlib #1233: IC-9700 in Sat Mode for QO-100

hb9gfx, 2023-02-09:
Description:

Settings:
I use the IC-9700 in SAT Mode to drive my QO-100 Groundstation in FT-4/FT-8
ZF is 144.540 on RX and 432.540 on TX
I added a Shift of -10.345.000 000 Mhz to have the 3cm in the Menu of JTDX
At the IC-9700 I added a SAT Memory  FT-8 QO-100 with the frequencies 144.540 on RX and 432.540 on TX
[image]

Issue:
While starting JTDX  for the IC-9700 in many cases the TRX is switching strange frequencies and going to 144.540 in 2m band. because the Frequency 3cm is not supported.

Please Add support for SAT Mode for QO-100 at IC-9700

mdblack98, 2023-02-09:
Did you try adding the frequency in the Settings/Frequencies tab?

Mike W9MDB

hb9gfx, 2023-02-09:
It works from time to time. But always starting the JTDX where hamlib is integrated it goes to 144.174Mhz. 

 

I think this is because from the IC-9700 it cannot do 10.489.540 Mhz. 

 

It overwrites the settings last saved by quitting the JTDX.

 

Next is in some cases it switches between Main and sub band without any reason on the radio. 

 

Also it would be great to set the “sub frequency” But this is also not supported by JTDX.  

 

73 de Stefan HB9GFX

mdblack98, 2023-02-09:
What are your target frequencies for the QO-100?

What satellite is this?

I'm trying to understand what's required here.

Mike W9MDB

hb9gfx, 2023-02-09:
Hi Mike,

 

for RX Target is 10.489.540.000 Mhz driven with ZF Fequency of 144.540Mhz from IC-9700 

for TX Target is 2.400.540.000 Mhz driven from ZF Frequency of 432.540Mhz from IC-9700

 

I am using a Transverter from DXPatrol. https://dxpatrol.pt/produto/dxpatrol-full-duplex-qo-100-groundstation/

 

QO -100 is a geostationary satellite which covers Europe, Asia, Africa and the far East cost of parts of South America

https://amsat-dl.org/en/eshail-2-2/

 

You need a standard TV LNB with a modification to add a 10Mhz GPS stabilized reference to the LNB for receiving

You need a 3,5 winds helix antenna to do transmission

All this is mounted to a 85cm offset dish 

 

You just need a view watts to get a good signal to the satellite.

 

You can see some pictures here of my station: https://xlx508-hb9gfx-ch.translate.goog/index.php?show=QO-100 <https://xlx508-hb9gfx-ch.translate.goog/index.php?show=QO-100&_x_tr_sl=de&_x_tr_tl=en&_x_tr_hl=de> &_x_tr_sl=de&_x_tr_tl=en&_x_tr_hl=de

 

Already translated with google to English if you klick the link.

 

73 de Stefan HB9GFX

mdblack98, 2024-11-04:
Could you please test the latest Hamlib and see if things are better now?  I did some fixes on the 9700 VFO capability to make it behave better.

hb9gfx, 2024-11-05:
Hi Michael,

 

I have sold my IC-9700 in the meantime and returned to IC-9100, which is the better radio 😊

 

73 de Stefan

closed 2024-11-05 by mdblack98