Hamlib/Hamlib #1249: Yaesu GS-232 and G-5500DC

PeteK6TJ, 2023-03-16:
The oldest GS-232 computer interface has no calibration capability, and expects N center scale and 360 degrees rotation just like the G-5400.  But the G-5500DC has N at zero and 450 degrees rotation.  In gs232a.c I did the forward and backward transforms and it works now.  The constant might well be 1.25 but mine measured as shown.  It's device 606 now.
gs232a_rot_set_position, line 172
/*  Calibration of G5500 PM 230313 */
    az = az / 1.255525;
    az = az + 180;
    if (az > 360) az = az - 360;
gs232a_rot_get_position, about line 219
/*   "calibration" for G5500DC PM 230313 */
    if (*az < 180.0) *az =  *az + 360.0;
    *az = *az - 180.0;
    *az = *az * 1.25525;

mdblack98, 2023-03-16:
This sounds like we need an azimuth correction factor we can set.  That would make it generic for any controller.

So in addition to these:
-o, --set-azoffset=VAL       set offset for azimuth
-O, --set-eloffset=VAL       set offset for elevation

We could have
-p, --set-azcorrection==VAL       set correction for azimuth
-P, --set-elcorrection==VAL       set correction for elevation

Sound OK?

Mike W9MDB

PeteK6TJ, 2023-03-16:
The G-5500DC can be at an azimuth in the range of {0-90} degrees in two 
ways.  One is near a reported count of zero and the other is nearly full 
scale.  The reason is so that it's unlikely every satellite pass will be 
interrupted with a need to rotate 360 due to end of Az travel.  I don't 
understand how tracking programs will handle this, or even if they do 
(I'm using gpredict).  So I'm not sure if the "360" number in the 
transforms needs to be settable too.  Just haven't thought it all the 
way through.

PeteK6TJ, 2023-03-24:
I think the 360 is related to degrees in a circle, not the limit of the rotator.  That is held elsewhere, and I don't have software that would drive it there at the moment.  So @mdblack98 just the ones you noted will get those who have the same hardware configuration back on the air.