Hamlib/Hamlib #1257: Rotctl model RT-21 does not support moving

NUCARADO, 2023-03-25:
Cannot operate Green Heron RT-21 through rotctl. Both models 403 and 405 respond to rot_move with "Feature not available".

NUCARADO, 2023-03-25:
There appears to be no rot_move or rot_park methods in rotators/rotorez/rotorez.c. "dumpcaps" reports:
Can Stop:               Y
Can Park:               N
Can Reset:              N
Can Move:               N

mdblack98, 2023-03-25:
The RT-21 does support "move".  It only supports set_pos.  That should do what you need.
Mike W9MDB

mdblack98, 2023-03-25:
Because the RT-21 doesn't have those functions.

NUCARADO, 2023-03-25:
Sorry, I incorrectly assumed "set position" changed an encoder offset.

The set position command moves azimuth but not elevation. It appears the driver sets the default max_el to 0. Rotctl seems to support "-C max_el=90", but the elevation still doesn't change.

NUCARADO, 2023-03-25:
`Rotator command: P
Azimuth: 0
Elevation: 20
rot_set_position called az=0.00 el=20.00
rot_set_position: south_zero=0
rt21_rot_set_position called
rotorez_send_priv_cmd called
write_block(): TX 10 bytes, method=2
0000    41 50 31 30 30 30 2e 30 0d 3b                       AP1000.0.;
rotorez_send_priv_cmd2 called
;write_block(): TX 10 bytes, method=2
0000    41 50 31 30 32 30 2e 30 0d 3b                       AP1020.0.;

Rotator command: p
rot_get_position called
rt21_rot_get_position called
rotorez_send_priv_cmd called
write_block(): TX 4 bytes, method=2
0000    42 49 31 3b                                         BI1;
read_string_generic called, rxmax=7 direct=1, expected_len=1
read_string_generic(): RX 5 characters, direct=1
0000    20 33 2e 37 3b                                       3.7;
rt21_rot_get_position: " 3.7;" after conversion = 3.7
rt21_rot_get_position: azimuth = 3.7 deg; elevation = 0.0 deg
rot_get_position: got az=3.70, el=0.00
Azimuth: 3.70
Elevation: 0.00`

NUCARADO, 2023-03-25:
Looks like this is two errors. "P" doesn't get the elevation motor moving, and "p" doesn't query for the second value.

mdblack98, 2023-03-26:
Do you have an RT-21azel?
Mike W9MDB

NUCARADO, 2023-03-26:
Yes I do.

closed 2023-03-28 by mdblack98