Hamlib/Hamlib #1271: TS-590SG does not like extra semicolons with PS command

mikaelnousiainen, 2023-04-11:
I'm trying to fix several bugs in TS-590 backend (now testing with SG model) and it seems to respond with a `?;` response if there is just a semicolon with no command (added in #1226). I think we'll need to make the extra semicolons rig-specific, e.g. enable only for TS-480.

Although I'm not sure why they're needed, as I've been running a remote station with a TS-480SAT for over 2 years with power ON/OFF working perfectly.

I can apply fixes here on my own, but just want to clarify what the issue with TS-480 is so I can fix this for TS-590SG.

mdblack98, 2023-04-11:
TS-480 requires that dummy data to power on from sleep mode.
Looks like the TS-590 is a whole 'nuther story.  Turn off flow control if used, send one ;, wait 200ms. then send PS1 within 2 seconds.

mikaelnousiainen, 2023-04-11:
Yeah that too, and when the rig is powered on you cannot send just "lonely" semicolons without getting a `?;` response that interferes with the response parsing...

mdblack98, 2023-04-11:
Most every other Yaesu/Kenwood works....sigh....

mdblack98, 2023-04-11:
Does the TS-590SG respond to PS; when in PS0; state?
Message ID: ***@***.***>

mikaelnousiainen, 2023-04-11:
It does, and when it's powered off, the extra semicolons are OK ... go figure:

```Rig command: \send_raw ; ;;PS
rigctl_parse: input_line: \send_raw ; ;;PS
rigctl_parse: vfo_opt=0
rigctl_send_raw:
 1:rig.c(7699):rig_send_raw entered
rig_send_raw: writing 4 bytes
write_block(): TX 4 bytes, method=2
0000    3b 3b 50 53                                         ;;PS            
rig_send_raw: reading frame terminated by ';'
read_string_generic called, rxmax=200 direct=1, expected_len=1
read_string_generic(): RX 4 characters, direct=1
0000    50 53 30 3b                                         PS0;            
 1:rig_send_raw: elapsed=3ms
 1:rig.c(7774):rig_send_raw returning(4) 
Send raw answer: PS0;
rigctl_parse: called, interactive=1
```

mikaelnousiainen, 2023-04-25:
This was fixed in #1274 by adding a rig model-specific flag to enable/disable the semicolons

closed 2023-04-25 by mikaelnousiainen