Hamlib/Hamlib #1277: Icom vfo swapping

mdblack98, 2023-04-21:
In Hamlib-4.5.5 vfo swapping is occuring in Icom rigs.
x25cmdfails is getting set to true when async is turned on.

mikaelnousiainen, 2023-05-10:
@mdblack98 Setting `x25cmdfails` to `-1` initially breaks things, e.g. `icom_get_freq` and `icom_get_split_freq` still expect only `x25cmdfails == 0` to do the check. Now the check isn't happening in these functions and `icom_get_freq` fails on an IC-7300 with `retval == -RIG_EINTERNAL` as `retval` is set to that value initially (which is confusing, I will clean up the logic).

What is the meaning of value `-1` for `x25cmdfails` ? If the goal here is to do a single detection only, it's not working right now.

I can fix this now, as I'm working on getting the latest master to work on different rigs again...

mikaelnousiainen, 2023-05-10:
Thinking of this logic here: the 0x25 command check should happen only if `x25cmdfails == -1` and then set to `0` or `1` based on the result of the check. This isn't the case right now in all the places that use `x25cmdfails` and attempt the check.

mikaelnousiainen, 2023-05-10:
Ok, so just to clarify this, there are two issues:

1) Nothing sets `x25cmdfails` to zero in `icom_get_freq` or `icom_get_split_freq`
2) `icom_get_freq` doesn't use 0x25 command at all now, since it is -1 all the time (and returns error code, since `retval` doesn't get changed from its initial value)

Shouldn't the existing code (before the change to set `x25cmdfails` to `-1`) disable use of 0x25 always if it's not supported (if the command fails)? I don't understand this change.

mikaelnousiainen, 2023-05-10:
Ok. I think I got it. After detecting once that command 0x25 works, it should not be marked as unsupported even if it fails on a subsequent call.

closed 2023-07-24 by mdblack98