Hamlib/Hamlib #1282: get_powerstat with NETrig

PianetaRadio, 2023-04-28:
Still problem with this issue, hamlib 4.5.5.

Radio (Yaesu FT-991A), power supply off then on, radio off:

```
rigctl -m 2 -vvvvv -Z --set-conf=auto_power_on=1
2023-04-23T20:40:50.842966+0100: rigctl.c(445) Startup: rigctl -m 2 -vvvvv -Z --set-conf=auto_power_on=1
2023-04-23T20:40:50.843235+0100: rigctl Hamlib 4.5.5 Apr 05 11:43:08Z 2023 SHA=6eecd3
2023-04-23T20:40:50.843293+0100: Report bugs to <hamlib-developer@lists.sourceforge.net>

2023-04-23T20:40:50.843367+0100: rig_check_rig_caps: p1=0xb6f1a158, p2=0xb6f1f8a4, rig_model=0xb6f1a158, macro_name=0xb6f1f8a4
2023-04-23T20:40:50.843558+0100: initrigs4_dummy: _init called
2023-04-23T20:40:50.843675+0100: rig_init: rig_model=Hamlib NET rigctl
2023-04-23T20:40:50.843831+0100: rig_init: rx_range_list1 is empty, using rx_range_list2
2023-04-23T20:40:50.843892+0100: rig_init: rig does not have tx_range!!
2023-04-23T20:40:50.844016+0100: netrigctl_init version 20230328.0
2023-04-23T20:40:50.844137+0100: rig_token_lookup called for auto_power_on
2023-04-23T20:40:50.844203+0100: rig_confparam_lookup called for auto_power_on
2023-04-23T20:40:50.844279+0100: rig_set_conf called
2023-04-23T20:40:50.844338+0100: rig_confparam_lookup called for 1073741948
2023-04-23T20:40:50.844398+0100: rig_set_conf: auto_power_on='1'
2023-04-23T20:40:50.844570+0100:  1:rig.c(817):rig_open entered
2023-04-23T20:40:50.844661+0100: rig_settings_get_path: path=/home/pi/.config//home/pi/hamlib_settings
2023-04-23T20:40:50.844848+0100: rig_settings_load_all: settings_file (/home/pi/.config//home/pi/hamlib_settings): No such file or directory
2023-04-23T20:40:50.844962+0100: rig_open: cwd=/home/pi
2023-04-23T20:40:50.845107+0100: rig_open: /home/pi/.config//home/pi/hamlib_settings does not exist
2023-04-23T20:40:50.845178+0100: rig_open: async_data_enable=1, async_data_supported=0
2023-04-23T20:40:50.845286+0100: rig_open: using network address 127.0.0.1:4532
2023-04-23T20:40:50.845370+0100: network_open: hoststr=127.0.0.1, portstr=4532
2023-04-23T20:40:50.845927+0100: network_open: client port=44165
2023-04-23T20:40:50.846008+0100:   2:rig.c(7481):async_data_handler_start entered
2023-04-23T20:40:50.846078+0100: async_data_handler_start: async data support disabled since async_data_enabled=0
2023-04-23T20:40:50.846181+0100:   2:rig.c(7488):async_data_handler_start returning(0) 
2023-04-23T20:40:50.846260+0100: rig.c(254):add_opened_rig returning2(0) 
2023-04-23T20:40:50.846409+0100: rig_open: 0x1ce1134 rs->comm_state==1?=1
2023-04-23T20:40:50.846545+0100:   2:rig.c(6084):rig_get_powerstat entered
2023-04-23T20:40:50.846668+0100: rig.c(6104) trace
2023-04-23T20:40:50.846788+0100: netrigctl_get_powerstat called
2023-04-23T20:40:50.846908+0100: netrigctl_transaction: called len=15
2023-04-23T20:40:50.847032+0100: rig_flush: called for network device
2023-04-23T20:40:50.847159+0100: network_flush called
2023-04-23T20:40:50.847478+0100: write_block(): TX 15 bytes, method=2
2023-04-23T20:40:50.847629+0100: 0000    5c 67 65 74 5f 70 6f 77 65 72 73 74 61 74 0a        \get_powerstat. 
2023-04-23T20:40:50.847758+0100: read_string_generic called, rxmax=1024 direct=1, expected_len=1
2023-04-23T20:41:00.857967+0100: read_string_generic(): Timed out 10.010 seconds after 0 chars, direct=1
2023-04-23T20:41:00.858072+0100: netrigctl_get_powerstat: PS command failed (ret=-5) so returning RIG_POWER_ON
2023-04-23T20:41:00.858144+0100:   2:rig.c(6109):rig_get_powerstat returning(0) 
2023-04-23T20:41:00.858214+0100:   2:netrigctl.c(261):netrigctl_open entered
2023-04-23T20:41:00.858278+0100: netrigctl_transaction: called len=9
2023-04-23T20:41:00.858340+0100: rig_flush: called for network device
2023-04-23T20:41:00.858406+0100: network_flush called
2023-04-23T20:41:00.858612+0100: write_block(): TX 9 bytes, method=2
2023-04-23T20:41:00.858685+0100: 0000    5c 63 68 6b 5f 76 66 6f 0a                          \chk_vfo.       
2023-04-23T20:41:00.858747+0100: read_string_generic called, rxmax=1024 direct=1, expected_len=1
2023-04-23T20:41:10.868889+0100: read_string_generic(): Timed out 10.010 seconds after 0 chars, direct=1
2023-04-23T20:41:10.869090+0100: netrigctl_open: chk_vfo error: netrigctl_get_powerstat called
netrigctl_transaction: called len=15
rig_flush: called for network device
network_flush called
write_block(): TX 15 bytes, method=2
0000    5c 67 65 74 5f 70 6f 77 65 72 73 74 61 74 0a        \get_powerstat. 
read_string_generic called, rxmax=1024 direct=1, expected_len=1
read_string_generic(): Timed out 10.010 seconds after 0 chars, direct=1
netrigctl_get_powerstat: PS command failed (ret=-5) so returning RIG_POWER_ON
  2:rig.c(6109):rig_get_powerstat returning(0) 
  2:netrigctl.c(261):netrigctl_open entered
netrigctl_transaction: called len=9
rig_flush: called for network device
network_flush called
write_block(): TX 9 bytes, method=2
0000    5c 63 68 6b 5f 76 66 6f 0a                          \chk_vfo.       
read_string_generic called, rxmax=1024 direct=1, expected_len=1
read_string_generic(): Timed out 10.010 seconds after 0 chars, direct=1
Communication timed out
Communication timed out

2023-04-23T20:41:10.869411+0100: netrigctl_open: vfo_mode=0
2023-04-23T20:41:10.869533+0100: netrigctl_transaction: called len=12
2023-04-23T20:41:10.869601+0100: rig_flush: called for network device
2023-04-23T20:41:10.869673+0100: network_flush called
2023-04-23T20:41:10.869916+0100: write_block(): TX 12 bytes, method=2
2023-04-23T20:41:10.869999+0100: 0000    5c 64 75 6d 70 5f 73 74 61 74 65 0a                 \dump_state.    
2023-04-23T20:41:10.870067+0100: read_string_generic called, rxmax=1024 direct=1, expected_len=1
2023-04-23T20:41:20.876247+0100: read_string_generic(): Timed out 10.006 seconds after 0 chars, direct=1
2023-04-23T20:41:20.876421+0100:   2:netrigctl.c(299):netrigctl_open returning(-5) Communication timed out

2023-04-23T20:41:20.876511+0100:   2:rig.c(7525):async_data_handler_stop entered
2023-04-23T20:41:20.876587+0100:   2:rig.c(7555):async_data_handler_stop returning(0) 
2023-04-23T20:41:20.876836+0100: network_close: close socket ret=0
2023-04-23T20:41:20.876917+0100:  1:rig.c(1301):rig_open returning(-5) Communication timed out

rig_open: error = (null) No such file or directory 
```

Remarks:

    Using rigctl with serial connection no problem, it reports power status off and I can turn on the radio on with set_powerstat.
    Then if I connect trough NET rig, it works.
    If I power off the radio, exit from rigctl and run again the previous command, now it works well reporting radio power status off.

mdblack98, 2023-04-29:
Try this package
https://www.dropbox.com/s/tgeouohpw8x7ezt/bin.zip?dl=0

Mike W9MDB

PianetaRadio, 2023-04-29:
Hi Mike,
tnx I will test under RPi, so I will wait for the last daily snapshot by Nate.

Keep updated

mdblack98, 2023-04-30:
And if you want to allow the rig to power off you may also want the -R option for rigctld which will close the rig when no clients are connected.  
Mike W9MDB

 

https://www.dropbox.com/s/tgeouohpw8x7ezt/bin.zip?dl=0

PianetaRadio, 2023-04-30:
No, problem not solved

```
rigctl -m 2 -vvvvv -Z --set-conf=auto_power_on=1
2023-04-30T19:58:33.427952+0100: rigctl.c(442) Startup: rigctl -m 2 -vvvvv -Z --set-conf=auto_power_on=1
2023-04-30T19:58:33.428282+0100: rigctl Hamlib 4.6~git Apr 29 22:10:25Z 2023 SHA=2b37e8 32-bit
2023-04-30T19:58:33.428355+0100: Report bugs to <hamlib-developer@lists.sourceforge.net>

2023-04-30T19:58:33.428442+0100: rig_check_rig_caps: p1=0xb6e6d5b8, p2=0xb6e72d04, rig_model=0xb6e6d5b8, macro_name=0xb6e72d04
2023-04-30T19:58:33.428626+0100: initrigs4_dummy: _init called
2023-04-30T19:58:33.428718+0100: rig_init: rig_model=Hamlib NET rigctl 20204010.0
2023-04-30T19:58:33.428836+0100: rig_init: rx_range_list1 is empty, using rx_range_list2
2023-04-30T19:58:33.428882+0100: rig_init: rig does not have tx_range!!
2023-04-30T19:58:33.428931+0100: netrigctl_init version 20204010.0
2023-04-30T19:58:33.428984+0100: rig_token_lookup called for auto_power_on
2023-04-30T19:58:33.429031+0100: rig_confparam_lookup called for auto_power_on
2023-04-30T19:58:33.429083+0100: rig_token_lookup called for auto_power_on
2023-04-30T19:58:33.429126+0100: rig_confparam_lookup called for auto_power_on
2023-04-30T19:58:33.429173+0100: rig_set_conf called
2023-04-30T19:58:33.429217+0100: rig_confparam_lookup called for 1073741948
2023-04-30T19:58:33.429261+0100: rig_set_conf: auto_power_on='1'
2023-04-30T19:58:33.429365+0100:  1:rig.c(827):rig_open entered
2023-04-30T19:58:33.429470+0100: rig_settings_get_path: path=/home/pi/.config/hamlib_settings
2023-04-30T19:58:33.429615+0100: rig_settings_load_all: settings_file (/home/pi/.config/hamlib_settings): No such file or directory
2023-04-30T19:58:33.429686+0100: rig_open: cwd=/home/pi
2023-04-30T19:58:33.429797+0100: rig_open: /home/pi/.config//home/pi/hamlib_settings does not exist
2023-04-30T19:58:33.429855+0100: rig_open: async_data_enable=1, async_data_supported=0
2023-04-30T19:58:33.429930+0100: rig_open: using network address 127.0.0.1:4532
2023-04-30T19:58:33.429993+0100: network_open: hoststr=127.0.0.1, portstr=4532
2023-04-30T19:58:33.430395+0100: network_open: client port=27333
2023-04-30T19:58:33.430450+0100:   2:rig.c(7520):async_data_handler_start entered
2023-04-30T19:58:33.430499+0100: async_data_handler_start: async data support disabled since async_data_enabled=0
2023-04-30T19:58:33.430588+0100:   2:rig.c(7527):async_data_handler_start returning(0) 
2023-04-30T19:58:33.430644+0100: rig.c(262):add_opened_rig returning2(0) 
2023-04-30T19:58:33.430693+0100: rig_open: 0x18cd134 rs->comm_state==1?=1
2023-04-30T19:58:33.430749+0100:   2:rig.c(6110):rig_get_powerstat entered
2023-04-30T19:58:33.430797+0100: rig.c(6130) trace
2023-04-30T19:58:33.430846+0100: netrigctl_get_powerstat called
2023-04-30T19:58:33.430892+0100: netrigctl_transaction: called len=15
2023-04-30T19:58:33.430942+0100: rig_flush: called for network device
2023-04-30T19:58:33.430996+0100: network_flush called
2023-04-30T19:58:33.431203+0100: write_block(): TX 15 bytes, method=2
2023-04-30T19:58:33.431296+0100: 0000    5c 67 65 74 5f 70 6f 77 65 72 73 74 61 74 0a        \get_powerstat. 
2023-04-30T19:58:33.431363+0100: read_string_generic called, rxmax=1024 direct=1, expected_len=1
2023-04-30T19:58:43.441522+0100: read_string_generic(): Timed out 10.010 seconds after 0 chars, direct=1
2023-04-30T19:58:43.441626+0100: netrigctl_get_powerstat: PS command failed (ret=-5) so returning RIG_POWER_ON
2023-04-30T19:58:43.441705+0100:   2:rig.c(6144):rig_get_powerstat returning(0) 
2023-04-30T19:58:43.441779+0100:   2:netrigctl.c(257):netrigctl_open entered
2023-04-30T19:58:43.441846+0100: netrigctl_transaction: called len=9
2023-04-30T19:58:43.441912+0100: rig_flush: called for network device
2023-04-30T19:58:43.441984+0100: network_flush called
2023-04-30T19:58:43.442270+0100: write_block(): TX 9 bytes, method=2
2023-04-30T19:58:43.442352+0100: 0000    5c 63 68 6b 5f 76 66 6f 0a                          \chk_vfo.       
2023-04-30T19:58:43.442421+0100: read_string_generic called, rxmax=1024 direct=1, expected_len=1
2023-04-30T19:58:53.452570+0100: read_string_generic(): Timed out 10.010 seconds after 0 chars, direct=1
2023-04-30T19:58:53.452787+0100: netrigctl_open: chk_vfo error: netrigctl_get_powerstat called
netrigctl_transaction: called len=15
rig_flush: called for network device
network_flush called
write_block(): TX 15 bytes, method=2
0000    5c 67 65 74 5f 70 6f 77 65 72 73 74 61 74 0a        \get_powerstat. 
read_string_generic called, rxmax=1024 direct=1, expected_len=1
read_string_generic(): Timed out 10.010 seconds after 0 chars, direct=1
netrigctl_get_powerstat: PS command failed (ret=-5) so returning RIG_POWER_ON
  2:rig.c(6144):rig_get_powerstat returning(0) 
  2:netrigctl.c(257):netrigctl_open entered
netrigctl_transaction: called len=9
rig_flush: called for network device
network_flush called
write_block(): TX 9 bytes, method=2
0000    5c 63 68 6b 5f 76 66 6f 0a                          \chk_vfo.       
read_string_generic called, rxmax=1024 direct=1, expected_len=1
read_string_generic(): Timed out 10.010 seconds after 0 chars, direct=1
Communication timed out
Communication timed out

2023-04-30T19:58:53.453105+0100: netrigctl_open: vfo_mode=0
2023-04-30T19:58:53.453210+0100: netrigctl_transaction: called len=12
2023-04-30T19:58:53.453269+0100: rig_flush: called for network device
2023-04-30T19:58:53.453330+0100: network_flush called
2023-04-30T19:58:53.453545+0100: write_block(): TX 12 bytes, method=2
[... cut ...]
2023-04-30T19:59:03.463826+0100: read_string_generic(): Timed out 10.010 seconds after 0 chars, direct=1
2023-04-30T19:59:03.463995+0100:   2:netrigctl.c(295):netrigctl_open returning(-5) Communication timed out

2023-04-30T19:59:03.464082+0100:   2:rig.c(7564):async_data_handler_stop entered
2023-04-30T19:59:03.464158+0100:   2:rig.c(7594):async_data_handler_stop returning(0) 
2023-04-30T19:59:03.464411+0100: network_close: close socket ret=0
2023-04-30T19:59:03.464493+0100:  1:rig.c(1315):rig_open returning(-5) Communication timed out
```

PianetaRadio, 2023-04-30:
I don't understand why the \get_powerstat doesn't receive reply and goes in timeout.
If I run the request directly (-m 1035 via serial port), it replies as expected:

```
rigctl -m 1035 -r /dev/ttyUSB1 -s 38400 -vvvvv -Z
2023-04-30T20:04:57.160382+0100: rigctl.c(442) Startup: rigctl -m 1035 -r /dev/ttyUSB1 -s 38400 -vvvvv -Z
2023-04-30T20:04:57.160702+0100: rigctl Hamlib 4.6~git Apr 29 22:10:25Z 2023 SHA=2b37e8 32-bit
2023-04-30T20:04:57.160764+0100: Report bugs to <hamlib-developer@lists.sourceforge.net>

2023-04-30T20:04:57.160841+0100: rig_check_rig_caps: p1=0xb6edc5b8, p2=0xb6ee1d04, rig_model=0xb6edc5b8, macro_name=0xb6ee1d04
2023-04-30T20:04:57.161049+0100: yaesu: initrigs4_yaesu called
2023-04-30T20:04:57.161393+0100: rig_init: rig_model=Yaesu FT-991 20230328.15
2023-04-30T20:04:57.161584+0100: rig_init: rig has VFO_A
2023-04-30T20:04:57.161646+0100: rig_init: rig has VFO_B
2023-04-30T20:04:57.161703+0100: rig_init: rig has VFO_MEM
2023-04-30T20:04:57.161836+0100: ft991_init called, version 20230328.15
2023-04-30T20:04:57.162016+0100:  1:newcat.c(466):newcat_init entered
2023-04-30T20:04:57.162149+0100:  1:newcat.c(507):newcat_init returning(0) 
2023-04-30T20:04:57.162284+0100:  1:rig.c(827):rig_open entered
2023-04-30T20:04:57.162384+0100: rig_settings_get_path: path=/home/pi/.config/hamlib_settings
2023-04-30T20:04:57.162584+0100: rig_settings_load_all: settings_file (/home/pi/.config/hamlib_settings): No such file or directory
2023-04-30T20:04:57.162681+0100: rig_open: cwd=/home/pi
2023-04-30T20:04:57.162855+0100: rig_open: /home/pi/.config//home/pi/hamlib_settings does not exist
2023-04-30T20:04:57.162930+0100: rig_open: async_data_enable=1, async_data_supported=0
2023-04-30T20:04:57.163061+0100: serial_open: /dev/ttyUSB1
2023-04-30T20:04:57.164063+0100: serial_setup: tcgetattr
2023-04-30T20:04:57.164186+0100: serial_setup: cfmakeraw
2023-04-30T20:04:57.164262+0100: serial_setup: cfsetispeed=38400,0x000f
2023-04-30T20:04:57.164338+0100: serial_setup: cfsetospeed=38400,0x000f
2023-04-30T20:04:57.164411+0100: serial_setup: data_bits=8
2023-04-30T20:04:57.164479+0100: serial_setup: parity=0
2023-04-30T20:04:57.164541+0100: serial_setup: Handshake=Hardware
2023-04-30T20:04:57.164618+0100: serial_setup: tcsetattr TCSANOW
2023-04-30T20:04:57.164765+0100: read_string_generic called, rxmax=4095 direct=1, expected_len=1
2023-04-30T20:04:57.165957+0100: tcflush
2023-04-30T20:04:57.216139+0100:   2:rig.c(7520):async_data_handler_start entered
2023-04-30T20:04:57.216209+0100: async_data_handler_start: async data support disabled since async_data_enabled=0
2023-04-30T20:04:57.216254+0100:   2:rig.c(7527):async_data_handler_start returning(0) 
2023-04-30T20:04:57.216298+0100: rig.c(262):add_opened_rig returning2(0) 
2023-04-30T20:04:57.216340+0100: rig_open: 0xada134 rs->comm_state==1?=1
2023-04-30T20:04:57.216390+0100:   2:rig.c(6110):rig_get_powerstat entered
2023-04-30T20:04:57.216431+0100: rig.c(6130) trace
2023-04-30T20:04:57.216509+0100:    3:newcat.c(3623):newcat_get_powerstat entered
2023-04-30T20:04:57.216553+0100: newcat_valid_command PS
2023-04-30T20:04:57.216597+0100: newcat.c(7668):newcat_valid_command returning2(1) 
2023-04-30T20:04:57.216679+0100: write_block(): TX 3 bytes, method=2
2023-04-30T20:04:57.216726+0100: 0000    50 53 3b                                            PS;             
2023-04-30T20:04:58.419045+0100:     4:newcat.c(10492):newcat_get_cmd entered
2023-04-30T20:04:58.419212+0100: rig_flush: called for serial device
2023-04-30T20:04:58.419333+0100: read_string_generic called, rxmax=4095 direct=1, expected_len=1
2023-04-30T20:04:58.420542+0100: tcflush
2023-04-30T20:04:58.420664+0100: cmd_str = PS;
2023-04-30T20:04:58.420807+0100: write_block(): TX 3 bytes, method=2
2023-04-30T20:04:58.420925+0100: 0000    50 53 3b                                            PS;             
2023-04-30T20:04:58.423114+0100: read_string_generic called, rxmax=129 direct=1, expected_len=1
2023-04-30T20:04:58.489675+0100: read_string_generic(): RX 4 characters, direct=1
2023-04-30T20:04:58.489829+0100: 0000    50 53 30 3b                                         PS0;            
2023-04-30T20:04:58.489944+0100: newcat_get_cmd: read count = 4, ret_data = PS0;
2023-04-30T20:04:58.490071+0100:     4:newcat.c(10710):newcat_get_cmd returning(0) 
2023-04-30T20:04:58.490188+0100:    3:newcat.c(3661):newcat_get_powerstat returning(0) 
2023-04-30T20:04:58.490304+0100:   2:rig.c(6144):rig_get_powerstat returning(0) 
2023-04-30T20:04:58.490420+0100: rig_open: rig power is off, use --set-conf=auto_power_on=1 if power on is wanted
```

mdblack98, 2023-04-30:
On Linux try the -R switch on rigctld.  

In my tests here the Linux side rigctld held on to the serial port and power off/on the 991 created a new serial port instead of using the old one.
So rigctld needs to let go of the serial port with no connections....ergo the -R switch should fix it.

Mike W9MDB

PianetaRadio, 2023-04-30:
1) Run rigctld with radio off

```
rigctld -m 1035 -r /dev/ttyUSB1 -s 38400 -vvvvv -Z -R
2023-05-01T00:36:00.740357+0100: rigctld.c(660) Startup: rigctld -m 1035 -r /dev/ttyUSB1 -s 38400 -vvvvv -Z -R
2023-05-01T00:36:00.740620+0100: rigctld Hamlib 4.6~git Apr 29 22:10:25Z 2023 SHA=2b37e8 32-bit
2023-05-01T00:36:00.740679+0100: Report bugs to <hamlib-developer@lists.sourceforge.net>

2023-05-01T00:36:00.740743+0100: Max# of rigctld client services=32
2023-05-01T00:36:00.740809+0100: rig_check_rig_caps: p1=0xb6ec45b8, p2=0xb6ec9d04, rig_model=0xb6ec45b8, macro_name=0xb6ec9d04
2023-05-01T00:36:00.741006+0100: yaesu: initrigs4_yaesu called
2023-05-01T00:36:00.741317+0100: rig_init: rig_model=Yaesu FT-991 20230328.15
2023-05-01T00:36:00.741497+0100: rig_init: rig has VFO_A
2023-05-01T00:36:00.741556+0100: rig_init: rig has VFO_B
2023-05-01T00:36:00.741611+0100: rig_init: rig has VFO_MEM
2023-05-01T00:36:00.741734+0100: ft991_init called, version 20230328.15
2023-05-01T00:36:00.741856+0100:  1:newcat.c(466):newcat_init entered
2023-05-01T00:36:00.741948+0100:  1:newcat.c(507):newcat_init returning(0) 
2023-05-01T00:36:00.742116+0100: main: twiddle=0, uplink=0, twiddle_rit=0
2023-05-01T00:36:00.742237+0100:  1:rig.c(827):rig_open entered
2023-05-01T00:36:00.742328+0100: rig_settings_get_path: path=/home/pi/.config/hamlib_settings
2023-05-01T00:36:00.742572+0100: rig_settings_load_all: settings_file (/home/pi/.config/hamlib_settings): No such file or directory
2023-05-01T00:36:00.742668+0100: rig_open: cwd=/home/pi
2023-05-01T00:36:00.742823+0100: rig_open: /home/pi/.config//home/pi/hamlib_settings does not exist
2023-05-01T00:36:00.742895+0100: rig_open: async_data_enable=1, async_data_supported=0
2023-05-01T00:36:00.743134+0100: serial_open: /dev/ttyUSB1
2023-05-01T00:36:00.747597+0100: serial_setup: tcgetattr
2023-05-01T00:36:00.747736+0100: serial_setup: cfmakeraw
2023-05-01T00:36:00.747815+0100: serial_setup: cfsetispeed=38400,0x000f
2023-05-01T00:36:00.747900+0100: serial_setup: cfsetospeed=38400,0x000f
2023-05-01T00:36:00.748096+0100: serial_setup: data_bits=8
2023-05-01T00:36:00.748173+0100: serial_setup: parity=0
2023-05-01T00:36:00.748243+0100: serial_setup: Handshake=Hardware
2023-05-01T00:36:00.748437+0100: serial_setup: tcsetattr TCSANOW
2023-05-01T00:36:00.753147+0100: read_string_generic called, rxmax=4095 direct=1, expected_len=1
2023-05-01T00:36:00.754363+0100: tcflush
2023-05-01T00:36:00.804673+0100:   2:rig.c(7520):async_data_handler_start entered
2023-05-01T00:36:00.804751+0100: async_data_handler_start: async data support disabled since async_data_enabled=0
2023-05-01T00:36:00.804794+0100:   2:rig.c(7527):async_data_handler_start returning(0) 
2023-05-01T00:36:00.804839+0100: rig.c(262):add_opened_rig returning2(0) 
2023-05-01T00:36:00.804878+0100: rig_open: 0xb9134 rs->comm_state==1?=1
2023-05-01T00:36:00.804926+0100:   2:rig.c(6110):rig_get_powerstat entered
2023-05-01T00:36:00.804966+0100: rig.c(6130) trace
2023-05-01T00:36:00.805004+0100:    3:newcat.c(3623):newcat_get_powerstat entered
2023-05-01T00:36:00.805045+0100: newcat_valid_command PS
2023-05-01T00:36:00.805089+0100: newcat.c(7668):newcat_valid_command returning2(1) 
2023-05-01T00:36:00.805177+0100: write_block(): TX 3 bytes, method=2
2023-05-01T00:36:00.805218+0100: 0000    50 53 3b                                            PS;             
2023-05-01T00:36:02.007581+0100:     4:newcat.c(10492):newcat_get_cmd entered
2023-05-01T00:36:02.008217+0100: rig_flush: called for serial device
2023-05-01T00:36:02.008703+0100: read_string_generic called, rxmax=4095 direct=1, expected_len=1
2023-05-01T00:36:02.010281+0100: tcflush
2023-05-01T00:36:02.010759+0100: cmd_str = PS;
2023-05-01T00:36:02.011278+0100: write_block(): TX 3 bytes, method=2
2023-05-01T00:36:02.011751+0100: 0000    50 53 3b                                            PS;             
2023-05-01T00:36:02.014380+0100: read_string_generic called, rxmax=129 direct=1, expected_len=1
2023-05-01T00:36:02.014997+0100: read_string_generic(): RX 4 characters, direct=1
2023-05-01T00:36:02.015482+0100: 0000    50 53 30 3b                                         PS0;            
2023-05-01T00:36:02.015951+0100: newcat_get_cmd: read count = 4, ret_data = PS0;
2023-05-01T00:36:02.016266+0100:     4:newcat.c(10710):newcat_get_cmd returning(0) 
2023-05-01T00:36:02.016576+0100:    3:newcat.c(3661):newcat_get_powerstat returning(0) 
2023-05-01T00:36:02.016707+0100:   2:rig.c(6144):rig_get_powerstat returning(0) 
2023-05-01T00:36:02.017018+0100: rig_open: rig power is off, use --set-conf=auto_power_on=1 if power on is wanted
Opened rig model 1035, 'FT-991'
rig_open: error = newcat_valid_command PS
newcat.c(7668):newcat_valid_command returning2(1) 
write_block(): TX 3 bytes, method=2
0000    50 53 3b                                            PS;             
    4:newcat.c(10492):newcat_get_cmd entered
rig_flush: called for serial device
[... 31 lines cut ...]
2023-05-01T00:36:02.025564+0100: tcflush
2023-05-01T00:36:02.025708+0100: write_block(): TX 4 bytes, method=2
2023-05-01T00:36:02.025783+0100: 0000    41 49 30 3b                                         AI0;            
2023-05-01T00:36:02.027992+0100: write_block(): TX 1 bytes, method=2
2023-05-01T00:36:02.028068+0100: 0000    3b                                                  ;               
2023-05-01T00:36:02.030232+0100: read_string_generic called, rxmax=129 direct=1, expected_len=1
[... cut ...]
2023-05-01T00:36:05.819694+0100:    3:rig.c(1577):rig_close returning(0) 
Closed rig model 1035, 'FT-991 - will reopen for clients'
2023-05-01T00:36:05.820637+0100: main: Using IPV4
2023-05-01T00:36:05.820729+0100:    3:network.c(977):network_multicast_publisher_start entered
2023-05-01T00:36:05.820800+0100: network.c(981):multicast address=0.0.0.0, port=4532
2023-05-01T00:36:05.820872+0100: network.c(986): not starting multicast publisher
2023-05-01T00:36:05.821056+0100: main: rigctld listening on port 4532

```

PianetaRadio, 2023-04-30:
2. Run rigctl, it fails

```
rigctl -m 2 -vvvvv -Z --set-conf=auto_power_on=1
2023-05-01T00:40:09.830071+0100: rigctl.c(442) Startup: rigctl -m 2 -vvvvv -Z --set-conf=auto_power_on=1
2023-05-01T00:40:09.830385+0100: rigctl Hamlib 4.6~git Apr 29 22:10:25Z 2023 SHA=2b37e8 32-bit
2023-05-01T00:40:09.830442+0100: Report bugs to <hamlib-developer@lists.sourceforge.net>

2023-05-01T00:40:09.830515+0100: rig_check_rig_caps: p1=0xb6e8c5b8, p2=0xb6e91d04, rig_model=0xb6e8c5b8, macro_name=0xb6e91d04
2023-05-01T00:40:09.830658+0100: initrigs4_dummy: _init called
2023-05-01T00:40:09.830783+0100: rig_init: rig_model=Hamlib NET rigctl 20204010.0
2023-05-01T00:40:09.830938+0100: rig_init: rx_range_list1 is empty, using rx_range_list2
2023-05-01T00:40:09.831001+0100: rig_init: rig does not have tx_range!!
2023-05-01T00:40:09.831069+0100: netrigctl_init version 20204010.0
2023-05-01T00:40:09.831138+0100: rig_token_lookup called for auto_power_on
2023-05-01T00:40:09.831199+0100: rig_confparam_lookup called for auto_power_on
2023-05-01T00:40:09.831264+0100: rig_token_lookup called for auto_power_on
2023-05-01T00:40:09.831323+0100: rig_confparam_lookup called for auto_power_on
2023-05-01T00:40:09.831387+0100: rig_set_conf called
2023-05-01T00:40:09.831446+0100: rig_confparam_lookup called for 1073741948
2023-05-01T00:40:09.831506+0100: rig_set_conf: auto_power_on='1'
2023-05-01T00:40:09.831644+0100:  1:rig.c(827):rig_open entered
2023-05-01T00:40:09.831782+0100: rig_settings_get_path: path=/home/pi/.config/hamlib_settings
2023-05-01T00:40:09.831977+0100: rig_settings_load_all: settings_file (/home/pi/.config/hamlib_settings): No such file or directory
2023-05-01T00:40:09.832160+0100: rig_open: cwd=/home/pi
2023-05-01T00:40:09.832317+0100: rig_open: /home/pi/.config//home/pi/hamlib_settings does not exist
2023-05-01T00:40:09.832392+0100: rig_open: async_data_enable=1, async_data_supported=0
2023-05-01T00:40:09.832494+0100: rig_open: using network address 127.0.0.1:4532
2023-05-01T00:40:09.832577+0100: network_open: hoststr=127.0.0.1, portstr=4532
2023-05-01T00:40:09.833093+0100: network_open: client port=34501
2023-05-01T00:40:09.833175+0100:   2:rig.c(7520):async_data_handler_start entered
2023-05-01T00:40:09.833259+0100: async_data_handler_start: async data support disabled since async_data_enabled=0
2023-05-01T00:40:09.833380+0100:   2:rig.c(7527):async_data_handler_start returning(0) 
2023-05-01T00:40:09.833458+0100: rig.c(262):add_opened_rig returning2(0) 
2023-05-01T00:40:09.833521+0100: rig_open: 0x13fa134 rs->comm_state==1?=1
2023-05-01T00:40:09.833594+0100:   2:rig.c(6110):rig_get_powerstat entered
2023-05-01T00:40:09.833657+0100: rig.c(6130) trace
2023-05-01T00:40:09.833716+0100: netrigctl_get_powerstat called
2023-05-01T00:40:09.833780+0100: netrigctl_transaction: called len=15
2023-05-01T00:40:09.833845+0100: rig_flush: called for network device
2023-05-01T00:40:09.833916+0100: network_flush called
2023-05-01T00:40:09.834157+0100: write_block(): TX 15 bytes, method=2
2023-05-01T00:40:09.834241+0100: 0000    5c 67 65 74 5f 70 6f 77 65 72 73 74 61 74 0a        \get_powerstat. 
2023-05-01T00:40:09.834311+0100: read_string_generic called, rxmax=1024 direct=1, expected_len=1
2023-05-01T00:40:19.844467+0100: read_string_generic(): Timed out 10.010 seconds after 0 chars, direct=1
2023-05-01T00:40:19.844580+0100: netrigctl_get_powerstat: PS command failed (ret=-5) so returning RIG_POWER_ON
2023-05-01T00:40:19.844662+0100:   2:rig.c(6144):rig_get_powerstat returning(0) 
2023-05-01T00:40:19.844739+0100:   2:netrigctl.c(257):netrigctl_open entered
2023-05-01T00:40:19.844821+0100: netrigctl_transaction: called len=9
2023-05-01T00:40:19.844896+0100: rig_flush: called for network device
2023-05-01T00:40:19.844972+0100: network_flush called
2023-05-01T00:40:19.845214+0100: write_block(): TX 9 bytes, method=2
2023-05-01T00:40:19.845299+0100: 0000    5c 63 68 6b 5f 76 66 6f 0a                          \chk_vfo.       
2023-05-01T00:40:19.845362+0100: read_string_generic called, rxmax=1024 direct=1, expected_len=1
2023-05-01T00:40:29.855466+0100: read_string_generic(): Timed out 10.010 seconds after 0 chars, direct=1
2023-05-01T00:40:29.855581+0100: netrigctl_open: chk_vfo error: netrigctl_get_powerstat called
netrigctl_transaction: called len=15
rig_flush: called for network device
network_flush called
write_block(): TX 15 bytes, method=2
0000    5c 67 65 74 5f 70 6f 77 65 72 73 74 61 74 0a        \get_powerstat. 
read_string_generic called, rxmax=1024 direct=1, expected_len=1
read_string_generic(): Timed out 10.010 seconds after 0 chars, direct=1
netrigctl_get_powerstat: PS command failed (ret=-5) so returning RIG_POWER_ON
  2:rig.c(6144):rig_get_powerstat returning(0) 
  2:netrigctl.c(257):netrigctl_open entered
netrigctl_transaction: called len=9
rig_flush: called for network device
network_flush called
write_block(): TX 9 bytes, method=2
0000    5c 63 68 6b 5f 76 66 6f 0a                          \chk_vfo.       
read_string_generic called, rxmax=1024 direct=1, expected_len=1
read_string_generic(): Timed out 10.010 seconds after 0 chars, direct=1
Communication timed out
Communication timed out

2023-05-01T00:40:29.855696+0100: netrigctl_open: vfo_mode=0
2023-05-01T00:40:29.855744+0100: netrigctl_transaction: called len=12
2023-05-01T00:40:29.855772+0100: rig_flush: called for network device
2023-05-01T00:40:29.855800+0100: network_flush called
2023-05-01T00:40:29.855902+0100: write_block(): TX 12 bytes, method=2
[... cut ...]
2023-05-01T00:40:39.862114+0100: read_string_generic(): Timed out 10.006 seconds after 0 chars, direct=1
2023-05-01T00:40:39.862216+0100:   2:netrigctl.c(295):netrigctl_open returning(-5) Communication timed out

2023-05-01T00:40:39.862253+0100:   2:rig.c(7564):async_data_handler_stop entered
2023-05-01T00:40:39.862285+0100:   2:rig.c(7594):async_data_handler_stop returning(0) 
2023-05-01T00:40:39.862432+0100: network_close: close socket ret=0
2023-05-01T00:40:39.862467+0100:  1:rig.c(1315):rig_open returning(-5) Communication timed out

```

PianetaRadio, 2023-04-30:
3. Output from rigctld

```
2023-05-01T00:45:06.245709+0100: Connection opened from 127.0.0.1:50568
2023-05-01T00:45:06.246112+0100: mutex_rigctld: client lock engaged
2023-05-01T00:45:06.246260+0100: mutex_rigctld: client lock disengaged
2023-05-01T00:45:06.246313+0100: mutex_rigctld: client lock engaged
2023-05-01T00:45:06.246362+0100:     4:rig.c(6110):rig_get_powerstat entered
2023-05-01T00:45:06.246439+0100:     4:rig.c(6115):rig_get_powerstat returning(-1) Invalid parameter

2023-05-01T00:45:06.246488+0100: mutex_rigctld: client lock disengaged
2023-05-01T00:45:06.246531+0100: mutex_rigctld: client lock engaged
2023-05-01T00:45:06.246574+0100:     4:rig.c(827):rig_open entered
2023-05-01T00:45:06.246630+0100: rig_settings_get_path: path=/home/pi/.config/hamlib_settings
2023-05-01T00:45:06.246760+0100: rig_settings_load_all: settings_file (/home/pi/.config/hamlib_settings): No such file or directory
2023-05-01T00:45:06.246827+0100: rig_open: cwd=/home/pi
2023-05-01T00:45:06.246907+0100: rig_open: /home/pi/.config//home/pi/hamlib_settings does not exist
2023-05-01T00:45:06.246952+0100: rig_open: async_data_enable=0, async_data_supported=0
2023-05-01T00:45:06.246998+0100: serial_open: /dev/ttyUSB1
2023-05-01T00:45:06.252278+0100: serial_setup: tcgetattr
2023-05-01T00:45:06.252334+0100: serial_setup: cfmakeraw
2023-05-01T00:45:06.252378+0100: serial_setup: cfsetispeed=38400,0x000f
2023-05-01T00:45:06.252420+0100: serial_setup: cfsetospeed=38400,0x000f
2023-05-01T00:45:06.252462+0100: serial_setup: data_bits=8
2023-05-01T00:45:06.252514+0100: serial_setup: parity=0
2023-05-01T00:45:06.252555+0100: serial_setup: Handshake=Hardware
2023-05-01T00:45:06.252599+0100: serial_setup: tcsetattr TCSANOW
2023-05-01T00:45:06.257803+0100: read_string_generic called, rxmax=4095 direct=1, expected_len=1
2023-05-01T00:45:06.258966+0100: tcflush
2023-05-01T00:45:06.309151+0100:      5:rig.c(7520):async_data_handler_start entered
2023-05-01T00:45:06.309244+0100: async_data_handler_start: async data support disabled since async_data_enabled=0
2023-05-01T00:45:06.309301+0100:      5:rig.c(7527):async_data_handler_start returning(0) 
2023-05-01T00:45:06.309354+0100: rig.c(262):add_opened_rig returning2(0) 
2023-05-01T00:45:06.309405+0100: rig_open: 0x1198134 rs->comm_state==1?=1
2023-05-01T00:45:06.309454+0100:      5:rig.c(6110):rig_get_powerstat entered
2023-05-01T00:45:06.309517+0100: rig.c(6130) trace
2023-05-01T00:45:06.309569+0100:       6:newcat.c(3623):newcat_get_powerstat entered
2023-05-01T00:45:06.309616+0100: newcat_valid_command PS
2023-05-01T00:45:06.309666+0100: newcat.c(7668):newcat_valid_command returning2(1) 
2023-05-01T00:45:06.309761+0100: write_block(): TX 3 bytes, method=2
2023-05-01T00:45:06.309815+0100: 0000    50 53 3b                                            PS;             
2023-05-01T00:45:07.512215+0100:        7:newcat.c(10492):newcat_get_cmd entered
2023-05-01T00:45:07.512318+0100: rig_flush: called for serial device
2023-05-01T00:45:07.512376+0100: read_string_generic called, rxmax=4095 direct=1, expected_len=1
2023-05-01T00:45:07.513538+0100: tcflush
2023-05-01T00:45:07.513615+0100: cmd_str = PS;
2023-05-01T00:45:07.513709+0100: write_block(): TX 3 bytes, method=2
2023-05-01T00:45:07.513769+0100: 0000    50 53 3b                                            PS;             
2023-05-01T00:45:07.515913+0100: read_string_generic called, rxmax=129 direct=1, expected_len=1
2023-05-01T00:45:07.516398+0100: read_string_generic(): RX 4 characters, direct=1
2023-05-01T00:45:07.516471+0100: 0000    50 53 30 3b                                         PS0;            
2023-05-01T00:45:07.516524+0100: newcat_get_cmd: read count = 4, ret_data = PS0;
2023-05-01T00:45:07.516583+0100:        7:newcat.c(10710):newcat_get_cmd returning(0) 
2023-05-01T00:45:07.516643+0100:       6:newcat.c(3661):newcat_get_powerstat returning(0) 
2023-05-01T00:45:07.516701+0100:      5:rig.c(6144):rig_get_powerstat returning(0) 
2023-05-01T00:45:07.516758+0100: rig_open: rig power is off, use --set-conf=auto_power_on=1 if power on is wanted
2023-05-01T00:45:07.516818+0100: handle_socket: rig_open reopened retcode=-20
2023-05-01T00:45:07.516901+0100: mutex_rigctld: client lock disengaged
2023-05-01T00:45:07.516958+0100: handle_socket: i/o error
2023-05-01T00:45:07.517015+0100: mutex_rigctld: client lock engaged
2023-05-01T00:45:07.517076+0100:      5:rig.c(1439):rig_close entered
2023-05-01T00:45:07.517137+0100:       6:newcat.c(688):newcat_close entered
2023-05-01T00:45:07.517197+0100:        7:newcat.c(7170):newcat_set_trn entered
2023-05-01T00:45:07.517256+0100: newcat_valid_command AI
2023-05-01T00:45:07.517320+0100: newcat.c(7668):newcat_valid_command returning2(1) 
2023-05-01T00:45:07.517379+0100: cmd_str = AI0;
2023-05-01T00:45:07.517436+0100:         8:newcat.c(10934):newcat_set_cmd entered
2023-05-01T00:45:07.517494+0100: rig_flush: called for serial device
2023-05-01T00:45:07.517553+0100: read_string_generic called, rxmax=4095 direct=1, expected_len=1
2023-05-01T00:45:07.518726+0100: tcflush
[... 423 lines cut ...]
2023-05-01T00:45:39.121482+0100:                 16:newcat.c(717):newcat_close returning(0) 
2023-05-01T00:45:39.121551+0100:                 16:rig.c(7564):async_data_handler_stop entered
2023-05-01T00:45:39.121617+0100:                 16:rig.c(7594):async_data_handler_stop returning(0) 
2023-05-01T00:45:39.121686+0100: ser_close: restoring options
[... cut ...]
2023-05-01T00:45:39.131244+0100: rigctld.c(1135) trace
2023-05-01T00:45:39.131326+0100:                15:network.c(1092):network_multicast_publisher_stop entered
2023-05-01T00:45:39.131396+0100:                15:network.c(1101):network_multicast_publisher_stop returning(0) 
2023-05-01T00:45:39.131466+0100: rigctld.c(1138) trace
2023-05-01T00:45:39.131541+0100:                15:newcat.c(521):newcat_cleanup entered
2023-05-01T00:45:39.131612+0100:                15:newcat.c(530):newcat_cleanup returning(0) 

```

PianetaRadio, 2023-04-30:
It seems that rigctld pass the reply PS0 to rigctl, but this one doens't keep the answer and goes in timeout.

mdblack98, 2023-05-01:
Did you use the -R switch on rigctld?

Got a debug log (from both rigctld and rigctl?)

Mike W9MDB

mdblack98, 2023-05-01:
It doesn't look like you used the auto_power_on with rigctld, did you?

    1. Output from rigctld

mdblack98, 2023-05-01:
You need to use the auto_power_on with rigctld.

mdblack98, 2023-05-01:
Give me two logs -- one from rigctld and one from rigctl for the same test please.

Mike W9MDB

PianetaRadio, 2023-05-01:
1. Radio power supply off
2. rigctld start as service on my RaspberryPi

```
rigctld -m 1035 -r /dev/ttyUSB1 -s 38400 -vvvvv -Z -R --set-conf=auto_power_on=1
2023-05-01T16:16:13.459666+0100: rigctld.c(660) Startup: rigctld -m 1035 -r /dev/ttyUSB1 -s 38400 -vvvvv -Z -R --set-conf=auto_power_on=1
2023-05-01T16:16:13.459958+0100: rigctld Hamlib 4.6~git Apr 29 22:10:25Z 2023 SHA=2b37e8 32-bit
2023-05-01T16:16:13.460021+0100: Report bugs to <hamlib-developer@lists.sourceforge.net>

2023-05-01T16:16:13.460083+0100: Max# of rigctld client services=32
2023-05-01T16:16:13.460153+0100: rig_check_rig_caps: p1=0xb6eee5b8, p2=0xb6ef3d04, rig_model=0xb6eee5b8, macro_name=0xb6ef3d04
2023-05-01T16:16:13.460354+0100: yaesu: initrigs4_yaesu called
2023-05-01T16:16:13.460653+0100: rig_init: rig_model=Yaesu FT-991 20230328.15
2023-05-01T16:16:13.460820+0100: rig_init: rig has VFO_A
2023-05-01T16:16:13.460880+0100: rig_init: rig has VFO_B
2023-05-01T16:16:13.460937+0100: rig_init: rig has VFO_MEM
2023-05-01T16:16:13.461055+0100: ft991_init called, version 20230328.15
2023-05-01T16:16:13.461186+0100:  1:newcat.c(466):newcat_init entered
2023-05-01T16:16:13.461278+0100:  1:newcat.c(507):newcat_init returning(0) 
2023-05-01T16:16:13.461412+0100: rig_token_lookup called for auto_power_on
2023-05-01T16:16:13.461479+0100: rig_confparam_lookup called for auto_power_on
2023-05-01T16:16:13.461552+0100: rig_token_lookup called for auto_power_on
2023-05-01T16:16:13.461612+0100: rig_confparam_lookup called for auto_power_on
2023-05-01T16:16:13.461680+0100: rig_set_conf called
2023-05-01T16:16:13.461740+0100: rig_confparam_lookup called for 1073741948
2023-05-01T16:16:13.461800+0100: rig_set_conf: auto_power_on='1'
2023-05-01T16:16:13.461897+0100: main: twiddle=0, uplink=0, twiddle_rit=0
2023-05-01T16:16:13.462033+0100:  1:rig.c(827):rig_open entered
2023-05-01T16:16:13.462188+0100: rig_settings_get_path: path=/home/pi/.config/hamlib_settings
2023-05-01T16:16:13.462412+0100: rig_settings_load_all: settings_file (/home/pi/.config/hamlib_settings): No such file or directory
2023-05-01T16:16:13.462572+0100: rig_open: cwd=/home/pi
2023-05-01T16:16:13.462749+0100: rig_open: /home/pi/.config//home/pi/hamlib_settings does not exist
2023-05-01T16:16:13.462825+0100: rig_open: async_data_enable=1, async_data_supported=0
2023-05-01T16:16:13.462906+0100: serial_open: /dev/ttyUSB1
2023-05-01T16:16:13.463022+0100: serial_open(229): open failed#1
2023-05-01T16:16:13.963270+0100: serial_open: Unable to open /dev/ttyUSB1 - No such file or directory
2023-05-01T16:16:13.963455+0100: port_open: serial_open(/dev/ttyUSB1) status=-6, err=No such file or directory
2023-05-01T16:16:13.963560+0100: rig_open: rs->comm_state==0?=0
2023-05-01T16:16:13.963686+0100:  1:rig.c(1033):rig_open returning(-6) IO error

Opened rig model 1035, 'FT-991'
rig_open: error = rig_token_lookup called for auto_power_on
rig_confparam_lookup called for auto_power_on
rig_set_conf called
rig_confparam_lookup called for 1073741948
rig_set_conf: auto_power_on='1'
main: twiddle=0, uplink=0, twiddle_rit=0
 1:rig.c(827):rig_open entered
rig_settings_get_path: path=/home/pi/.config/hamlib_settings
rig_settings_load_all: settings_file (/home/pi/.config/hamlib_settings): No such file or directory
rig_open: cwd=/home/pi
rig_open: /home/pi/.config//home/pi/hamlib_settings does not exist
rig_open: async_data_enable=1, async_data_supported=0
serial_open: /dev/ttyUSB1
serial_open(229): open failed#1
serial_open: Unable to open /dev/ttyUSB1 - No such file or directory
port_open: serial_open(/dev/ttyUSB1) status=-6, err=No such file or directory
rig_open: rs->comm_state==0?=0
 1:rig.c(1033):rig_open returning(-6) IO error

IO error
 /dev/ttyUSB1 No such file or directory 
2023-05-01T16:16:13.963926+0100: Backend version: 20230328.15, Status: Stable
2023-05-01T16:16:13.964536+0100:  1:rig.c(1439):rig_close entered
2023-05-01T16:16:13.964638+0100:  1:rig.c(1451):rig_close returning(-1) Invalid parameter

Closed rig model 1035, 'FT-991 - will reopen for clients'
2023-05-01T16:16:13.965190+0100: main: Using IPV4
2023-05-01T16:16:13.965308+0100:  1:network.c(977):network_multicast_publisher_start entered
2023-05-01T16:16:13.965410+0100: network.c(981):multicast address=0.0.0.0, port=4532
[... 1238 lines cut ...]
2023-05-01T16:26:21.704042+0100: cmd_str = NA0;
2023-05-01T16:26:21.704115+0100: write_block(): TX 4 bytes, method=2
2023-05-01T16:26:21.704170+0100: 0000    4e 41 30 3b                                         NA0;            
2023-05-01T16:26:21.706304+0100: read_string_generic called, rxmax=129 direct=1, expected_len=1
2023-05-01T16:26:21.707688+0100: read_string_generic(): RX 5 characters, direct=1
2023-05-01T16:26:21.707745+0100: 0000    4e 41 30 30 3b                                      NA00;           
2023-05-01T16:26:21.707796+0100: newcat_get_cmd: read count = 5, ret_data = NA00;
2023-05-01T16:26:21.707849+0100:          9:newcat.c(10710):newcat_get_cmd returning(0) 
2023-05-01T16:26:21.707906+0100:         8:newcat.c(8117):get_narrow returning(0) 
2023-05-01T16:26:21.707962+0100: newcat_get_rx_bandwidth: RETURNFUNC(RIG_OK)
2023-05-01T16:26:21.708013+0100:        7:newcat.c(10271):newcat_get_rx_bandwidth returning(0) 
2023-05-01T16:26:21.708064+0100:       6:newcat.c(1626):newcat_get_mode returning(0) 
2023-05-01T16:26:21.708116+0100: rig_get_mode: retcode after get_mode=0
2023-05-01T16:26:21.708198+0100: rig_get_mode(2585): debug
[... cut ...]
2023-05-01T16:26:21.708516+0100:      5:rig.c(2601):rig_get_mode returning(0) 
2023-05-01T16:26:21.708581+0100:     4:rigctl_parse.c(2275):rigctl_get_mode returning(0) 
2023-05-01T16:26:21.708700+0100: mutex_rigctld: client lock disengaged
2023-05-01T16:26:21.708756+0100: mutex_rigctld: client lock engaged
2023-05-01T16:26:21.708809+0100: mutex_rigctld: client lock disengaged
2023-05-01T16:26:21.708863+0100: handle_socket: doing rigctl_parse vfo_mode=0, secure=0
2023-05-01T16:26:21.708914+0100: rigctl_parse: called, interactive=1
```

PianetaRadio, 2023-05-01:
So, if I run, as you suggested, rigctld with auto_power_on=1, when a client ask for connection, rigctld power on the radio automatically and all work as expected.

The question is, if I don't want the auto_power_on and want to manually power on the radio once connected, how to do?

mdblack98, 2023-05-01:
It works here on Windows.
Turn 12V power off on rig.
Start rigctld -- gets errors but still starts.
Turn 12V power on (rig still powered off) -- USB ports reappear
rigctl -m --set-conf-auto-poweron=1 
It all works.

If this sequence doesn't work on Linux may have to do with the serial ports not getting re-established.

I need to see the debug logs again.  I have a virtual Linux system and the serial ports get hosed when powering off the rig and have to restart the VM.

Mike W9MDB

PianetaRadio, 2023-05-01:
1. Power supply 12V off (radio USB port not available on the system).
2. Run rigctld (auto_power_on disabled):

```
rigctld -m 1035 -r /dev/ttyUSB1 -s 38400 -vvvvv -Z -R
2023-05-01T19:58:10.808936+0100: rigctld.c(660) Startup: rigctld -m 1035 -r /dev/ttyUSB1 -s 38400 -vvvvv -Z -R
2023-05-01T19:58:10.809244+0100: rigctld Hamlib 4.6~git Apr 29 22:10:25Z 2023 SHA=2b37e8 32-bit
2023-05-01T19:58:10.809305+0100: Report bugs to <hamlib-developer@lists.sourceforge.net>

2023-05-01T19:58:10.809371+0100: Max# of rigctld client services=32
2023-05-01T19:58:10.809441+0100: rig_check_rig_caps: p1=0xb6ec65b8, p2=0xb6ecbd04, rig_model=0xb6ec65b8, macro_name=0xb6ecbd04
2023-05-01T19:58:10.809689+0100: yaesu: initrigs4_yaesu called
2023-05-01T19:58:10.810028+0100: rig_init: rig_model=Yaesu FT-991 20230328.15
2023-05-01T19:58:10.810195+0100: rig_init: rig has VFO_A
2023-05-01T19:58:10.810255+0100: rig_init: rig has VFO_B
2023-05-01T19:58:10.810311+0100: rig_init: rig has VFO_MEM
2023-05-01T19:58:10.810424+0100: ft991_init called, version 20230328.15
2023-05-01T19:58:10.810551+0100:  1:newcat.c(466):newcat_init entered
2023-05-01T19:58:10.810643+0100:  1:newcat.c(507):newcat_init returning(0) 
2023-05-01T19:58:10.810777+0100: main: twiddle=0, uplink=0, twiddle_rit=0
2023-05-01T19:58:10.810904+0100:  1:rig.c(827):rig_open entered
2023-05-01T19:58:10.811055+0100: rig_settings_get_path: path=/home/pi/.config/hamlib_settings
2023-05-01T19:58:10.811309+0100: rig_settings_load_all: settings_file (/home/pi/.config/hamlib_settings): No such file or directory
2023-05-01T19:58:10.811408+0100: rig_open: cwd=/home/pi
2023-05-01T19:58:10.811570+0100: rig_open: /home/pi/.config//home/pi/hamlib_settings does not exist
2023-05-01T19:58:10.811645+0100: rig_open: async_data_enable=1, async_data_supported=0
2023-05-01T19:58:10.811744+0100: serial_open: /dev/ttyUSB1
2023-05-01T19:58:10.811864+0100: serial_open(229): open failed#1
2023-05-01T19:58:11.312051+0100: serial_open: Unable to open /dev/ttyUSB1 - No such file or directory
2023-05-01T19:58:11.312106+0100: port_open: serial_open(/dev/ttyUSB1) status=-6, err=No such file or directory
2023-05-01T19:58:11.312139+0100: rig_open: rs->comm_state==0?=0
2023-05-01T19:58:11.312195+0100:  1:rig.c(1033):rig_open returning(-6) IO error

Opened rig model 1035, 'FT-991'
rig_open: error = rig_init: rig has VFO_B
rig_init: rig has VFO_MEM
ft991_init called, version 20230328.15
 1:newcat.c(466):newcat_init entered
 1:newcat.c(507):newcat_init returning(0) 
main: twiddle=0, uplink=0, twiddle_rit=0
 1:rig.c(827):rig_open entered
rig_settings_get_path: path=/home/pi/.config/hamlib_settings
rig_settings_load_all: settings_file (/home/pi/.config/hamlib_settings): No such file or directory
rig_open: cwd=/home/pi
rig_open: /home/pi/.config//home/pi/hamlib_settings does not exist
rig_open: async_data_enable=1, async_data_supported=0
serial_open: /dev/ttyUSB1
serial_open(229): open failed#1
serial_open: Unable to open /dev/ttyUSB1 - No such file or directory
port_open: serial_open(/dev/ttyUSB1) status=-6, err=No such file or directory
rig_open: rs->comm_state==0?=0
 1:rig.c(1033):rig_open returning(-6) IO error

IO error
 /dev/ttyUSB1 No such file or directory 
2023-05-01T19:58:11.312295+0100: Backend version: 20230328.15, Status: Stable
2023-05-01T19:58:11.312432+0100:  1:rig.c(1439):rig_close entered
2023-05-01T19:58:11.312462+0100:  1:rig.c(1451):rig_close returning(-1) Invalid parameter

Closed rig model 1035, 'FT-991 - will reopen for clients'
2023-05-01T19:58:11.312879+0100: main: Using IPV4
2023-05-01T19:58:11.312919+0100:  1:network.c(977):network_multicast_publisher_start entered
2023-05-01T19:58:11.312949+0100: network.c(981):multicast address=0.0.0.0, port=4532
2023-05-01T19:58:11.312977+0100: network.c(986): not starting multicast publisher
2023-05-01T19:58:11.313057+0100: main: rigctld listening on port 4532
```

3. Power on 12V power supply (/dev/ttyUSB1 appears on the system).
4. Run rigctl with auto_power_on=1 option:

[... 751 lines cut ...]
2023-05-01T20:05:14.487621+0100:                   18:rig.c(7594):async_data_handler_stop returning(0) 
2023-05-01T20:05:14.487686+0100: ser_close: restoring options
2023-05-01T20:05:14.495509+0100: rig_close(1575): 0x118d134 rs->comm_state==0?=0
2023-05-01T20:05:14.495590+0100:                  17:rig.c(1577):rig_close returning(0) 
2023-05-01T20:05:14.495660+0100: mutex_rigctld: client lock disengaged
2023-05-01T20:05:14.495738+0100: handle_socket: rig_close retcode=0
2023-05-01T20:05:14.495785+0100: mutex_rigctld: client lock engaged
2023-05-01T20:05:14.495875+0100: rigctld.c(1120) trace
2023-05-01T20:05:14.495944+0100: 1 outstanding client(s)
2023-05-01T20:05:14.496016+0100:                  17:rig.c(1439):rig_close entered
2023-05-01T20:05:14.496090+0100:                  17:rig.c(1451):rig_close returning(-1) Invalid parameter

2023-05-01T20:05:14.496167+0100: rigctld.c(1128) trace
2023-05-01T20:05:14.496236+0100: mutex_rigctld: client lock disengaged
2023-05-01T20:05:14.496305+0100: rigctld.c(1130) trace
2023-05-01T20:05:14.496369+0100: rigctld.c(1135) trace
2023-05-01T20:05:14.496452+0100:                  17:network.c(1092):network_multicast_publisher_stop entered
2023-05-01T20:05:14.496522+0100:                  17:network.c(1101):network_multicast_publisher_stop returning(0) 
2023-05-01T20:05:14.496638+0100: rigctld.c(1138) trace
2023-05-01T20:05:14.496715+0100:                  17:newcat.c(521):newcat_cleanup entered
2023-05-01T20:05:14.496788+0100:                  17:newcat.c(530):newcat_cleanup returning(0) 
```

7. As you can see, rigctld receives the request from the client, convert the command get_power_stat, send it to the radio (Yaesu PS;) and receive the right answer (PS0;) but for some reason doesn't send the information to the client.
8. Rigctld continue to send commands to the radio in a kind of loop after the client is disconnected for the timeout, I have to stop it by forcing with Ctrl+C.

mdblack98, 2023-05-02:
You need to start rigctld with auto_power_on otherwise it will never try to auto start the rig.
You can also use the -Y switch on rigctl to have it continue even though it gets errors.
Mike W9MDB

mdblack98, 2023-05-03:
> Mike,
> this is the key point, I don't need the auto power on.
> I want that my client connects rigctld and (if the rig has get and set powerstat caps and if the rig is off) waits that the user presses the button power on the software GUI, in this case it sends power on to the rig and starts the polling.
> This was possible without errors up to version 4.5.3, if I remember well.

Try the latest master...it might work now.  Hopefully doesn't break any other rigs.  The problem was with Flex PowerSDR and SDR++ that don't have the PS command.

Mike W9MDB

PianetaRadio, 2023-05-04:
I done a fast test, but seems that the problem is not yet solved. Tomorrow I will send you more details.

About your worry for the rigs that haven't powerstat capability, the applications generally should check if the rig has get_powerstat, if not use another command (for example read the frequency) to understand if the rig is or not powered.

In my application if rig has powerstat capability and return get_powerstat 1, I do also a double check by asking the frequency and only if ok start the polling of other variables.

PianetaRadio, 2023-05-05:
Mike, issue not yet solved.

1) Launch rigctld with radio power supply off, it works as usual:

```
rigctld -m 1035 -r /dev/ttyUSB1 -s 38400 -vvvvv -Z -R
2023-05-05T19:25:53.432572+0100: rigctld.c(660) Startup: rigctld -m 1035 -r /dev/ttyUSB1 -s 38400 -vvvvv -Z -R
2023-05-05T19:25:53.432946+0100: rigctld Hamlib 4.6~git May 04 12:20:16Z 2023 SHA=1ab828 32-bit
2023-05-05T19:25:53.433069+0100: Report bugs to <hamlib-developer@lists.sourceforge.net>

2023-05-05T19:25:53.433226+0100: Max# of rigctld client services=32
2023-05-05T19:25:53.433361+0100: rig_check_rig_caps: p1=0xb6e915c0, p2=0xb6e96d0c, rig_model=0xb6e915c0, macro_name=0xb6e96d0c
2023-05-05T19:25:53.433761+0100: yaesu: initrigs4_yaesu called
2023-05-05T19:25:53.434119+0100: rig_init: rig_model=Yaesu FT-991 20230328.15
2023-05-05T19:25:53.434350+0100: rig_init: rig has VFO_A
2023-05-05T19:25:53.434473+0100: rig_init: rig has VFO_B
2023-05-05T19:25:53.434590+0100: rig_init: rig has VFO_MEM
2023-05-05T19:25:53.434839+0100: ft991_init called, version 20230328.15
2023-05-05T19:25:53.435103+0100:  1:newcat.c(466):newcat_init entered
2023-05-05T19:25:53.435259+0100:  1:newcat.c(507):newcat_init returning(0) 
2023-05-05T19:25:53.435510+0100: main: twiddle=0, uplink=0, twiddle_rit=0
2023-05-05T19:25:53.435763+0100:  1:rig.c(827):rig_open entered
2023-05-05T19:25:53.436033+0100: rig_settings_get_path: path=/home/pi/.config/hamlib_settings
2023-05-05T19:25:53.436334+0100: rig_settings_load_all: settings_file (/home/pi/.config/hamlib_settings): No such file or directory
2023-05-05T19:25:53.436494+0100: rig_open: cwd=/home/pi
2023-05-05T19:25:53.436729+0100: rig_open: /home/pi/.config//home/pi/hamlib_settings does not exist
2023-05-05T19:25:53.436869+0100: rig_open: async_data_enable=1, async_data_supported=0
2023-05-05T19:25:53.437029+0100: serial_open: /dev/ttyUSB1
2023-05-05T19:25:53.437215+0100: serial_open(229): open failed#1
2023-05-05T19:25:53.937535+0100: serial_open: Unable to open /dev/ttyUSB1 - No such file or directory
2023-05-05T19:25:53.937590+0100: port_open: serial_open(/dev/ttyUSB1) status=-6, err=No such file or directory
2023-05-05T19:25:53.937623+0100: rig_open: rs->comm_state==0?=0
2023-05-05T19:25:53.937680+0100:  1:rig.c(1033):rig_open returning(-6) IO error

Opened rig model 1035, 'FT-991'
rig_open: error = rig_init: rig has VFO_B
rig_init: rig has VFO_MEM
ft991_init called, version 20230328.15
 1:newcat.c(466):newcat_init entered
 1:newcat.c(507):newcat_init returning(0) 
main: twiddle=0, uplink=0, twiddle_rit=0
 1:rig.c(827):rig_open entered
rig_settings_get_path: path=/home/pi/.config/hamlib_settings
rig_settings_load_all: settings_file (/home/pi/.config/hamlib_settings): No such file or directory
rig_open: cwd=/home/pi
rig_open: /home/pi/.config//home/pi/hamlib_settings does not exist
rig_open: async_data_enable=1, async_data_supported=0
serial_open: /dev/ttyUSB1
serial_open(229): open failed#1
serial_open: Unable to open /dev/ttyUSB1 - No such file or directory
port_open: serial_open(/dev/ttyUSB1) status=-6, err=No such file or directory
rig_open: rs->comm_state==0?=0
 1:rig.c(1033):rig_open returning(-6) IO error

IO error
 /dev/ttyUSB1 No such file or directory 
2023-05-05T19:25:53.937781+0100: Backend version: 20230328.15, Status: Stable
2023-05-05T19:25:53.938227+0100:  1:rig.c(1439):rig_close entered
2023-05-05T19:25:53.938282+0100:  1:rig.c(1451):rig_close returning(-1) Invalid parameter

Closed rig model 1035, 'FT-991 - will reopen for clients'
2023-05-05T19:25:53.938740+0100: main: Using IPV4
2023-05-05T19:25:53.938808+0100:  1:network.c(977):network_multicast_publisher_start entered
2023-05-05T19:25:53.938860+0100: network.c(981):multicast address=0.0.0.0, port=4532
2023-05-05T19:25:53.938911+0100: network.c(986): not starting multicast publisher
2023-05-05T19:25:53.939018+0100: main: rigctld listening on port 4532

```

2) Power on the power supply, radio is off.
[... 858 lines cut ...]
2023-05-05T19:28:24.283402+0100: rigctl_parse: vfo_opt=0
2023-05-05T19:28:24.283489+0100:             12:rigctl_parse.c(4370):rigctl_dump_state entered
2023-05-05T19:28:24.284068+0100:             12:rigctl_parse.c(4534):rigctl_dump_state returning(0) 
2023-05-05T19:28:24.284429+0100: mutex_rigctld: client lock disengaged
2023-05-05T19:28:24.284561+0100: mutex_rigctld: client lock engaged
2023-05-05T19:28:24.284707+0100: mutex_rigctld: client lock disengaged
2023-05-05T19:28:24.284786+0100: handle_socket: doing rigctl_parse vfo_mode=0, secure=0
2023-05-05T19:28:24.284862+0100: rigctl_parse: called, interactive=1
2023-05-05T19:28:24.286271+0100: fscanf: parsing '' with '%c'
2023-05-05T19:28:24.286549+0100: scanfc: ret=-1
2023-05-05T19:28:24.286665+0100: scanfc: errno=104, Connection reset by peer
2023-05-05T19:28:24.286750+0100: rigctl_parse: nothing to scan#1? retcode=-1, last_cmd=
2023-05-05T19:28:24.286859+0100: handle_socket: rigctl_parse retcode=2
2023-05-05T19:28:24.286959+0100:             12:rig.c(1439):rig_close entered
2023-05-05T19:28:24.287069+0100:              13:newcat.c(688):newcat_close entered
2023-05-05T19:28:24.287182+0100:              13:newcat.c(717):newcat_close returning(0) 
2023-05-05T19:28:24.287297+0100:              13:rig.c(7564):async_data_handler_stop entered
2023-05-05T19:28:24.287394+0100:              13:rig.c(7594):async_data_handler_stop returning(0) 
2023-05-05T19:28:24.287500+0100: ser_close: restoring options
2023-05-05T19:28:24.296503+0100: rig_close(1575): 0x10ff134 rs->comm_state==0?=0
2023-05-05T19:28:24.296613+0100:             12:rig.c(1577):rig_close returning(0) 
Closed rig model FT-991.  Will reopen for new clients
2023-05-05T19:28:24.296724+0100: Connection closed from 127.0.0.1:45212

```

PianetaRadio, 2023-05-05:
Now I do the same with my application CatRadio.

Log from rigctld

```
2023-05-05T19:40:27.957074+0100: Connection opened from 127.0.0.1:45218
2023-05-05T19:40:27.957525+0100: mutex_rigctld: client lock engaged
2023-05-05T19:40:27.957697+0100: mutex_rigctld: client lock disengaged
2023-05-05T19:40:27.957750+0100: mutex_rigctld: client lock engaged
2023-05-05T19:40:27.957807+0100:   2:rig.c(6110):rig_get_powerstat entered
2023-05-05T19:40:27.957859+0100:   2:rig.c(6115):rig_get_powerstat returning(-1) Invalid parameter

2023-05-05T19:40:27.957911+0100: mutex_rigctld: client lock disengaged
2023-05-05T19:40:27.957957+0100: mutex_rigctld: client lock engaged
2023-05-05T19:40:27.958003+0100:   2:rig.c(827):rig_open entered
2023-05-05T19:40:27.958062+0100: rig_settings_get_path: path=/home/pi/.config/hamlib_settings
2023-05-05T19:40:27.958205+0100: rig_settings_load_all: settings_file (/home/pi/.config/hamlib_settings): No such file or directory
2023-05-05T19:40:27.958280+0100: rig_open: cwd=/home/pi
2023-05-05T19:40:27.958371+0100: rig_open: /home/pi/.config//home/pi/hamlib_settings does not exist
2023-05-05T19:40:27.958421+0100: rig_open: async_data_enable=0, async_data_supported=0
2023-05-05T19:40:27.958473+0100: serial_open: /dev/ttyUSB1
2023-05-05T19:40:27.964116+0100: serial_setup: tcgetattr
2023-05-05T19:40:27.964186+0100: serial_setup: cfmakeraw
2023-05-05T19:40:27.964238+0100: serial_setup: cfsetispeed=38400,0x000f
2023-05-05T19:40:27.964292+0100: serial_setup: cfsetospeed=38400,0x000f
2023-05-05T19:40:27.964342+0100: serial_setup: data_bits=8
2023-05-05T19:40:27.964390+0100: serial_setup: parity=0
2023-05-05T19:40:27.964437+0100: serial_setup: Handshake=Hardware
2023-05-05T19:40:27.964492+0100: serial_setup: tcsetattr TCSANOW
2023-05-05T19:40:27.969693+0100: read_string_generic called, rxmax=4095 direct=1, expected_len=1
2023-05-05T19:40:27.970880+0100: read_string_generic: flag=0
2023-05-05T19:40:28.022159+0100: read_string_generic: flag=1
2023-05-05T19:40:28.022269+0100: tcflush
2023-05-05T19:40:28.072478+0100:    3:rig.c(7520):async_data_handler_start entered
2023-05-05T19:40:28.072604+0100: async_data_handler_start: async data support disabled since async_data_enabled=0
2023-05-05T19:40:28.072675+0100:    3:rig.c(7527):async_data_handler_start returning(0) 
2023-05-05T19:40:28.072748+0100: rig.c(262):add_opened_rig returning2(0) 
2023-05-05T19:40:28.072809+0100: rig_open: 0x1423134 rs->comm_state==1?=1
2023-05-05T19:40:28.072885+0100:    3:rig.c(6110):rig_get_powerstat entered
2023-05-05T19:40:28.072944+0100: rig.c(6130) trace
2023-05-05T19:40:28.073067+0100:     4:newcat.c(3623):newcat_get_powerstat entered
2023-05-05T19:40:28.073132+0100: newcat_valid_command PS
2023-05-05T19:40:28.073196+0100: newcat.c(7668):newcat_valid_command returning2(1) 
2023-05-05T19:40:28.073318+0100: write_block(): TX 3 bytes, method=2
2023-05-05T19:40:28.073385+0100: 0000    50 53 3b                                            PS;             
2023-05-05T19:40:29.275747+0100:      5:newcat.c(10492):newcat_get_cmd entered
2023-05-05T19:40:29.275862+0100: rig_flush: called for serial device
2023-05-05T19:40:29.275938+0100: read_string_generic called, rxmax=4095 direct=1, expected_len=1
2023-05-05T19:40:29.277268+0100: read_string_generic: flag=0
2023-05-05T19:40:29.328559+0100: read_string_generic: flag=1
2023-05-05T19:40:29.328681+0100: 0000    50 53 30 3b                                         PS0;            
2023-05-05T19:40:29.328743+0100: tcflush
2023-05-05T19:40:29.328815+0100: cmd_str = PS;
2023-05-05T19:40:29.328920+0100: write_block(): TX 3 bytes, method=2
2023-05-05T19:40:29.328990+0100: 0000    50 53 3b                                            PS;             
2023-05-05T19:40:29.331136+0100: read_string_generic called, rxmax=129 direct=1, expected_len=1
2023-05-05T19:40:29.331595+0100: read_string_generic(): RX 4 characters, direct=1
2023-05-05T19:40:29.331661+0100: 0000    50 53 30 3b                                         PS0;            
2023-05-05T19:40:29.331720+0100: newcat_get_cmd: read count = 4, ret_data = PS0;
2023-05-05T19:40:29.331783+0100:      5:newcat.c(10710):newcat_get_cmd returning(0) 
2023-05-05T19:40:29.331846+0100:     4:newcat.c(3661):newcat_get_powerstat returning(0) 
2023-05-05T19:40:29.331909+0100:    3:rig.c(6144):rig_get_powerstat returning(0) 
2023-05-05T19:40:29.331971+0100: rig_open: rig power is off, use --set-conf=auto_power_on=1 if power on is wanted
2023-05-05T19:40:29.332034+0100: handle_socket: rig_open reopened retcode=-20
2023-05-05T19:40:29.332092+0100: mutex_rigctld: client lock disengaged
2023-05-05T19:40:29.332150+0100: handle_socket: i/o error
2023-05-05T19:40:29.332209+0100: mutex_rigctld: client lock engaged
2023-05-05T19:40:29.332270+0100:    3:rig.c(1439):rig_close entered
2023-05-05T19:40:29.332429+0100:     4:newcat.c(688):newcat_close entered
2023-05-05T19:40:29.332503+0100:      5:newcat.c(7170):newcat_set_trn entered
[... 751 lines cut ...]
  2:rig.c(6144):rig_get_powerstat returning(0) 
  2:netrigctl.c(257):netrigctl_open entered
netrigctl_transaction: called len=9
rig_flush: called for network device
network_flush called
write_block(): TX 9 bytes, method=2
read_string_generic called, rxmax=1024 direct=1, expected_len=1
read_string_generic: flag=0
read_string_generic: flag=1
read_string_generic(): Timed out 20.057 seconds after 0 chars, direct=1
Communication timed out
Communication timed out

[... cut ...]
2023-05-05T19:41:08.082124+0100: network_flush called
2023-05-05T19:41:18.092537+0100: read_string_generic: flag=0
2023-05-05T19:41:27.834217+0100:   2:netrigctl.c(346):netrigctl_open returning(-8) Protocol error

2023-05-05T19:41:27.834433+0100:   2:rig.c(7564):async_data_handler_stop entered
2023-05-05T19:41:27.834527+0100:   2:rig.c(7594):async_data_handler_stop returning(0) 
2023-05-05T19:41:27.837568+0100: network_close: close socket ret=0
2023-05-05T19:41:27.837982+0100:  1:rig.c(1315):rig_open returning(-8) Protocol error

```

PianetaRadio, 2023-05-12:
Test as before:
1) Launch rigctld with radio power supply off;
2) Radio power supply on;
3) Launch rigctl:

```
rigctl -m 2 -vvvvv -Z
2023-05-12T19:58:56.710187+0100: rigctl.c(442) Startup: rigctl -m 2 -vvvvv -Z
2023-05-12T19:58:56.710416+0100: rigctl Hamlib 4.6~git May 11 21:36:12Z 2023 SHA=dfc5e7 32-bit
2023-05-12T19:58:56.710465+0100: Report bugs to <hamlib-developer@lists.sourceforge.net>

2023-05-12T19:58:56.710527+0100: rig_check_rig_caps: p1=0xb6e795e0, p2=0xb6e7ed2c, rig_model=0xb6e795e0, macro_name=0xb6e7ed2c
2023-05-12T19:58:56.710691+0100: initrigs4_dummy: _init called
2023-05-12T19:58:56.710790+0100: rig_init: rig_model=Hamlib NET rigctl 20230503.0
2023-05-12T19:58:56.710924+0100: rig_init: rx_range_list1 is empty, using rx_range_list2
2023-05-12T19:58:56.710975+0100: rig_init: rig does not have tx_range!!
2023-05-12T19:58:56.711083+0100: netrigctl_init version 20230503.0
2023-05-12T19:58:56.711199+0100:  1:rig.c(830):rig_open entered
2023-05-12T19:58:56.711326+0100: rig_settings_get_path: path=/home/pi/.config/hamlib_settings
2023-05-12T19:58:56.711508+0100: rig_settings_load_all: settings_file (/home/pi/.config/hamlib_settings): No such file or directory
2023-05-12T19:58:56.711596+0100: rig_open: cwd=/home/pi
2023-05-12T19:58:56.711754+0100: rig_open: /home/pi/.config//home/pi/hamlib_settings does not exist
2023-05-12T19:58:56.711826+0100: rig_open: async_data_enable=1, async_data_supported=0
2023-05-12T19:58:56.711951+0100: rig_open: using network address 127.0.0.1:4532
2023-05-12T19:58:56.712092+0100: network_open: hoststr=127.0.0.1, portstr=4532
2023-05-12T19:58:56.712720+0100: network_open: client port=45232
2023-05-12T19:58:56.712797+0100:   2:rig.c(7529):async_data_handler_start entered
2023-05-12T19:58:56.712858+0100: async_data_handler_start: async data support disabled since async_data_enabled=0
2023-05-12T19:58:56.712998+0100:   2:rig.c(7536):async_data_handler_start returning(0) 
2023-05-12T19:58:56.713074+0100: rig.c(262):add_opened_rig returning2(0) 
2023-05-12T19:58:56.713140+0100: rig_open: 0x97d134 rs->comm_state==1?=1
2023-05-12T19:58:56.713216+0100:   2:rig.c(6119):rig_get_powerstat entered
2023-05-12T19:58:56.713279+0100: rig.c(6139) trace
2023-05-12T19:58:56.713348+0100: netrigctl_get_powerstat called
2023-05-12T19:58:56.713413+0100: netrigctl_transaction: called len=15
2023-05-12T19:58:56.713486+0100: rig_flush: called for network device
2023-05-12T19:58:56.713555+0100: network_flush called
2023-05-12T19:58:56.713842+0100: write_block(): TX 15 bytes, method=2
2023-05-12T19:58:56.713923+0100: 0000    5c 67 65 74 5f 70 6f 77 65 72 73 74 61 74 0a        \get_powerstat. 
2023-05-12T19:58:56.764357+0100: netrigctl_get_powerstat: PS command failed (ret=-5) so returning RIG_POWER_ON
2023-05-12T19:58:56.764453+0100:   2:rig.c(6153):rig_get_powerstat returning(0) 
2023-05-12T19:58:56.764520+0100:   2:netrigctl.c(257):netrigctl_open entered
2023-05-12T19:58:56.764582+0100: netrigctl_transaction: called len=9
2023-05-12T19:58:56.764693+0100: rig_flush: called for network device
2023-05-12T19:58:56.764759+0100: network_flush called
2023-05-12T19:58:56.764996+0100: write_block(): TX 9 bytes, method=2
2023-05-12T19:58:56.765069+0100: 0000    5c 63 68 6b 5f 76 66 6f 0a                          \chk_vfo.       
2023-05-12T19:58:56.815357+0100: netrigctl_open: chk_vfo error: write_block(): TX 15 bytes, method=2
0000    5c 67 65 74 5f 70 6f 77 65 72 73 74 61 74 0a        \get_powerstat. 
read_string_generic called, rxmax=1024 direct=1, expected_len=1
read_string_generic: flag=0
read_string_generic: flag=1
read_string_generic(): Timed out 0.050 seconds after 0 chars, direct=1
netrigctl_get_powerstat: PS command failed (ret=-5) so returning RIG_POWER_ON
  2:rig.c(6153):rig_get_powerstat returning(0) 
  2:netrigctl.c(257):netrigctl_open entered
netrigctl_transaction: called len=9
rig_flush: called for network device
network_flush called
write_block(): TX 9 bytes, method=2
0000    5c 63 68 6b 5f 76 66 6f 0a                          \chk_vfo.       
read_string_generic called, rxmax=1024 direct=1, expected_len=1
read_string_generic: flag=0
read_string_generic: flag=1
read_string_generic(): Timed out 0.050 seconds after 0 chars, direct=1
Communication timed out
Communication timed out

2023-05-12T19:58:56.815777+0100: netrigctl_open: vfo_mode=0
2023-05-12T19:58:56.815849+0100: netrigctl_transaction: called len=12
[... 507 lines cut ...]
2023-05-12T19:59:28.581415+0100: rigctl_parse: cmd=dump_state handle=4
2023-05-12T19:59:28.581482+0100: mutex_rigctld: client lock engaged
2023-05-12T19:59:28.581547+0100: rigctl(d): � 'currVFO' '' '' ''
2023-05-12T19:59:28.581612+0100: rigctl_parse: vfo_opt=0
2023-05-12T19:59:28.581679+0100:       6:rigctl_parse.c(4372):rigctl_dump_state entered
2023-05-12T19:59:28.582106+0100:       6:rigctl_parse.c(4536):rigctl_dump_state returning(0) 
2023-05-12T19:59:28.582199+0100: mutex_rigctld: client lock disengaged
2023-05-12T19:59:28.582260+0100: mutex_rigctld: client lock engaged
2023-05-12T19:59:28.582318+0100: mutex_rigctld: client lock disengaged
2023-05-12T19:59:28.582377+0100: handle_socket: doing rigctl_parse vfo_mode=0, secure=0
2023-05-12T19:59:28.582436+0100: rigctl_parse: called, interactive=1
2023-05-12T19:59:28.582517+0100: scanfc: ret=-1
2023-05-12T19:59:28.582576+0100: rigctl_parse: nothing to scan#1? retcode=-1, last_cmd=
2023-05-12T19:59:28.582634+0100: handle_socket: rigctl_parse retcode=2
[... cut ...]
2023-05-12T19:59:28.582877+0100:        7:rig.c(7573):async_data_handler_stop entered
2023-05-12T19:59:28.582940+0100:        7:rig.c(7603):async_data_handler_stop returning(0) 
2023-05-12T19:59:28.583003+0100: ser_close: restoring options
2023-05-12T19:59:28.590992+0100: rig_close(1581): 0x1edb134 rs->comm_state==0?=0
2023-05-12T19:59:28.591137+0100:       6:rig.c(1583):rig_close returning(0) 
Closed rig model FT-991.  Will reopen for new clients
2023-05-12T19:59:28.591350+0100: Connection closed from 127.0.0.1:45232
```

PianetaRadio, 2023-05-21:
Well, now communication doesn't  go in time out and rigctl doesn't fail.
But there is a strange behaviour, if I send a power off command, after a while the radio power on again.

More details will follow.

mikaelnousiainen, 2023-05-21:
@PianetaRadio There were at least a couple of regressions with get/set_powerstat causing rigs to automatically power on after power off and such. Additionally rigctl(d) wouldn't start if rig was powered off -- which has been possible always. Please try my fork/branch here, I will later merge my changes after I test with a couple rigs more (incl. TS-480).

https://github.com/mikaelnousiainen/Hamlib/tree/power-status-bug-fixes

mdblack98, 2023-05-22:
https://n0nb.users.sourceforge.net/

Mike W9MDB

Mikael, could you provide a tarball of your branch to compile on my Raspberry Pi?

TNX
Johnny

mikaelnousiainen, 2023-05-23:
I think @PianetaRadio meant he wanted to test my changes (yeah, we have similar names, Michael vs Mikael :joy:).

I don't really have an easy way to provide tarballs, but can you just try to install Git on your Raspberry Pi and then clone:

```
git clone https://github.com/mikaelnousiainen/Hamlib
git checkout power-status-bug-fixes
```

After that you can compile as usual:

```
./bootstrap && ./configure && make -j4
```

PianetaRadio, 2023-05-23:
@mikaelnousiainen thank you, I was missing the ./bootstrap command that is not used when I compile from tarball archive.

First test with FT-991A and my software CatRadio passing trough rigctld. Power off now is ok.
But when I connect, the radio power on automatically (option auto_power_on=0).

Rigctl -m 2 instead do not detect properly the power status (always off) and also causes automatic power on when launched.

```
2023-05-23T21:19:33.508806+0100: rigctl.c(442) Startup: rigctl -m 2 -vvvvv -Z
2023-05-23T21:19:33.509092+0100: rigctl Hamlib 4.6~git May 21 15:15:22Z 2023 SHA=52358d 32-bit
2023-05-23T21:19:33.509144+0100: Report bugs to <hamlib-developer@lists.sourceforge.net>

2023-05-23T21:19:33.509251+0100: rig_check_rig_caps: p1=0xb6eee6b0, p2=0xb6ef3dfc, rig_model=0xb6eee6b0, macro_name=0xb6ef3dfc
2023-05-23T21:19:33.509417+0100: initrigs4_dummy: _init called
2023-05-23T21:19:33.509501+0100: rig_init: rig_model=Hamlib NET rigctl 20230503.0
2023-05-23T21:19:33.509609+0100: rig_init: rx_range_list1 is empty, using rx_range_list2
2023-05-23T21:19:33.509654+0100: rig_init: rig does not have tx_range!!
2023-05-23T21:19:33.509699+0100: netrigctl_init version 20230503.0
2023-05-23T21:19:33.509788+0100: rig.c(835):rig_open entered
2023-05-23T21:19:33.509896+0100: rig_settings_get_path: path=/home/pi/.config/hamlib_settings
2023-05-23T21:19:33.510043+0100: rig_settings_load_all: settings_file (/home/pi/.config/hamlib_settings): No such file or directory
2023-05-23T21:19:33.510124+0100: rig_open: cwd=/home/pi/Hamlib
2023-05-23T21:19:33.510223+0100: rig_open: /home/pi/.config//home/pi/Hamlib/hamlib_settings does not exist
2023-05-23T21:19:33.510272+0100: rig_open: async_data_enable=1, async_data_supported=0
2023-05-23T21:19:33.510348+0100: rig_open: using network address 127.0.0.1:4532
2023-05-23T21:19:33.510402+0100: network_open: hoststr=127.0.0.1, portstr=4532
2023-05-23T21:19:33.510765+0100: network_open: client port=16049
2023-05-23T21:19:33.510817+0100:  1:rig.c(7638):async_data_handler_start entered
2023-05-23T21:19:33.510856+0100: async_data_handler_start: async data support disabled since async_data_enabled=0
2023-05-23T21:19:33.510947+0100:  1:rig.c(7645):async_data_handler_start returning(0) 
2023-05-23T21:19:33.510995+0100: rig.c(262):add_opened_rig returning2(0) 
2023-05-23T21:19:33.511034+0100: rig_open: 0xf46134 rs->comm_state==1?=1
2023-05-23T21:19:33.611503+0100:  1:rig.c(6199):rig_get_powerstat entered
2023-05-23T21:19:33.611747+0100: rig.c(6219) trace
2023-05-23T21:19:33.611901+0100: netrigctl_get_powerstat called
2023-05-23T21:19:33.612055+0100: netrigctl_transaction: called len=15
2023-05-23T21:19:33.612212+0100: rig_flush_force: called for network device
2023-05-23T21:19:33.612367+0100: network_flush called
2023-05-23T21:19:33.612699+0100: write_block(): TX 15 bytes, method=2
2023-05-23T21:19:33.612891+0100: 0000    5c 67 65 74 5f 70 6f 77 65 72 73 74 61 74 0a        \get_powerstat. 
2023-05-23T21:19:40.324135+0100: read_string_generic(): RX 2 characters, direct=1
2023-05-23T21:19:40.324192+0100: 0000    30 0a                                               0.              
2023-05-23T21:19:40.324232+0100:  1:rig.c(6233):rig_get_powerstat returning(0) 
2023-05-23T21:19:40.324266+0100: rig_open: rig power is off, use --set-conf=auto_power_on=1 if power on is wanted
2023-05-23T21:19:40.324315+0100: rig.c(1300):rig_open returning2(-20) Rig is not powered on

```

mikaelnousiainen, 2023-05-23:
@PianetaRadio Great that you got the code compiled.

Sorry, but I do not understand what the problem is. By default -- and based on your logs -- rigctld does NOT power on the rig automatically. The rig remains powered off until you execute command `\set_powerstat1`. This is what I fixed too. Your command did _not_ specify `--set-conf=auto_power_on=1` so that's the correct behavior.

Can you specify in more detail what is going wrong?

For me, Yaesu FT-991A power ON/OFF works perfectly now. I'm using rigctld daemon, but the client is my own (not using another rigctl instance to connect to it).

mdblack98, 2023-05-23:
He does not want it to auto power on.  He wants it to remain off until set_power_stat 1 is called.

I had this working correctly...the only thing left was the problem about turning itself on after a while.

Mike W9MDB

mikaelnousiainen, 2023-05-23:
This is at least exactly how my branch works. I've tested it on multiple Yaesu rigs.

I'll supply a PR of the branch, as it fixes a couple of important issues, such as being able to open rigctld when the rig is powered off (this is how Hamlib has always been working).

PianetaRadio, 2023-05-24:
1) Sometimes when I launch my application or rigctl (both with auto-power-on=0) the radio power on automatically, but hamlib is still reporting RIG_POWER_OFF.

2) Often rigctld returns error from get_powerstat and RIG_POWER_ON is assumed, causing problem to my software GUI:
```
2023-05-24T19:42:15.930263+0100: initrigs4_dummy: _init called
2023-05-24T19:42:15.931219+0100: rig_init: rig_model=Hamlib NET rigctl 20230503.0
2023-05-24T19:42:15.931559+0100: rig_init: rig does not have tx_range!!
2023-05-24T19:42:15.931972+0100: rig.c(835):rig_open entered
2023-05-24T19:42:15.932289+0100: rig_settings_load_all: settings_file (/home/pi/.config/hamlib_settings): No such file or directory
2023-05-24T19:42:15.932377+0100: rig_open: cwd=/home/pi/CatRadio
2023-05-24T19:42:15.932511+0100: rig_open: /home/pi/.config//home/pi/CatRadio/hamlib_settings does not exist
2023-05-24T19:42:15.933194+0100:  1:rig.c(7638):async_data_handler_start entered
2023-05-24T19:42:15.933368+0100:  1:rig.c(7645):async_data_handler_start returning(0) 
2023-05-24T19:42:15.933446+0100: rig.c(262):add_opened_rig returning2(0) 
2023-05-24T19:42:15.933507+0100: rig_open: 0x1daeabc rs->comm_state==1?=1
2023-05-24T19:42:16.033818+0100:  1:rig.c(6199):rig_get_powerstat entered
2023-05-24T19:42:16.033965+0100: netrigctl_get_powerstat called
2023-05-24T19:42:16.034016+0100: netrigctl_transaction: called len=15
2023-05-24T19:42:16.034093+0100: network_flush called
2023-05-24T19:42:16.107046+0100: netrigctl_get_powerstat: PS command failed (ret=-5) so returning RIG_POWER_ON
2023-05-24T19:42:16.107196+0100:  1:rig.c(6233):rig_get_powerstat returning(0) 
```
In my GUI I done a double check, after receiving RIG_POWER_ON, by reading the VFO frequency, but It seems that in this case Hamlib do not return an error but a frequency reading =0.

3) Sometimes, after a rig_close -> rig_open, I have connection timeout and it is solved only by closing and reloading rigctld:
```
2023-05-24T19:34:43.943732+0100: initrigs4_dummy: _init called
2023-05-24T19:34:43.944772+0100: rig_init: rig_model=Hamlib NET rigctl 20230503.0
2023-05-24T19:34:43.944936+0100: rig_init: rig does not have tx_range!!
2023-05-24T19:34:43.945101+0100: rig.c(835):rig_open entered
2023-05-24T19:34:43.945338+0100: rig_settings_load_all: settings_file (/home/pi/.config/hamlib_settings): No such file or directory
2023-05-24T19:34:43.945454+0100: rig_open: cwd=/home/pi/CatRadio
2023-05-24T19:34:43.945588+0100: rig_open: /home/pi/.config//home/pi/CatRadio/hamlib_settings does not exist
2023-05-24T19:34:43.946255+0100:  1:rig.c(7638):async_data_handler_start entered
2023-05-24T19:34:43.946441+0100:  1:rig.c(7645):async_data_handler_start returning(0) 
2023-05-24T19:34:43.946518+0100: rig.c(262):add_opened_rig returning2(0) 
2023-05-24T19:34:43.946583+0100: rig_open: 0x94aa74 rs->comm_state==1?=1
2023-05-24T19:34:44.046906+0100:  1:rig.c(6199):rig_get_powerstat entered
2023-05-24T19:34:44.047065+0100: netrigctl_get_powerstat called
2023-05-24T19:34:44.047115+0100: netrigctl_transaction: called len=15
2023-05-24T19:34:44.047193+0100: network_flush called
2023-05-24T19:34:54.057562+0100: netrigctl_get_powerstat: PS command failed (ret=-5) so returning RIG_POWER_ON
2023-05-24T19:34:54.057699+0100:  1:rig.c(6233):rig_get_powerstat returning(0) 
2023-05-24T19:34:54.057821+0100:  1:netrigctl.c(257):netrigctl_open entered
2023-05-24T19:34:54.057877+0100: netrigctl_transaction: called len=9
2023-05-24T19:34:54.057952+0100: network_flush called
2023-05-24T19:34:57.628696+0100: netrigctl_open: chk_vfo error:  1:rig.c(6199):rig_get_powerstat entered
rig.c(6219) trace
netrigctl_get_powerstat called
netrigctl_transaction: called len=15
rig_flush_force: called for network device
network_flush called
write_block(): TX 15 bytes, method=2
read_string_generic called, rxmax=1024 direct=1, expected_len=1
read_string_generic(): Timed out 10.010 seconds after 0 chars, direct=1
netrigctl_get_powerstat: PS command failed (ret=-5) so returning RIG_POWER_ON
 1:rig.c(6233):rig_get_powerstat returning(0) 
 1:netrigctl.c(257):netrigctl_open entered
netrigctl_transaction: called len=9
rig_flush_force: called for network device
network_flush called
write_block(): TX 9 bytes, method=2
read_string_generic called, rxmax=1024 direct=1, expected_len=1
read_string_generic(): RX 8 characters, direct=1
Communication timed out
Communication timed out

2023-05-24T19:34:57.628822+0100: netrigctl_open: vfo_mode=0
2023-05-24T19:34:57.628919+0100: netrigctl_transaction: called len=12
2023-05-24T19:34:57.628998+0100: network_flush called
2023-05-24T19:35:07.639360+0100:  1:netrigctl.c(295):netrigctl_open returning(-5) Communication timed out

2023-05-24T19:35:07.639497+0100:  1:rig.c(7682):async_data_handler_stop entered
2023-05-24T19:35:07.639562+0100:  1:rig.c(7715):async_data_handler_stop returning(0) 
2023-05-24T19:35:07.639776+0100: network_close: close socket ret=0
2023-05-24T19:35:07.639854+0100: rig.c(1328):rig_open returning2(-5) Communication timed out

```

mdblack98, 2023-05-24:
Are you using Mikael's fork or the Hamlib master?

 
 

   
   -

PianetaRadio, 2023-05-25:
@mdblack98 I'm using Mikael's fork.

mikaelnousiainen, 2023-05-25:
@PianetaRadio Is your rigctl client version the same as the rigctld daemon? Both have an effect on how this works. You need to run both from my fork to test this properly.

PianetaRadio, 2023-05-25:
> @PianetaRadio Is your rigctl client version the same as the rigctld daemon? Both have an effect on how this works. You need to run both from my fork to test this properly.

Yes, I confirm.

mikaelnousiainen, 2023-05-29:
@PianetaRadio @mdblack98 I guess this issue can be closed, as it was fixed by #1302 ? I will work on further improvements on power status stuff in another PR, but it's no longer related to this issue.

closed 2023-05-29 by mdblack98

PianetaRadio, 2023-05-29:
Ok