Hamlib/Hamlib #1285: TS-890 VFO A/B switch not recognised by FreeDV with PR #385

Tyrbiter, 2023-05-06:
This change has been added to freedv-gui, apparently to fix a problem with another radio, see here:

https://github.com/drowe67/freedv-gui/pull/385

The change to how the VFOs are selected in freedv now means that VFOcurr is not updated by Hamlib's TS-890 code after a VFO swap (either A to B or B to A) unless the freedv modem is stopped and restarted, freedv and hamlib debug output is shown in the link above.

mdblack98, 2023-05-06:
I don't see a set_vfo being called anywhere -- are you manually swapping VFOs?

 
 

drowe67/freedv-gui#385

Tyrbiter, 2023-05-06:
Yes, freedv doesn't change the VFO itself, it simply uses what the radio has set. It checks the frequency every 5 seconds when the modem is running.

The code change that has caused this new behaviour is here:

https://github.com/drowe67/freedv-gui/pull/385/commits/7efa24529ee0ff500277e49a464a35b4f16779c1

mdblack98, 2023-05-06:
The rig_get_vfo was removed.  That's the only way to see what the current vfo actually is.

closed 2023-05-06 by mdblack98

Tyrbiter, 2023-05-06:
OK Mike, you're right, I will pass that back and see if we can't come up with something that works for everyone.