Hamlib/Hamlib #1286: FT-847 always defaults to main VFO

kconkas, 2023-05-11:
I am trying to use FT-847 (serial number starting with 8K, hence not one of those early "problematic" CAT models) in satellite mode with gpredict. The issue I am experiencing is uplink and downlink frequencies periodically alternate in the main VFO, without any changes whatsoever to the sub VFO.

To troubleshoot this, I tried with rigctl only and it turns out that it returns only the main VFO irrespective of main or sub VFO requested. For example, for my rig in satellite mode with main (RX) VFO set to 435.900 MHz and sub VFO set to 145.900 MHz I get the following:
```
$ rigctl -vvvv -m 1001 -r /dev/ttyUSB0 -s 57600
rigctl.c(442) Startup: rigctl -vvvv -m 1001 -r /dev/ttyUSB0 -s 57600
rigctl Hamlib 4.6~git Mar 14 22:35:33Z 2023 SHA=24a4a0 64-bit
Report bugs to <hamlib-developer@lists.sourceforge.net>

yaesu: initrigs4_yaesu called
rig_init: rig_model=Yaesu FT-847
rig_init: rig has VFO_SUB
rig_init: rig has VFO_MAIN
ft847_init called 
 1:rig.c(827):rig_open entered
rig_settings_load_all: settings_file (/home/kconkas/.config/hamlib_settings): No such file or directory
rig_open: cwd=/home/kconkas/Downloads/mydde
rig_open: /home/kconkas/Downloads/mydde/hamlib_settings does not exist
serial_open: /dev/ttyUSB0
tcflush
  2:rig.c(7508):async_data_handler_start entered
  2:rig.c(7515):async_data_handler_start returning(0) 
rig.c(262):add_opened_rig returning2(0) 
rig_open: 0x1427c7c rs->comm_state==1?=1
ft847_open: called
ft847.c(1086):ft847_open returning2(0) 
rig_get_freq(2044) called vfo=VFOA
rig.c(2053) vfo=Main, curr_vfo=Main
rig_get_freq(2141): vfo_opt=0, model=1001
get_freq_and_mode: vfo =Main 
read_block_generic called, direct=1
from_bcd_be called
  2:rig.c(2619):rig_passband_normal entered
  2:rig_passband_normal: return filter#0, width=2200
  2:rig.c(2640):rig_passband_normal returning(2200) 
rig_get_freq(2044) called vfo=VFOB
rig.c(2053) vfo=Sub, curr_vfo=Main
rig_get_freq(2141): vfo_opt=0, model=1001
get_freq_and_mode: vfo =Sub 
read_block_generic called, direct=1
from_bcd_be called
  2:rig.c(2619):rig_passband_normal entered
  2:rig_passband_normal: return filter#0, width=2200
  2:rig.c(2640):rig_passband_normal returning(2200) 
  2:rig.c(5147):rig_get_split_vfo entered
ft847_get_split_vfo called
  2:rig.c(5212):rig_get_split_vfo returning(0) 
rig_open(1388): Current split=0, tx_vfo=Sub
  2:rig.c(2465):rig_get_mode entered
get_freq_and_mode: vfo =Main 
read_block_generic called, direct=1
from_bcd_be called
   3:rig.c(2619):rig_passband_normal entered
   3:rig_passband_normal: return filter#0, width=2200
   3:rig.c(2640):rig_passband_normal returning(2200) 
   3:cache.c(35):rig_set_cache_mode entered
   3:cache.c(118):rig_set_cache_mode returning(0) 
  2:rig.c(2598):rig_get_mode returning(0) 
 1:rig.c(1412):rig_open returning(0) 
Opened rig model 1001, 'FT-847'
Backend version: 20220525.0, Status: Stable

Rig command: f
 1:rigctl_parse.c(2115):rigctl_get_freq entered
rig_get_freq(2044) called vfo=currVFO
rig.c(2053) vfo=currVFO, curr_vfo=Main
rig_get_freq(2141): vfo_opt=0, model=1001
get_freq_and_mode: vfo =Main 
read_block_generic called, direct=1
from_bcd_be called
  2:rig.c(2619):rig_passband_normal entered
  2:rig_passband_normal: return filter#0, width=2200
  2:rig.c(2640):rig_passband_normal returning(2200) 
Frequency: 435900000
 1:rigctl_parse.c(2141):rigctl_get_freq returning(0) 

Rig command: i
 1:rigctl_parse.c(2896):rigctl_get_split_freq entered
  2:rig.c(4267):rig_get_split_freq entered
vfo_fixup(1995): split=0, vfo==TX tx_vfo=TX
get_freq_and_mode: vfo =TX 
read_block_generic called, direct=1
from_bcd_be called
   3:rig.c(2619):rig_passband_normal entered
   3:rig_passband_normal: return filter#0, width=2200
   3:rig.c(2640):rig_passband_normal returning(2200) 
  2:rig.c(4293):rig_get_split_freq returning(0) 
TX Frequency: 435900000
 1:rigctl_parse.c(2912):rigctl_get_split_freq returning(0) 
```

The same also happens if I try it with `--vfo` option:
```
$ rigctl -vvvv --vfo -m 1001 -r /dev/ttyUSB0 -s 57600
rigctl.c(442) Startup: rigctl -vvvv --vfo -m 1001 -r /dev/ttyUSB0 -s 57600
rigctl Hamlib 4.6~git Mar 14 22:35:33Z 2023 SHA=24a4a0 64-bit
Report bugs to <hamlib-developer@lists.sourceforge.net>

yaesu: initrigs4_yaesu called
rig_init: rig_model=Yaesu FT-847
rig_init: rig has VFO_SUB
rig_init: rig has VFO_MAIN
ft847_init called 
 1:rig.c(827):rig_open entered
rig_settings_load_all: settings_file (/home/kconkas/.config/hamlib_settings): No such file or directory
rig_open: cwd=/home/kconkas/Downloads/mydde
rig_open: /home/kconkas/Downloads/mydde/hamlib_settings does not exist
serial_open: /dev/ttyUSB0
tcflush
  2:rig.c(7508):async_data_handler_start entered
  2:rig.c(7515):async_data_handler_start returning(0) 
rig.c(262):add_opened_rig returning2(0) 
rig_open: 0x21f9c7c rs->comm_state==1?=1
ft847_open: called
ft847.c(1086):ft847_open returning2(0) 
rig_get_freq(2044) called vfo=VFOA
rig.c(2053) vfo=Main, curr_vfo=Main
rig_get_freq(2141): vfo_opt=0, model=1001
get_freq_and_mode: vfo =Main 
read_block_generic called, direct=1
from_bcd_be called
  2:rig.c(2619):rig_passband_normal entered
  2:rig_passband_normal: return filter#0, width=2200
  2:rig.c(2640):rig_passband_normal returning(2200) 
rig_get_freq(2044) called vfo=VFOB
rig.c(2053) vfo=Sub, curr_vfo=Main
rig_get_freq(2141): vfo_opt=0, model=1001
get_freq_and_mode: vfo =Sub 
read_block_generic called, direct=1
from_bcd_be called
  2:rig.c(2619):rig_passband_normal entered
  2:rig_passband_normal: return filter#0, width=2200
  2:rig.c(2640):rig_passband_normal returning(2200) 
  2:rig.c(5147):rig_get_split_vfo entered
ft847_get_split_vfo called
  2:rig.c(5212):rig_get_split_vfo returning(0) 
rig_open(1388): Current split=0, tx_vfo=Sub
  2:rig.c(2465):rig_get_mode entered
get_freq_and_mode: vfo =Main 
read_block_generic called, direct=1
from_bcd_be called
   3:rig.c(2619):rig_passband_normal entered
   3:rig_passband_normal: return filter#0, width=2200
   3:rig.c(2640):rig_passband_normal returning(2200) 
   3:cache.c(35):rig_set_cache_mode entered
   3:cache.c(118):rig_set_cache_mode returning(0) 
  2:rig.c(2598):rig_get_mode returning(0) 
 1:rig.c(1412):rig_open returning(0) 
Opened rig model 1001, 'FT-847'
Backend version: 20220525.0, Status: Stable

Rig command: f
VFO: Main
rig_parse_vfo called
 1:rigctl_parse.c(2115):rigctl_get_freq entered
rig_get_freq(2044) called vfo=Main
rig.c(2053) vfo=Main, curr_vfo=Main
rig_get_freq(2141): vfo_opt=0, model=1001
get_freq_and_mode: vfo =Main 
read_block_generic called, direct=1
from_bcd_be called
  2:rig.c(2619):rig_passband_normal entered
  2:rig_passband_normal: return filter#0, width=2200
  2:rig.c(2640):rig_passband_normal returning(2200) 
Frequency: 435900000
 1:rigctl_parse.c(2141):rigctl_get_freq returning(0) 

Rig command: f
VFO: Sub
rig_parse_vfo called
 1:rigctl_parse.c(2115):rigctl_get_freq entered
rig_get_freq(2044) called vfo=Sub
rig.c(2053) vfo=Sub, curr_vfo=Main
rig_get_freq(2141): vfo_opt=0, model=1001
get_freq_and_mode: vfo =Sub 
read_block_generic called, direct=1
from_bcd_be called
  2:rig.c(2619):rig_passband_normal entered
  2:rig_passband_normal: return filter#0, width=2200
  2:rig.c(2640):rig_passband_normal returning(2200) 
Frequency: 435900000
 1:rigctl_parse.c(2141):rigctl_get_freq returning(0) 

```

It seems to me like this is a hamlib issue because using MacDoppler (which comes with its own CAT implementation) on a MacOS box I can set both VFOs independently and have no issues with tracking satellites.

Has anyone got this working? If so, I'd appreciate any advice about what I'm doing wrong.

kconkas, 2023-05-11:
If this helps, dump_caps returns:
```
 rigctl -m 1001 -r /dev/ttyUSB0 -s 57600

Rig command: 1
Caps dump for model: 1001
Model name:	FT-847
Mfg name:	Yaesu
Backend version:	20220525.0
Backend copyright:	LGPL
Backend status:	Stable
Rig type:	Other Receiver Transmitter Transceiver 
PTT type:	Rig capable
DCD type:	Rig capable
Port type:	RS-232
Serial speed: 4800..57600 bauds, 8N2 
Write delay: 50mS, timeout 1000mS, 0 retry
Post Write delay: 50mS
Has targetable VFO: Y
Has async data support: N
Announce: 0x0
Max RIT: -0.0kHz/+0.0kHz
Max XIT: -0.0kHz/+0.0kHz
Max IF-SHIFT: -0.0kHz/+0.0kHz
Preamp: None
Attenuator: None
AGC levels: 9=NONE

CTCSS: 67.0 69.3 71.9 74.4 77.0 79.7 82.5 85.4 88.5 91.5 94.8 97.4 100.0 103.5 107.2 110.9 114.8 118.8 123.0 127.3 131.8 136.5 141.3 146.2 151.4 156.7 162.2 167.9 173.8 179.9 186.2 192.8 203.5 210.7 218.1 225.7 233.6 241.8 250.3 Hz, 39 tones
DCS: 23 25 26 31 32 36 43 47 51 53 54 65 71 72 73 74 114 115 116 122 125 131 132 134 143 145 152 155 156 162 165 172 174 205 212 223 225 226 243 244 245 246 251 252 255 261 263 265 266 271 274 306 311 315 325 331 332 343 346 351 356 364 365 371 411 412 413 423 431 432 445 446 452 454 455 462 464 465 466 503 506 516 523 526 532 546 565 606 612 624 627 631 632 654 662 664 703 712 723 731 732 734 743 754, 104 codes
Get functions: 
Set functions: TONE TSQL 
Extra functions:
Get level: RAWSTR(0..255/0) ALC(0..1/0.01) STRENGTH(0..60/0) 
Set level: BAND_SELECT(0..16/1) 
Extra levels:
Get parameters: 
Set parameters: 
Extra parameters:
Mode list: AM CW USB LSB FM CWR 
VFO list: Sub Main 
VFO Ops: 
Scan Ops: 
Number of banks:	0
Memory name desc size:	0
Memories: None
TX ranges #1 for ITU1:
	1810000 Hz - 2000000 Hz
		VFO list: Sub Main 
		Mode list: AM CW USB LSB FM 
		Antenna list: ANT1 
		Low power: 5 W, High power: 100 W
	3500000 Hz - 3800000 Hz
		VFO list: Sub Main 
		Mode list: AM CW USB LSB FM 
		Antenna list: ANT1 
		Low power: 5 W, High power: 100 W
	7000000 Hz - 7200000 Hz
		VFO list: Sub Main 
		Mode list: AM CW USB LSB FM 
		Antenna list: ANT1 
		Low power: 5 W, High power: 100 W
	10100000 Hz - 10150000 Hz
		VFO list: Sub Main 
		Mode list: AM CW USB LSB FM 
		Antenna list: ANT1 
		Low power: 5 W, High power: 100 W
	14000000 Hz - 14350000 Hz
		VFO list: Sub Main 
		Mode list: AM CW USB LSB FM 
[... 350 lines cut ...]
Can get Transceive:	N
Can set Func:	Y
Can get Func:	N
Can set Level:	N
Can get Level:	Y
Can set Param:	N
Can get Param:	N
Can send DTMF:	N
Can recv DTMF:	N
Can send Morse:	N
Can send Voice:	N
Can decode Events:	N
Can set Bank:	N
Can set Mem:	N
Can get Mem:	N
Can set Channel:	N
Can get Channel:	N
Can ctl Mem/VFO:	N
Can Scan:	N
Can get Info:	N
Can get power2mW:	N
Can get mW2power:	N

Overall backend warnings: 0   
```

kconkas, 2023-05-11:
And another update: it looks like this is working if I _set_ the sub VFO frequency before trying to read it:
```
$ rigctl -m 1001 -r /dev/ttyUSB0 -s 57600

Rig command: i
TX Frequency: 435900000

Rig command: I 145585000

Rig command: i
TX Frequency: 145585000
```

mdblack98, 2023-05-11:
Could you please add 1 more 'v' to the -vvvv?

Mike W9MDB

kconkas, 2023-05-12:
@mdblack98 thanks for looking into this. As requested, the output is:
```
$ rigctl -vvvvv -m 1001 -r /dev/ttyUSB0 -s 57600
rigctl.c(442) Startup: rigctl -vvvvv -m 1001 -r /dev/ttyUSB0 -s 57600
rigctl Hamlib 4.6~git Mar 14 22:35:33Z 2023 SHA=24a4a0 64-bit
Report bugs to <hamlib-developer@lists.sourceforge.net>

rig_check_rig_caps: p1=0x7fdb361be480, p2=0x7fdb361c4a00, rig_model=0x7fdb361be480, macro_name=0x7fdb361c4a00
yaesu: initrigs4_yaesu called
rig_init: rig_model=Yaesu FT-847
rig_init: rig has VFO_SUB
rig_init: rig has VFO_MAIN
ft847_init called 
 1:rig.c(827):rig_open entered
rig_settings_get_path: path=/home/kconkas/.config/hamlib_settings
rig_settings_load_all: settings_file (/home/kconkas/.config/hamlib_settings): No such file or directory
rig_open: cwd=/home/kconkas/Downloads
rig_open: /home/kconkas/Downloads/hamlib_settings does not exist
rig_open: async_data_enable=0, async_data_supported=0
serial_open: /dev/ttyUSB0
serial_setup: tcgetattr
serial_setup: cfmakeraw
serial_setup: cfsetispeed=57600,0x1001
serial_setup: cfsetospeed=57600,0x1001
serial_setup: data_bits=8
serial_setup: parity=0
serial_setup: Handshake=None
serial_setup: tcsetattr TCSANOW
read_string_generic called, rxmax=4095 direct=1, expected_len=1
tcflush
  2:rig.c(7508):async_data_handler_start entered
async_data_handler_start: async data support disabled since async_data_enabled=0
  2:rig.c(7515):async_data_handler_start returning(0) 
rig.c(262):add_opened_rig returning2(0) 
rig_open: 0x57bc7c rs->comm_state==1?=1
ft847_open: called
write_block(): TX 5 bytes, method=1
0000    00 00 00 00 00                                      .....           
ft847.c(1086):ft847_open returning2(0) 
rig.c(1317) trace
rig_open: vfo_curr=currVFO, tx_vfo=TX
vfo_fixup:(from rig_open:1349) vfo=VFOA, vfo_curr=currVFO, split=0
rig_open: No set_vfo function rig so default vfo = Main
rig_get_freq(2044) called vfo=VFOA
vfo_fixup:(from rig_get_freq:2051) vfo=VFOA, vfo_curr=Main, split=0
rig.c(2053) vfo=Main, curr_vfo=Main
rig_get_freq: cache miss age=1000000ms, cached_vfo=Main, asked_vfo=Main, use_cached_freq=0
rig_get_freq(2141): vfo_opt=0, model=1001
get_freq_and_mode: vfo =Main 
write_block(): TX 5 bytes, method=1
0000    00 00 00 00 03                                      .....           
read_block_generic called, direct=1
read_block_generic(): RX 5 bytes, direct=1
0000    43 55 81 07 01                                      CU...           
from_bcd_be called
  2:rig.c(2619):rig_passband_normal entered
  2:rig_passband_normal: return filter#0, width=2200
  2:rig.c(2640):rig_passband_normal returning(2200) 
 1:rig_get_freq: elapsed=302ms
rig_get_freq(2044) called vfo=VFOB
vfo_fixup:(from rig_get_freq:2051) vfo=VFOB, vfo_curr=Main, split=0
vfo_fixup: final vfo=Sub
rig.c(2053) vfo=Sub, curr_vfo=Main
rig_get_freq: cache miss age=1000000ms, cached_vfo=Sub, asked_vfo=Sub, use_cached_freq=0
rig_get_freq(2141): vfo_opt=0, model=1001
get_freq_and_mode: vfo =Sub 
write_block(): TX 5 bytes, method=1
0000    00 00 00 00 03                                      .....           
read_block_generic called, direct=1
read_block_generic(): RX 5 bytes, direct=1
[... 142 lines cut ...]
rigctl_parse: vfo_opt=0
 1:rigctl_parse.c(2896):rigctl_get_split_freq entered
  2:rig.c(4267):rig_get_split_freq entered
vfo_fixup:(from rig_get_split_freq:4281) vfo=TX, vfo_curr=Main, split=1
vfo_fixup(1995): split=1, vfo==TX tx_vfo=Sub
vfo_fixup: RIG_VFO_TX changed to Sub, split=1, satmode=0
rig.c(4311) trace
get_freq_and_mode: vfo =Sub 
write_block(): TX 5 bytes, method=1
0000    00 00 00 00 23                                      ....#           
read_block_generic called, direct=1
read_block_generic(): RX 5 bytes, direct=1
0000    14 58 50 00 00                                      .XP..           
from_bcd_be called
   3:rig.c(2619):rig_passband_normal entered
   3:rig_passband_normal: return filter#0, width=2200
   3:rig.c(2640):rig_passband_normal returning(2200) 
  2:rig_get_split_freq: elapsed=302ms
  2:rig.c(4314):rig_get_split_freq returning(0) 
TX Frequency: 145850000
 1:rigctl_parse.c(2912):rigctl_get_split_freq returning(0) 
rigctl_parse: called, interactive=1

Rig command: 
```

mdblack98, 2023-05-12:
Can you please try the latest in the master repo?
Should be fixed now.

Mike W9MDB

kconkas, 2023-05-12:
@mdblack98 thanks. I've compiled the latest from master and this fixed my problem in satellite mode. However; I am now unable to get the main VFO output in non-satellite mode (even though the frequency does seem to get read correctly):
```
$ /usr/local/bin/rigctl -vvvvv -m 1001 -r /dev/ttyUSB0 -s 57600
rigctl.c(442) Startup: /usr/local/bin/rigctl -vvvvv -m 1001 -r /dev/ttyUSB0 -s 57600
rigctl Hamlib 4.6~git May 11 21:36:12Z 2023 SHA=dfc5e7 64-bit
Report bugs to <hamlib-developer@lists.sourceforge.net>

rig_check_rig_caps: p1=0x7f9dda463ac0, p2=0x7f9dda46a040, rig_model=0x7f9dda463ac0, macro_name=0x7f9dda46a040
yaesu: initrigs4_yaesu called
rig_init: rig_model=Yaesu FT-847 20230511.0
rig_init: rig has VFO_SUB
rig_init: rig has VFO_MAIN
ft847_init called 
 1:rig.c(830):rig_open entered
rig_settings_get_path: path=/home/kconkas/.config/hamlib_settings
rig_settings_load_all: settings_file (/home/kconkas/.config/hamlib_settings): No such file or directory
rig_open: cwd=/home/kconkas/src/Hamlib
rig_open: /home/kconkas/src/Hamlib/hamlib_settings does not exist
rig_open: async_data_enable=1, async_data_supported=0
serial_open: /dev/ttyUSB0
serial_setup: tcgetattr
serial_setup: cfmakeraw
serial_setup: cfsetispeed=57600,0x1001
serial_setup: cfsetospeed=57600,0x1001
serial_setup: data_bits=8
serial_setup: parity=0
serial_setup: Handshake=None
serial_setup: tcsetattr TCSANOW
tcflush
  2:rig.c(7529):async_data_handler_start entered
async_data_handler_start: async data support disabled since async_data_enabled=0
  2:rig.c(7536):async_data_handler_start returning(0) 
rig.c(262):add_opened_rig returning2(0) 
rig_open: 0x2426c7c rs->comm_state==1?=1
ft847_open: called
write_block(): TX 5 bytes, method=1
0000    00 00 00 00 00                                      .....           
ft847.c(1086):ft847_open returning2(0) 
rig.c(1328) trace
rig_open: vfo_curr=currVFO, tx_vfo=TX
vfo_fixup:(from rig_open:1360) vfo=VFOA, vfo_curr=currVFO, split=0
rig_open: No set_vfo function rig so default vfo = Main
rig_get_freq(2054) called vfo=VFOA
vfo_fixup:(from rig_get_freq:2061) vfo=VFOA, vfo_curr=Main, split=0
rig.c(2063) vfo=Main, curr_vfo=Main
rig_get_freq: cache miss age=1000000ms, cached_vfo=Main, asked_vfo=Main, use_cached_freq=0
rig_get_freq(2151): vfo_opt=0, model=1001
get_freq_and_mode: vfo =Main 
write_block(): TX 5 bytes, method=1
0000    00 00 00 00 03                                      .....           
read_block_generic called, direct=1
read_block_generic(): RX 5 bytes, direct=1
0000    02 48 90 00 01                                      .H...           
from_bcd_be called
  2:rig.c(2629):rig_passband_normal entered
  2:rig_passband_normal: return filter#0, width=2200
  2:rig.c(2650):rig_passband_normal returning(2200) 
 1:rig_get_freq: elapsed=302ms
rig_get_freq(2054) called vfo=VFOB
vfo_fixup:(from rig_get_freq:2061) vfo=VFOB, vfo_curr=Main, split=0
vfo_fixup: final vfo=Sub
rig.c(2063) vfo=Sub, curr_vfo=Main
rig_get_freq: cache miss age=1000000ms, cached_vfo=Sub, asked_vfo=Sub, use_cached_freq=0
rig_get_freq(2151): vfo_opt=0, model=1001
get_freq_and_mode: vfo =Sub 
write_block(): TX 5 bytes, method=1
0000    00 00 00 00 23                                      ....#           
read_block_generic called, direct=1
read_block_generic(): Timed out 0.500602 seconds after 0 chars, direct=1
ft847: read_block returned -5
[... 123 lines cut ...]
rig_open: 0x2426c7c rs->comm_state==1?=1
ft847_open: called
write_block(): TX 5 bytes, method=1
0000    00 00 00 00 00                                      .....           
ft847.c(1086):ft847_open returning2(0) 
rig.c(1328) trace
rig_open: vfo_curr=Main, tx_vfo=TX
vfo_fixup:(from rig_open:1360) vfo=VFOA, vfo_curr=Main, split=0
rig_open: No set_vfo function rig so default vfo = Main
rig_get_freq(2054) called vfo=VFOA
vfo_fixup:(from rig_get_freq:2061) vfo=VFOA, vfo_curr=Main, split=0
rig.c(2063) vfo=Main, curr_vfo=Main
rig_get_freq: cache miss age=1993ms, cached_vfo=Main, asked_vfo=Main, use_cached_freq=0
rig_get_freq(2151): vfo_opt=0, model=1001
get_freq_and_mode: vfo =Main 
write_block(): TX 5 bytes, method=1
0000    00 00 00 00 13                                      .....           
read_block_generic called, direct=1
read_block_generic(): Timed out 0.500605 seconds after 0 chars, direct=1
ft847: read_block returned -5
 1:rig_get_freq: elapsed=802ms
 1:rig.c(1423):rig_open returning(0) 
main: rig_open retcode=0
rigctl_parse: called, interactive=1
```

mdblack98, 2023-05-12:
Please try again...

kconkas, 2023-05-12:
@mdblack98 thank you so much for fixing this and so quickly, hamradio and open source at its best. I can confirm it's working now both for sat and non-sat mode. There's just one tiny slight snag in the sat mode: I can only read the secondary VFO after I have written to it, but this is good enough to me (gpredict, where I connect to `rigctld` from, sets both frequencies):
```
$ /usr/local/bin/rigctl -vvvvv -m 1001 -r /dev/ttyUSB0 -s 57600
rigctl.c(442) Startup: /usr/local/bin/rigctl -vvvvv -m 1001 -r /dev/ttyUSB0 -s 57600
rigctl Hamlib 4.6~git May 12 16:19:02Z 2023 SHA=1f98f0 64-bit
Report bugs to <hamlib-developer@lists.sourceforge.net>

rig_check_rig_caps: p1=0x7febb6bffac0, p2=0x7febb6c06040, rig_model=0x7febb6bffac0, macro_name=0x7febb6c06040
yaesu: initrigs4_yaesu called
rig_init: rig_model=Yaesu FT-847 20230512.0
rig_init: rig has VFO_SUB
rig_init: rig has VFO_MAIN
ft847_init called 
 1:rig.c(830):rig_open entered
rig_settings_get_path: path=/home/kconkas/.config/hamlib_settings
rig_settings_load_all: settings_file (/home/kconkas/.config/hamlib_settings): No such file or directory
rig_open: cwd=/home/kconkas/src/Hamlib
rig_open: /home/kconkas/src/Hamlib/hamlib_settings does not exist
rig_open: async_data_enable=1, async_data_supported=0
serial_open: /dev/ttyUSB0
serial_setup: tcgetattr
serial_setup: cfmakeraw
serial_setup: cfsetispeed=57600,0x1001
serial_setup: cfsetospeed=57600,0x1001
serial_setup: data_bits=8
serial_setup: parity=0
serial_setup: Handshake=None
serial_setup: tcsetattr TCSANOW
tcflush
  2:rig.c(7529):async_data_handler_start entered
async_data_handler_start: async data support disabled since async_data_enabled=0
  2:rig.c(7536):async_data_handler_start returning(0) 
rig.c(262):add_opened_rig returning2(0) 
rig_open: 0x13c6c7c rs->comm_state==1?=1
ft847_open: called
write_block(): TX 5 bytes, method=1
0000    00 00 00 00 00                                      .....           
ft847.c(1086):ft847_open returning2(0) 
rig.c(1328) trace
rig_open: vfo_curr=currVFO, tx_vfo=TX
vfo_fixup:(from rig_open:1360) vfo=VFOA, vfo_curr=currVFO, split=0
rig_open: No set_vfo function rig so default vfo = Main
rig_get_freq(2054) called vfo=VFOA
vfo_fixup:(from rig_get_freq:2061) vfo=VFOA, vfo_curr=Main, split=0
rig.c(2063) vfo=Main, curr_vfo=Main
rig_get_freq: cache miss age=1000000ms, cached_vfo=Main, asked_vfo=Main, use_cached_freq=0
rig_get_freq(2151): vfo_opt=0, model=1001
get_freq_and_mode: vfo =Main 
write_block(): TX 5 bytes, method=1
0000    00 00 00 00 03                                      .....           
read_block_generic called, direct=1
read_block_generic(): RX 5 bytes, direct=1
0000    43 59 00 00 01                                      CY...           
from_bcd_be called
  2:rig.c(2629):rig_passband_normal entered
  2:rig_passband_normal: return filter#0, width=2200
  2:rig.c(2650):rig_passband_normal returning(2200) 
 1:rig_get_freq: elapsed=302ms
rig_get_freq(2054) called vfo=VFOB
vfo_fixup:(from rig_get_freq:2061) vfo=VFOB, vfo_curr=Main, split=0
vfo_fixup: final vfo=Sub
rig.c(2063) vfo=Sub, curr_vfo=Main
rig_get_freq: cache miss age=1000000ms, cached_vfo=Sub, asked_vfo=Sub, use_cached_freq=0
rig_get_freq(2151): vfo_opt=0, model=1001
get_freq_and_mode: vfo =Sub 
write_block(): TX 5 bytes, method=1
0000    00 00 00 00 23                                      ....#           
read_block_generic called, direct=1
read_block_generic(): RX 5 bytes, direct=1
0000    14 59 00 00 00                                      .Y...           
[... 140 lines cut ...]
Rig command: i
rigctl_parse: input_line: i
rigctl_parse: vfo_opt=0
 1:rigctl_parse.c(2893):rigctl_get_split_freq entered
  2:rig.c(4279):rig_get_split_freq entered
vfo_fixup:(from rig_get_split_freq:4293) vfo=TX, vfo_curr=Main, split=1
vfo_fixup(2004): split=1, vfo==TX tx_vfo=Sub
vfo_fixup: RIG_VFO_TX changed to Sub, split=1, satmode=0
rig.c(4323) trace
get_freq_and_mode: vfo =Sub 
write_block(): TX 5 bytes, method=1
0000    00 00 00 00 23                                      ....#           
read_block_generic called, direct=1
read_block_generic(): RX 5 bytes, direct=1
0000    14 59 05 00 00                                      .Y...           
from_bcd_be called
   3:rig.c(2629):rig_passband_normal entered
   3:rig_passband_normal: return filter#0, width=2200
   3:rig.c(2650):rig_passband_normal returning(2200) 
  2:rig_get_split_freq: elapsed=302ms
  2:rig.c(4326):rig_get_split_freq returning(0) 
TX Frequency: 145905000
 1:rigctl_parse.c(2909):rigctl_get_split_freq returning(0) 
rigctl_parse: called, interactive=1
```

closed 2023-09-29 by mdblack98