Hamlib/Hamlib #1287: Extra retry and delay in read_string_generic() slows down serial communication

mikaelnousiainen, 2023-05-12:
@mdblack98 It seems this commit (https://github.com/Hamlib/Hamlib/commit/4ada2117131f85e438727d28f4606e5357304136 ) slows down serial communication quite a bit, especially in non-async mode (testing on IC-7600 and FT-991A), because serial_flush() now leads to a retry and extra 50ms pause.

Is there some specific reason to do this? I have not seen any significant issues with low-level serial comms on any rig so far... (and I've tested many)

If there is, it needs to be enabled on a per-rig and per-call basis, so that serial_flush() doesn't lead to extra waiting.

mdblack98, 2023-05-12:
We were having problems on different interfaces where tcflush() doesn't work on some and the read char method doesn't work on others so ended up having to do both.

We could start making exceptions by rig so if you want to do that first for the 7300 we can continue to add others that we can test.

Mike W9MDB

mikaelnousiainen, 2023-05-14:
@mdblack98 I've attempted to fix this by making the extra timeout retries configurable via a set-conf option in #1289 . I think that is the correct approach, since at least most modern rigs with USB port would never need this. It can now be enabled on a per-case basis, where retries can be used with a problematic USB serial interface.

Let's move this discussion to #1289 .

closed 2023-05-14 by mikaelnousiainen