Hamlib/Hamlib #1292: Is using half of the timeout correct behavior?

mikaelnousiainen, 2023-05-14:
@mdblack98 This is more of a question just to get the next Hamlib release out fast: Since the read timeout retry count is now optional and configurable (in #1289), is this commit about using half of the timeout time now unnecessary (or even related)?: 

https://github.com/Hamlib/Hamlib/commit/5f55b0ea486fdd1f5158543bb3d5ea349a2beaf7

I'm not sure what the change was based on?

mdblack98, 2023-05-14:
The timeout option you put in there needs to be reversed -- assume it's needed until proven otherwise...so we can set it for "this rigs works without the 2nd attempt".
If the two-attempt timeout is enabled then we still need half the timeout to prevent longer timeouts from happening.
This will all go away if we can rid of flush and tcflush completely.  Good test for this is to a port split and run an app on both ports and see if they get collision errors (which they do now).  If we can allow all the rigs to do async processing we can get rid of the flush calls.  Just need the rig to ignore anything it doesn't expect although the ?; response might be problematic.  So I imagine this might have to be per rig also.

mdblack98, 2023-05-14:
And my planned release schedule right now is to put out 4.6 a bit before WSJT-X final is put out.
4.5.6 may never be released.  4.6 has diverged so far from 4.5.5 backporting is difficult

We're using the RC candidates to test 4.6.

Mike W9MDB

Tyrbiter, 2023-05-14:
There are currently 106 open entries in the 4.6 milestone Mike, would it be sensible to chop those down somewhat as resolving them all during the WSJTX-rc process seems challenging?

mdblack98, 2023-05-14:
It's a long way away from releasing 4.6 -- everything goes into the next milestone until we run out of time.

The goal is to release 4.6 a week or two before the general WSJT-X release.

Mike W9MDB

Tyrbiter, 2023-05-14:
OK, fair enough, I just wanted to mention it as I didn't want you to be flogged to meet the schedule.

mikaelnousiainen, 2023-05-14:
> The timeout option you put in there needs to be reversed -- assume it's needed until proven otherwise...so we can set it for "this rigs works without the 2nd attempt". If the two-attempt timeout is [...]

@mdblack98 I'm not entirely sure about this approach. Based on some 20-30 popular Kenwood/Yaesu/Icom (all they way from IC-7000 and Yaesu FT-897) rig models that I've received feedback about, there have never been issues with rig needing constant retries. And this is even when used with WSJT-X.

An additional aspect here are serial interfaces for older rigs that don't have USB connectivity: it could be the USB<->Serial interface causing this and not the actual rig, so going even on a per-rig setting wouldn't be accurate.

Anyway, to go forward with this: maybe we can set the timeout_retry default value to be 1, so that it can be disabled if the user wishes to do so. Also, the additional should be disabled during flush, as flush will always time out (we are still using flushing, after all). I can create another PR today to do this.

Also, a good question is what the retry sleep should be. I lowered it to 10ms on purpose, so by controlling the number of retries one can get longer times easily.

I can also add a rig caps setting for this, so we can disable it per rig too. Without a rig specific setting, it would still default to 1 retry.

I'll push out a PR soon.

mdblack98, 2023-05-14:
10ms is probably not enough.  Slowest rig is 300baud so 33ms per byte.  Could make it scale by serial speed...something like 1/(serial_rate/10)*2 with a minimum of 1ms;  So 300 baud gives 66ms and 1200 baud 16ms and 19200 and up 1ms

These flush problems exist with some virtual serial ports, microham devices, and some others.
So we can't break all those.

Mike W9MDB

@mdblack98 I'm not entirely sure about this approach. Based on some 20-30 popular Kenwood/Yaesu/Icom (all they way from IC-7000 and Yaesu FT-897) rig models that I've received feedback about, there have never been issues with rig needing constant retries. And this is even when used with WSJT-X.

An additional aspect here are serial interfaces for older rigs that don't have USB connectivity: it could be the USB<->Serial interface causing this and not the actual rig, so going even on a per-rig setting wouldn't be accurate.

Anyway, to go forward with this: maybe we can set the timeout_retry default value to be 1, so that it can be disabled if the user wishes to do so. Also, the additional should be disabled during flush, as flush will always time out (we are still using flushing, after all). I can create another PR today to do this.

Also, a good question is what the retry sleep should be. I lowered it to 10ms on purpose, so by controlling the number of retries one can get longer times easily.

I can also add a rig caps setting for this, so we can disable it per rig too. Without a rig specific setting, it would still default to 1 retry.

I'll push out a PR soon.

mikaelnousiainen, 2023-05-14:
See #1294

mikaelnousiainen, 2023-05-29:
Fixde in #1294

closed 2023-05-29 by mikaelnousiainen