Hamlib/Hamlib #1296: Quickly switching between VFO A and VFO B

Michael-AUG, 2023-05-20:
Hi everyone,

I use Hamlib (and rigctld in particular) for sending my VFO frequency to CloudLog. It runs as an autorun on an attached Raspberry Pi.

When I turn the radio on (the Pi is always running), every two or three times, the radio spends about five minutes quickly switching between VFO A and VFO B. Sometimes I don't notice this in time and begin transmitting, meaning my TX it split between the two frequencies. Not ideal!

Does anyone have any advice? Or is this a bug?

Many thanks indeed.
Michael

mdblack98, 2023-05-20:
Please run rigctld with this added:
-vvvvv -Z >log.txt 2>&1
Duplicate the problem, ctrl-c the rigctld process, then send me the log.
Mike W9MDB

Michael-AUG, 2023-05-20:
Thanks Mike

So - I wasn't able to replicate it entirely because I am usually running the rigctld on the Pi as a service  from startup, however, I used the command: $ rigctld -r /dev/ttyUSB0 -m 3070 -vvvvv -Z >log.txt 2>&1

The file is attached.
[log.txt](https://github.com/Hamlib/Hamlib/files/11522785/log.txt)

73 Michael GM5AUG

Michael-AUG, 2023-05-20:
Sorry - didn't mean to close the Issue!

mdblack98, 2023-05-21:
The latest version should at least get rid of the get_powerstat error...though it does not address the VFO switching.
Exactly how are you sending frequency to Cloudlog?  And are you trying to send both A and B frequencies?  Every time you ask for B frequency there will be a VFO swap that occurs.  That's the only way to get it.

Michael-AUG, 2023-05-21:
I use the application CloudLogPyCAT (https://github.com/mbridak/CloudLogPyCAT) to deal with the CloudLog side, Mike.
MT - GM5AUG

mdblack98, 2023-05-21:
I don't have time to debug 3rd party applications.

Hamlib only does what it's asked to do.  I imagine this cloudlog application does not know about --vfo mode in rigctld which avoids VFO swapping on rigs that can directly address VFOB.
But that's only about half of the rigs.  The other half cannot directly address VFOB and have to vfo swap to get VFOB frequency.

If they don't have the ability to turn off VFOB frequency that's a problem for the rigs that cannot get VFOB freq without a VFO swap.

Mike

Michael-AUG, 2023-05-21:
I realise you don't have time to debug 3rd party applications Mike (no need to be narky with me). I guess my main interest was that the issue is not occurring when I launch the 3rd party application but when I either a) turn the rig on or b) launch rigctld.
Which makes me think the problem may actually be at the hamlib end??
May I respectfully ask you to consider reopening this case and not writing it off abruptly?

mdblack98, 2023-05-21:
In order to avoid VFO swapping you need two things.

#1 rigctld has to run in --vfo mode
#2 Your rig has to be able to get VFOB frequency without swapping vfos.

The IC-7100 cannot get VFOB frequency without swapping vfos so #2 is your situation that is not fixable at all.

Can you compile and run the latest Hamlib and provide another debug log?  I am wondering why VFOB is being requested at all.  

Mike W9MDB

ejkeerbs, 2023-08-25:
I have this same issue - fast VFO switching - with Hamlib 4.5.5 on Fedora 38. I am using the precompiled version that's available through repositories available with Fedora. I use Hamlib with CQRLOG logging software, and occasionally with wsjtx or jtdx. My radio is an Icom IC-7610.

The fast switching rarely starts immediately upon launching CQRLOG - ironically, it seems to start most frequently when my YL and I key up the transmitter! Most of our operating is on a traffic-handling voice net, and we listen anywhere from 5 to 30 minutes before we can check in. Apparently, the switching is fast enough that none of the net controls has noticed there is an issue, yet I notice because my VFO B is usually set to receive WWV and my frequency-reading linear amp complains that WWV is outside of ham bands.

I do NOT have this issue with Hamlib 4.5.4, and I am currently preventing Fedora from upgrading to 4.5.5 after confirming that this is an issue ONLY with version 4.5.5.

I don't think this matters, but will note it anyway. I am using a commercial USB-to-RS232 adapter, attached to a home-brew RS232-to-TTL converter that connects my linear, radio, and computer. I developed this setup when I had an IC746Pro and a (now retired) SteppIR antenna, as I could only interface to one device with the 746.

I have tried capturing logfiles per the above suggestion - there is a place to add additional commands for rigctld in the CQRLOG configuration screens - yet I have not been successful. I think rigctld is trying to save the logfile to a folder that is only accessible for the root user, as I have never found a file named "log.txt" when doing a file search.

Eric
AD7HM

mdblack98, 2023-08-25:
Can you compile yourself?

Grab the tar file from here and try the 4.6 development version.

https://n0nb.users.sourceforge.net/

Mike W9MDB

ejkeerbs, 2023-09-03:
Per the suggestion of Mike, W9MDB, I pulled the 4.6 development version code down (on 29 August)  to my machine and compiled it. Well...okay...took 2 attempts as I did not have one of the build dependencies installed.

I did _not_ see the fast VFO switching issue when I was using CQRLOG this evening. I'm going to continue with this software for another week...might stick with it instead of reinstalling the 4.5.4 stable version if I don't run into any other issues.

73,

Eric
AD7HM

closed 2023-09-03 by mdblack98