Hamlib/Hamlib #1303: New rig: Anytone AT-D578UVIII

jrobertfisher, 2023-05-27:
Hi Mike,

Nate told me to reach out to get the Anytone AT-D578UVIII added as a rig. I was able to figure out how to get CAT PTT working with it. https://github.com/jrobertfisher/AT-D578UV-software-mic/wiki 

[Anytone AT-D578UVIII]
CmdType=HEX
PTTOn=4101000000000006
PTTOff=4100000000000006

mdblack98, 2023-05-27:
I've just about got the beginnings of this coded...

Please contact me directly and we can hook up and test it.
***@***.***

Mike W9MDB

jrobertfisher, 2023-05-27:
just emailed your yahoo account that was posted on QRZ. Github stripped your email address.

Tyrbiter, 2023-05-27:
Something in these last 5 commits has broken things here, freedv now shows a hamlib error on modem start, it doesn't read anything from the radio after that. TS-890, working perfectly on a previous local build if I downgrade.

closed 2023-05-29 by mdblack98