Hamlib/Hamlib #1308: FT-8x7: Cannot switch between PKTUSB and PKTLSB

tmiw, 2023-06-03:
I recently made some updates to the FreeDV application to support rig control. As part of this, FreeDV calls `rig_set_mode` to ensure that the application is following the agreed-upon standard for FreeDV (`RIG_MODE_PKTLSB` or `RIG_MODE_LSB` for 40 meters and below and the USB equivalents for 20 and above). During testing with my FT-817 I noticed that hamlib was not switching between `RIG_MODE_PKTLSB` and `RIG_MODE_PKTUSB`. Here's an example of what happens on 4.5.4 when I use rigctl:

```
Mooneers-16-MacBook-Pro-2461:build_osx mooneer$ rigctl -V
rigctl Hamlib 4.5.4 Jan 10 01:31:41Z 2023 SHA=921cc5
Mooneers-16-MacBook-Pro-2461:build_osx mooneer$ rigctl -m 1020 -r /dev/cu.usbserial-141320 -vvv
Opened rig model 1020, 'FT-817'

Rig command: m
Mode: PKTUSB
Passband: 0

Rig command: M PKTLSB 3000
set_mode: error =   2:rig.c(2542):rig_get_mode returning(0) 
 1:rigctl_parse.c(2260):rigctl_get_mode returning(0) 
rigctl_parse: called, interactive=1
rigctl_parse: input_line: M PKTLSB 3000
rigctl_parse: vfo_opt=0
 1:rigctl_parse.c(2209):rigctl_set_mode entered
rig_parse_mode called
rig_set_mode called, vfo=currVFO, mode=PKTLSB, width=3000, curr_vfo=VFOA
vfo_fixup:(from rig_set_mode:2291) vfo=VFOA, vfo_curr=VFOA, split=0
rig.c(2314) trace
ft817_set_mode: generic mode = PKTLSB
rig_set_mode: targetable retcode after set_mode(PKTLSB)=-1
Invalid parameter

 1:rig_set_mode: elapsed=0ms
 1:rig.c(2369):rig_set_mode returning(-1) Invalid parameter

0:rigctl_parse.c(2226):rigctl_set_mode returning(-1) Invalid parameter

Invalid parameter
```

(after manually switching menu 26 to "USER-L")

```
Rig command: m
Mode: PKTLSB
Passband: 0
```

Interestingly, the latest 4.6-git (as of commit 2491d208d945859f294f6f56ea0b52396350562e) exhibits different behavior (showing RTTYR no matter what):

```
Mooneers-16-MacBook-Pro-2461:build_osx mooneer$ ../hamlib/bin/rigctl -V
rigctl Hamlib 4.6~git Jun 02 20:16:38Z 2023 SHA=2491d2 64-bit
Mooneers-16-MacBook-Pro-2461:build_osx mooneer$ ../hamlib/bin/rigctl -m 1020 -r /dev/cu.usbserial-141320 -vvv
Opened rig model 1020, 'FT-817'

Rig command: m
Mode: RTTYR
Passband: 2200

Rig command: M PKTLSB 2200
set_mode: error = rigctl_parse: input_line: M PKTLSB 2200
rigctl_parse: vfo_opt=0
rigctl_parse.c(2224):rigctl_set_mode entered
rig_parse_mode called
 1:rig.c(2340):rig_set_mode entered
lock: rig_set_mode
rig_set_mode called, vfo=currVFO, mode=PKTLSB, width=2200, curr_vfo=VFOA
vfo_fixup:(from rig_set_mode:2390) vfo=VFOA, vfo_curr=VFOA, split=0
rig.c(2414) trace
ft817_set_mode: generic mode = PKTLSB
rig_set_mode: targetable retcode after set_mode(PKTLSB)=-1
Invalid parameter

 1:rig_set_mode: elapsed=0ms
unlock: rig_set_mode
 1:rig.c(2473):rig_set_mode returning(-1) Invalid parameter

rigctl_parse.c(2241):rigctl_set_mode returning2(-1) Invalid parameter

Invalid parameter

[switched menu 26 to USER-L again here]

Rig command: m
Mode: RTTYR
Passband: 2200

Rig command: 
```

From what I can tell, it seems that hamlib isn't sending the commands to update menu 26 when going between PKTUSB and PKTLSB (which do appear at least [theoretically possible to do](http://www.ka7oei.com/ft817_meow.html)). This may be intentional, though, so I figure I'd put it out there to confirm (along with always showing RTTYR on 4.6-git).

mdblack98, 2023-06-04:
The 817 cannot set the menu 26 modes via CAT control -- can only be done by the menu.  Also no way to read that status.

PKTUSB should put it in DIG mode and you select USER-U or USER-L yourself for upper/lower sideband.

PKTLSB is not defined and if it was would just do the same thing as PKTUSB and I think that would be more misleading than rejecting it.  But I'm open to suggestions.

Mike W9MDB

Interestingly, the latest 4.6-git (as of commit 2491d20) exhibits different behavior (showing RTTYR no matter what):

mdblack98, 2023-06-04:
I added PKTLSB for the FT817 and warning message about Menu 26.  That's the best I can do without writing to the EEPROM.  I'm a bit reluctant to write to the EEPROM to avoid contributing to burn out although if we validate the mode is actually changing we might be able to implement it since changing the VFO mode is an infrequent operation...we just need to avoid re-writing it when it doesn't need to change.
At this point getting the mode should return the correct mode of LSB, USB, CW, CWR, AM, WFM,  FM, RTTYR, PKTLSB, PKTUSB based on the Menu 26 setting. But setting USER-U USER-L still takes menu 26.  Looks like the FT-817 does not cache the writes but actually updates the EEPROM on each write.

tmiw, 2023-06-05:
Just synced to 55cbf80719a3b6972731ad9d85c725f0998a4e11 and am getting the following while running through all the modes:

```
Mooneers-16-MacBook-Pro-2461:hamlib-code mooneer$ ../hamlib/bin/rigctl -m 1020 -r /dev/cu.usbserial-141220 -vvv
Opened rig model 1020, 'FT-817'

Rig command: m
Mode: RTTYR
Passband: 2200

Rig command: M PKTLSB
Passband: 0
rig_set_mode: FT817 must use Menu 26 to switch between upper/lower sideband, Hamlib only sets DIG mode

Rig command: m
Mode: RTTYR
Passband: 2200

Rig command: M PKTUSB
Passband: 0
rig_set_mode: FT817 must use Menu 26 to switch between upper/lower sideband, Hamlib only sets DIG mode

Rig command: m
Mode: RTTYR
Passband: 2200

Rig command: M RTTYR
Passband: 0
rig_set_mode: FT817 must use Menu 26 to switch between upper/lower sideband, Hamlib only sets DIG mode
set_mode: error = rigctl_parse: vfo_opt=0
rigctl_parse.c(2224):rigctl_set_mode entered
rig_parse_mode called
 1:rig.c(2345):rig_set_mode entered
lock: rig_set_mode
rig_set_mode called, vfo=currVFO, mode=RTTYR, width=0, curr_vfo=VFOA
vfo_fixup:(from rig_set_mode:2395) vfo=VFOA, vfo_curr=VFOA, split=0
rig.c(2419) trace
ft817_set_mode: generic mode = RTTYR
rig_set_mode: targetable retcode after set_mode(RTTYR)=-1
rig_set_mode: FT817 must use Menu 26 to switch between upper/lower sideband, Hamlib only sets DIG mode
Invalid parameter

 1:rig_set_mode: elapsed=0ms
unlock: rig_set_mode
 1:rig.c(2482):rig_set_mode returning(-1) Invalid parameter

rigctl_parse.c(2241):rigctl_set_mode returning2(-1) Invalid parameter

Invalid parameter

Rig command: M RTTY
Passband: 0
rig_set_mode: FT817 must use Menu 26 to switch between upper/lower sideband, Hamlib only sets DIG mode

Rig command: M RTTYR
Passband: 0
rig_set_mode: FT817 must use Menu 26 to switch between upper/lower sideband, Hamlib only sets DIG mode
set_mode: error = rigctl_parse: vfo_opt=0
rigctl_parse.c(2224):rigctl_set_mode entered
rig_parse_mode called
 1:rig.c(2345):rig_set_mode entered
lock: rig_set_mode
rig_set_mode called, vfo=currVFO, mode=RTTYR, width=0, curr_vfo=VFOA
vfo_fixup:(from rig_set_mode:2395) vfo=VFOA, vfo_curr=VFOA, split=0
rig.c(2419) trace
ft817_set_mode: generic mode = RTTYR
rig_set_mode: targetable retcode after set_mode(RTTYR)=-1
rig_set_mode: FT817 must use Menu 26 to switch between upper/lower sideband, Hamlib only sets DIG mode
Invalid parameter

 1:rig_set_mode: elapsed=0ms
unlock: rig_set_mode
 1:rig.c(2482):rig_set_mode returning(-1) Invalid parameter

rigctl_parse.c(2241):rigctl_set_mode returning2(-1) Invalid parameter

Invalid parameter

Rig command: M USB
Passband: 0
rig_set_mode: FT817 must use Menu 26 to switch between upper/lower sideband, Hamlib only sets DIG mode

Rig command: M LSB
Passband: 0
rig_set_mode: FT817 must use Menu 26 to switch between upper/lower sideband, Hamlib only sets DIG mode

Rig command: m
Mode: LSB
Passband: 2200

Rig command: M USB
Passband: 0
rig_set_mode: FT817 must use Menu 26 to switch between upper/lower sideband, Hamlib only sets DIG mode

Rig command: m
Mode: USB
Passband: 2200

Rig command: M FM
Passband: 0
rig_set_mode: FT817 must use Menu 26 to switch between upper/lower sideband, Hamlib only sets DIG mode

Rig command: m
Mode: FM
Passband: 9000

Rig command: M AM
Passband: 0
rig_set_mode: FT817 must use Menu 26 to switch between upper/lower sideband, Hamlib only sets DIG mode

Rig command: m
Mode: AM
Passband: 6000

Rig command: M CW
Passband: 0
rig_set_mode: FT817 must use Menu 26 to switch between upper/lower sideband, Hamlib only sets DIG mode

Rig command: m
Mode: CW
Passband: 2200

Rig command: M CWR
Passband: 0
rig_set_mode: FT817 must use Menu 26 to switch between upper/lower sideband, Hamlib only sets DIG mode

Rig command: m
Mode: CWR
Passband: 2200

Rig command: M WFM
Passband: 0
rig_set_mode: FT817 must use Menu 26 to switch between upper/lower sideband, Hamlib only sets DIG mode
set_mode: error = rigctl_parse: vfo_opt=0
rigctl_parse.c(2224):rigctl_set_mode entered
rig_parse_mode called
 1:rig.c(2345):rig_set_mode entered
lock: rig_set_mode
rig_set_mode called, vfo=currVFO, mode=WFM, width=0, curr_vfo=VFOA
vfo_fixup:(from rig_set_mode:2395) vfo=VFOA, vfo_curr=VFOA, split=0
rig.c(2419) trace
ft817_set_mode: generic mode = WFM
rig_set_mode: targetable retcode after set_mode(WFM)=-1
rig_set_mode: FT817 must use Menu 26 to switch between upper/lower sideband, Hamlib only sets DIG mode
Invalid parameter

 1:rig_set_mode: elapsed=0ms
unlock: rig_set_mode
 1:rig.c(2482):rig_set_mode returning(-1) Invalid parameter

rigctl_parse.c(2241):rigctl_set_mode returning2(-1) Invalid parameter

Invalid parameter

Rig command:
```

Looks like PKTUSB/PKTLSB are still reporting as RTTYR at the moment regardless of the Menu 26 setting.

Anyway, the preferred option is for the Menu 26 EEPROM setting to get updated on mode change (which based on the testing done in https://github.com/drowe67/freedv-gui/pull/408 doesn't seem like it would burn out the EEPROM), but failing that, the existing 4.5 behavior works too (correct reporting of PKTUSB/PKTLSB when the mode's queried); as the current sideband turns red in the FreeDV application if it's not as expected, that would be a sufficient indication that the mode didn't change properly.