Hamlib/Hamlib #1309: IC-7300 PASSBAND_NOCHANGE not working correctly

mdblack98, 2023-06-06:
Hi!

Workflow:

    issue command:  +\set_mode currVFO USB -1
    manually set filter other than #1:  set filter #2
    issue command:  +\set_mode currVFO LSB -1

Bug:

    filter returns always to #1 when sideband changes.

Setup:

    Icom IC7300 firmware 1.41
    rigctld Hamlib 4.6~git Jun 05 13:34:59Z 2023 SHA=acc426 64-bit
    Fedora release 37 (Thirty Seven)

Question:

    Bandwidth "-1" should keep the selected bandwidth, this case keep
selected filter #2, but it does not.
    I understand that filter resets to #1 when mode changes (CW->SSB,
SSB->AM etc.) but this case mode is same SSB, just sideband changes.

    Is this a bug, intentional feature or Icom property?

closed 2023-06-10 by mdblack98

PD5DJ, 2025-05-11:
I have the same issue using an ICOM 7610..

Freq: 7123khz
Mode: USB
Passband: 2100hz

When setting "+M LSB -1" passband is always set to 2400hz bandwith..

Freq: 7123khz
Mode: LSB
Passband: 2400hz

Still a bug?

dforsi, 2025-05-18:
Please open a new issue attaching a log file from `rigctl` with trace enabled:
```
rigctl -m 3078 -vvvvv m M USD 2100 m M LSB -1 m
```
or change the commands if you find a shorter or better way to show the problem in you setup.