Hamlib/Hamlib #1314: Icom IC-905 10Ghz/3CM Changing VFO on Transceiver causes OOB in WSJT-X

djsincla, 2023-06-17:
All other bands appear ok, but when changing VFO from say 10386.100.000 to 10386.099.000 on Transceiver, WSJT-X will change from normal 3CM 10,386.100 000 to OOB 386.099.000 dropping the 10.

djsincla, 2023-06-17:
Here is a picture of the error: https://imgur.com/a/h1FD2IP

mdblack98, 2023-06-17:
Try this DLL....

https://www.dropbox.com/s/snmkzu8eif89yqs/libhamlib-4.dll?dl=0

Mike W9MDB

djsincla, 2023-06-17:
MacOS ;)

Dwayne AB6A

mdblack98, 2023-06-17:
Can you build yourself?

If not I've got a user that can build a Mac library.

Mike W9MDB

djsincla, 2023-06-17:
I believe I can build myself… Just pull the latest repository?

Dwayne AB6A

mdblack98, 2023-06-17:
Perhaps Craig would nice enough to build a new dylib?

Mike

mdblack98, 2023-06-17:
Yup.

djsincla, 2023-06-17:
No worries - I can do that. I will let you know.

Dwayne AB6A

mdblack98, 2023-08-01:
Still waiting to hear if this is OK now.

djsincla, 2023-08-01:
I will confirm this weekend.Regards Dwayne AB6A