Hamlib/Hamlib #1328: Hy-Gain DCU-1 position reporting

danieljkemp, 2023-07-06:
Good morning!

Looking through the docs I see that the hygain dcu1/dcu1x does not list support for getting current position.

I *think* it is possible based on other discussions and docs that I have found.

Is there a technical reasons it is not supported?

If it is possible, I'd be willing to try and work on a PR, and test, but my C is a bit on the rusty side ;)

If anyone has notes/suggestions/docs that would help I'd really appreciate it!

mdblack98, 2023-07-07:
Try the Rotor-EZ entry and see if that  can get position.

If it does I can add it for the DCU-1.

Mike W9MDB

danieljkemp, 2023-07-07:
Ack, sorry, turns out it is actually a DCU2, closing this since I can't actually help test

Thanks for responding!

73
-Dan

closed 2023-07-07 by danieljkemp

mdblack98, 2023-07-07:
According to the DCU1 manual there are only two commands.  API and AM.  Set target and start rotation respectively.
https://www.manualslib.com/manual/1956569/Hy-Gain-Dcu-1.html