Hamlib/Hamlib #1331: BAND_UP and BAND_DOWN leads to protocol error (-8) with FT-450

ftl, 2023-07-10:
I use Hamlib 4.6~git Jun 07 12:26:39Z 2023 SHA=05cf55 64-bit on Arch Linux with a FT-450D.

When I send `vfo_op BAND_UP` or `vfo_op BAND_DOWN`, I get a protocol error (-8). 

The FT-450D sends no answer for the corresponding `BU0;` and `BD0;` commands. rigctl seems to send a `FA;` immediately after the band up/down command and seems to interpret the answer for `FA;` as the answer to the band up/down command, as you can see the log output below:

```
cmd_str = BD0;
     5:newcat.c(10796):newcat_set_cmd_validate entered
newcat_set_cmd_validate: priv->cmd_str=BD0;
write_block(): TX 4 bytes, method=2
0000    42 44 30 3b                                         BD0;            
write_block(): TX 3 bytes, method=2
0000    46 41 3b                                            FA;             
read_string_generic(): RX 11 characters, direct=1
0000    46 41 31 30 31 33 36 30 30 30 3b                    FA10136000;     
newcat_set_cmd_validate: cmd validation failed, 'BD0;'!='FA10136000;', try#1
write_block(): TX 4 bytes, method=2
0000    42 44 30 3b                                         BD0;            
write_block(): TX 3 bytes, method=2
0000    46 41 3b                                            FA;             
read_string_generic(): RX 11 characters, direct=1
0000    46 41 30 37 30 32 30 30 30 30 3b                    FA07020000;     
newcat_set_cmd_validate: cmd validation failed, 'BD0;'!='FA07020000;', try#2
write_block(): TX 4 bytes, method=2
0000    42 44 30 3b                                         BD0;            
write_block(): TX 3 bytes, method=2
0000    46 41 3b                                            FA;             
read_string_generic(): RX 11 characters, direct=1
0000    46 41 30 33 35 34 37 31 34 30 3b                    FA03547140;     
newcat_set_cmd_validate: cmd validation failed, 'BD0;'!='FA03547140;', try#3
write_block(): TX 4 bytes, method=2
0000    42 44 30 3b                                         BD0;            
write_block(): TX 3 bytes, method=2
0000    46 41 3b                                            FA;             
read_string_generic(): RX 11 characters, direct=1
0000    46 41 32 38 30 31 36 33 37 30 3b                    FA28016370;     
newcat_set_cmd_validate: cmd validation failed, 'BD0;'!='FA28016370;', try#4
write_block(): TX 4 bytes, method=2
0000    42 44 30 3b                                         BD0;            
write_block(): TX 3 bytes, method=2
0000    46 41 3b                                            FA;             
read_string_generic(): RX 11 characters, direct=1
0000    46 41 32 34 39 31 35 37 30 30 3b                    FA24915700;     
newcat_set_cmd_validate: cmd validation failed, 'BD0;'!='FA24915700;', try#5
write_block(): TX 4 bytes, method=2
0000    42 44 30 3b                                         BD0;            
write_block(): TX 3 bytes, method=2
0000    46 41 3b                                            FA;             
read_string_generic(): RX 11 characters, direct=1
0000    46 41 32 31 30 31 37 31 30 30 3b                    FA21017100;     
newcat_set_cmd_validate: cmd validation failed, 'BD0;'!='FA21017100;', try#6
write_block(): TX 4 bytes, method=2
0000    42 44 30 3b                                         BD0;            
write_block(): TX 3 bytes, method=2
0000    46 41 3b                                            FA;             
read_string_generic(): RX 11 characters, direct=1
0000    46 41 32 30 39 39 39 39 39 30 3b                    FA20999990;     
newcat_set_cmd_validate: cmd validation failed, 'BD0;'!='FA20999990;', try#7
write_block(): TX 4 bytes, method=2
0000    42 44 30 3b                                         BD0;            
write_block(): TX 3 bytes, method=2
0000    46 41 3b                                            FA;             
read_string_generic(): RX 11 characters, direct=1
0000    46 41 31 38 31 30 30 30 30 30 3b                    FA18100000;     
newcat_set_cmd_validate: cmd validation failed, 'BD0;'!='FA18100000;', try#8
     5:newcat.c(10969):newcat_set_cmd_validate returning(-8) Protocol error

```
With every retry, the FT-450D makes another band switch.

I know that band up/down has worked with an older (I think 3.x) version of hamlib, but I can't say exactly which version, I did not use my FT-450D for some time.

mdblack98, 2023-07-10:
Hopefully can compile yourself?  Should be fixed  now.
Please try the latest from https://github.com/Hamlib/Hamlib
Mike W9MDB
Message ID: ***@***.***>

mdblack98, 2023-07-11:
> That was fast, thank you @mdblack98! Unfortunately your change did not fix the whole problem. There is only one band change happening now, but I still get a protocol error:
> mutex_rigctld: client lock engaged
> mutex_rigctld: client lock disengaged
> handle_socket: doing rigctl_parse vfo_mode=0, secure=0
> rigctl_parse: called, interactive=1
> rigctl_parse: cmd=G(47) handle=7
> mutex_rigctld: client lock engaged
> rigctl(d): G 'currVFO' 'BAND_DOWN' '' ''
> rigctl_parse: vfo_opt=0
> rigctl_parse.c(3771):rigctl_vfo_op entered
> rig_parse_vfo_op called
>   3:rig.c(6483):rig_vfo_op entered
>    4:rig.c(6450):rig_has_vfo_op entered
>    4:rig.c(6458):rig_has_vfo_op returning(256)
>    4:newcat.c(7109):newcat_vfo_op entered
>     5:newcat.c(7961):newcat_set_vfo_from_alias entered
> newcat_set_vfo_from_alias: alias vfo = currVFO
>     5:newcat.c(8019):newcat_set_vfo_from_alias returning(0)
>     5:newcat.c(10991):newcat_set_cmd entered
> cmd_str = BD0;
>      6:newcat.c(10797):newcat_set_cmd_validate entered

You need to be running the new rigctld.  You are probably running /usr/bin/rigctld instead of /usr/local/bin/rigctld

Make sure you have uninstalled any hamlib packages.

Mike W9MDB

newcat_set_cmd_validate: cmd validation failed, 'BD0;'!='', try#1
     6:newcat.c(10972):newcat_set_cmd_validate returning(-8) Protocol error

newcat_set_cmd: set_cmd_validate failed
    5:newcat.c(11013):newcat_set_cmd returning(-8) Protocol error

   4:newcat.c(7193):newcat_vfo_op returning(-8) Protocol error

  3:rig_vfo_op: elapsed=79ms
  3:rig.c(6507):rig_vfo_op returning(-8) Protocol error

rigctl_parse.c(3790):rigctl_vfo_op returning2(-8) Protocol error

rigctl_parse: return#1 RPRT -8
mutex_rigctld: client lock disengaged
handle_socket: rigctl_parse retcode=-8
handle_socket: i/o error

Could you please have a look again?

ftl, 2023-07-11:
My bad, sorry. Now I built the latest sources from master and it works like a charm. Thank you very much!

73, Florian

closed 2023-07-11 by mdblack98