Hamlib/Hamlib #1337: Request : integrate my hamlib REST API

DL4OCE, 2023-07-17:
Hi folks,

I've started writing a REST API to network-enable hamlib, at least as far as rigctl and rotctl are concerned.

You can easily access a method of it by invoking an HTTP request against i.e.

http://192.168.12.10/trx/0/frequency ,

which will internally be translated to "rigctl -m {model} -p {interface} {interface parameters} f", for example. 

Of course, you can also set values as per

curl -XPOST 192.168.12.10/rigctl_api/trx/0/mode -d'{"mode":"FM","passband":9000}'

In a very early alpha, I've already implemented most functionality of these two tools, and it's already capcable of using more than one TRX / rotator.

Have a look at it at [](https://github.com/DL4OCE/hamlib_api_dl4oce) . All it takes for a test-drive is a VM or somethin as tiny as a Raspberry Pi Zero 2W.

My goal would be to enable users to use their rotators by simply supplying the rotator and the attached RasPi with power and connect it to the LAN by WiFi to avoid putting high voltage lines over the yard as well as building browser-based TRX GUIs, for example. Probably it will also be easy to convince logbook software creators to support the new, easy interface, in their programs? And son on...

Let me know if you've got any questions about this and if you're interested in integrating this with the hamlib bundle, or if I should keep it separate...

Vy 73 de Arne, DL4OCE ..._._

DL4OCE, 2023-07-17:
[image]

DL4OCE, 2023-07-17:
[image]

And it works:

> pi@hamlib:/var/www/html/hamlib_api# curl localhost/hamlib_api/trx/0/frequency
{"REQUEST_URI":"\/hamlib_api\/trx\/0\/frequency","REQUEST_BODY":"","response":{"freq":"144519490"}}

mdblack98, 2023-07-17:
We should be able to integrate this....
Just create a fork of hamlib and do a pull request when it's ready.

closed 2023-07-17 by mdblack98

DL4OCE, 2023-07-17:
Hi Mike,

I don't think it will be that easy. I didn't write the API in C++, so you won't just have another binary after compiling the source code, but as a combination of apache2 httpd and php that uses system calls to the hamlib's binaries. It needs it's own installer (can be found in the /prereqs sub-directory), and I'd really like to do some specification work with someone to have a really solid API route format in the end that doesn't need to be changed.

mdblack98, 2023-07-17:
Oh...I though you were adding the REST API in hamlib itself.
That was on my long-term to-do list.
The idea would be that you wouldn't  need a web server at all -- hamlib would  have  it's own  REST  port.  So you would just run hamlib on the remote and no webserver needed except for a GUI if wanted.That's really just a modification of what rotctld already does....just a different command format.
The REST API would be the thing to define.

mdblack98, 2023-07-17:
> Yah, sounds good. Too bad I'm completely unexperienced in writing socket-based TCP servers in Cpp (which is a matter of hours in php). ;-)
>
> But I get the idea, and I think it's a good idea to have another binary that takes care of HTTP requests or even just a switch that enables the API mode in every hamlib module.
>
> For the API routes part, I invite you to steal the format I already defined in usage.html in the project.
>
> But as my API is nearly 95 % implemented, already, I really wouldn't want to just dump all the work. There's also more reasons why I'd chose something like a "real" web technology just in case people would like to add encryption and/or authentication to it.
>
> Two different approaches. Probably you'd like to integrate the php hamlib REST API as a part of hamlib, temporarily? Until someone's written something better?

You can always wrap a web server around a REST API.  No need for the REST side of the house to have encryption or security.
That way the web server can run locally or remote.
Are you talking to rotctld with your code?

Encryption would be pretty useless here IMHO.

Would you like to use https://www.postman.com/ 
to develop/document the API?

Mike W9MDB

mdblack98, 2023-07-17:
> Until today, I'm talking to rigctl and rotctl. These methods are already working:
>
> // ==== rigctl ====
>
> curl localhost/rigctl_api/trx/0/frequency
>
> curl -XPOST localhost/rigctl_api/trx/0/frequency -d'{"newValue":"145425000"}'
>
> curl localhost/rigctl_api/trx/0/mode
>
> curl -XPOST localhost/rigctl_api/trx/0/mode -d'{"mode":"FM","passband":9000}'
>
> curl -XPOST localhost/rigctl_api/trx/0/mode -d'{"mode":"USB","passband":2400}'
>
> curl localhost/rigctl_api/trx/0/split_frequency
> curl -XPOST localhost/rigctl_api/trx/0/split_frequency -d'{"newValue":145725100}'
>
> curl localhost/rigctl_api/trx/0/split_mode
> curl -XPOST localhost/rigctl_api/trx/0/split_mode -d'{"mode":"FM","passband":9000}'
>
> curl localhost/rigctl_api/trx/0/split_frequency_mode
> curl -XPOST localhost/rigctl_api/trx/0/split_frequency_mode -d'{"frequency":145425300, "mode":"FM","passband":9000}'
>
> curl localhost/rigctl_api/trx/0/split_vfo
> curl -XPOST localhost/rigctl_api/trx/0/split_vfo -d'{"split_mode":1,"tx_vfo":"VFOA"}'
> [... 150 lines cut ...]
> curl localhost/rigctl_api/rotator/0/state
>
> curl localhost/rigctl_api/rotator/0/capabilities
>
> No plans on documenting, so far, no. One thing I'm thinking about is creating an OpenAPI documentation that enables the user to test-drive the methods, but honestly, there should be more important front-ends like logbook programs, contest rig control software, browser-based TRX GUIs and son on, as soon as this thing is stable and accepted..

Are you using a pipe to rigctl/rotctl?  Or running them every time you send a command?

Mike W9MDB

DL4OCE, 2023-07-17:
That's one of the disadvantages of stateless HTTP: keeping a connection alive is not something you'd recommend :-D

Instead, each time you call my script

curl -XPOST localhost/rigctl_api/trx/0/frequency -d'{"newValue":"145425000"}' ,

it performs a sys-call similar to

/usr/bin/rigctl -m 1020 -p /dev/ttyUSB0 F 145425000 

That makes things a bit slow. I could also switch to rigctld, does that make more sense?

mdblack98, 2023-07-17:
rigctld would be a LOT faster.

Some rigs have a fair amount of overhead when opening them.

You really don't want the overhead of a TCP connection to rigctld either if you can avoid it but it's not too bad.

time echo "f " | nc -W 1 localhost 4532
145000000

real    0m0.022s
user    0m0.000s
sys    0m0.002s

The NodeRed people went through this same exercise and found keeping a pipe open to rigctld was the best solution.

Mike W9MDB

DL4OCE, 2023-07-17:
Now I remember, how I improved the speed of my QO-100 downlink syncing tool :-D 

Yes, connecting to the daemon is waaaaay faster, especially on my FT817 I'm using for testing, at the moment! Thanks!

DL4OCE, 2024-08-28:
Hi Michael,

as I'd like to implement that REST API of mine as part of the hamlib project, could you probably point me at a REST API library that fits your requirements in terms of complexity, performance, features, licence, ...? There's so many out there, I gave up searching for the best one 😕

Requirements from my side would be

  *
at least HTTP methods GET and POST
  *
route parsing (i.e. the route of POST http://server<http://server/hamlib/rigctl/qrg?frequency=145425000>/hamlib/rigctl/qrg?frequency=145425000<http://server/hamlib/rigctl/qrg?frequency=145425000> would be the bold printed portion)
  *
handling JSON (could be used from another library)

BR

Arne, DL4OCE    ..._._

mdblack98, 2024-08-30:
Hamlib is written in C (with some addtional side code in C++ but I do not want to depend on C++ at all)
Take a look at libcurl