Hamlib/Hamlib #1338: Rig Control Error Connecting to FT-891

KB3MDD, 2023-07-18:
I am trying to transmit using WSJT-X (v2.6.0), using FT-891; However, when I press "Test CAT" radio is stuck on transmit and eventually I get HamLib Error. 

I am using good quality USB (High Speed Ferret Choke) and DigiRig. Radio is new.

I followed Digital mode setting for FT-891/WSJT-X (https://www.jpole-antenna.com/2020/08/12/yaesu-ft-891-setup-for-digital-modes/)

mdblack98, 2023-07-18:
Sounds like RFI Problems due to noise on the USB cable

Tests
If problems are occurring only during transmit:
    #1 Reduce power to zero and see if the problem stops -- if it does stop than it is definitely RFI.  You will see certain higher power levels on certain bands that cause problems.
Then, if problems are occurring during non-transmit periods it indicates a system problem with USB devices so...
    #1 Check USB Power Management option is turned off on all USB devices
        Device Manager for Windows.
        For Linux set autosuspend=-1 https://docs.kernel.org/driver-api/usb/power-management.html

RFI Fixes:
    #1 Free - Move USB cables to another port -- some ports are more susceptible than others.
    #2 Free -- Check your grounding system.  rod-outside-the-shack is a common problem when it's not bonded to the main house ground. 
        Common grounding mistakes, sources, and solutions:
        A. Ground rod outside the shack that is not bonded to the main house ground.
        B. Shack equipment bonded incorrectly (e.g. daisy chained instead of common ground point)
        C. Desktop computer grounded to the house ground and not the shack ground.  Run a separate RF ground from the computer chassis to your station RF ground.
           For a laptop use the retaining screw of a DB9 or DB25 connector shell, if your device still has them.
        D. Ethernet cables that bring RFI into the computer...which then ends up going to the rig too since the ethernet shield is tied to the case which is tied to USB shield which is tied to pin 4 on the USB cable (a very common problem on most all USB devices -- see my QRZ page).
           Ethernet patch cables up through CAT6 are UTP, which stands for UNSHIELDED Twisted Pairs, four to be specific.  There is NO separate shield conductor in the jacket, nor a metallic shield around the RJ45 connector itself.
           Just use a ferrite toroid at each end.
        E. Wall warts -- 24VAC supplies in sprinkler and alarm systems are notorious for picking up RFI into your electrical system.
           24 VAC transformers can be RF-bypassed using .005 ufd caps from each output lead to safety ground. You can often use the cover plate mounting screw as your ground connection.
        F. Speaker wires The same approach as E also works for external speaker audio leads.
        G. Lamps (yes...lamps around the house have unshielded wires as do many other appliances).
        H. Washer/Dryers are notorious for generating and picking up RFI.  In general, newer high-efficiency models have more RF problems.  
           Ferrite toroids INSIDE the appliance housing can work wonders if the wiring harness has connectors in the AC line input, OR an external noise filter for the AC line cord of a washing machine can reduce RF spurs by 25 dB or more.
        I. HVAC systems with variable speed blower control systems both cause RF noise and react badly to RF fields -- we believe adding torroids inside the unit on the power lines will work.
        J. If you use a powered USB expansion hub, add a ferrite toroid on the cable coming from the USB power supply.
        K. SignaLink -- You can ground the metal box shell by simply wrapping an 18ga wire (or use a small crimped ring or spade terminal) under the head of any of the screws holding the rear panel, then connect to your station RF ground. 
           The case is isolated from both USB and analog audio signal grounds, so this does not affect use of the USB shield isolators.
        L. DC power supply -- both linear and switching -- READ THE PS MANUAL FIRST!  This step may void some manufacturers' warranty and UL/CSA approvals. 
           Remove any jumpers between the DC negative output lead and PS chassis or line cord ground  Add a .005 ufd cap from each DC output lead to chassis ground if not already there. 
           NOTE: Samlex DC outputs are already isolated and bypassed, but many others, including Astron, may randomly have the negative side grounded and no RF bypassing. 
        B through L may all need chokes.    
        http://www.k9yc.com/GroundingAndAudio.pdf 
    #3 Free -- start unplugging devices around the house and see if there's one device that is acting as a bad source of RFI.  This presupposes you can easily repeat the problem on your rig setup.
    #4 Cheap -- Add some USB shield isolators (see my QRZ page).  I use one on my SignaLink for example.
    #5 Minimal $$ -- Good USB cables like this
    https://www.amazon.ca/Tripp-U023-006-Device-Ferrite-Chokes/dp/B003MQ29B2/ref=sr_1_5?crid=11YRNPWDVWGCU&keywords=usb+cable+with+choke&qid=1658187349&sprefix=usb+cable+with+choke%2Caps%2C139&sr=8-5
    #6 Maybe free (if you have chokes...otherwise can get a bit costly) -- add chokes to USB cables first, then all other cables including power, ethernet, and control cables.
    Fair-Rite torroids are good quality -- do NOT buy cheap Chinese ones --  https://www.fair-rite.com/product/toroids-5943003801/  You can use clip-ons but torroids allow multiple wraps and give better results.
    https://www.fair-rite.com/product/round-cable-snap-its-431176451/
    I couldn't find type 31 torroids at Fair-Rite as of 20220721 but Palomar has some palomar-engineers.com/ferrite-products/ferrite-cores/ferrite-ring-toroid-combo-pack/
    Newark also carries them https://newark.com/c/passive-components/emc-rfi-suppression/ferrites-ferrite-assortments?brand=fair-rite
    RFI Problems

Mike W9MDB

KB3MDD, 2023-07-20:
Thanks for the reply!

I tried following things:

1. Reduced power to 5 W.
2. Using high quality USB cable (gold plated).

still I get following errors during CAT Test:

Hamlib error: write_block(): TX 3 bytes, method=2
0000    46 41 3b                                            FA;             
read_string_generic called, rxmax=129 direct=1, expected_len=1
read_string_generic(): Timed out 2.119 seconds after 0 chars, direct=1
rig_flush: called for serial device
read_string_generic called, rxmax=4095 direct=1, expected_len=1
cmd_str = FA;
write_block():1079 failed 0 - No error
   3:newcat.c(10568):newcat_get_cmd returning(-6) IO error

  2:newcat.c(1418):newcat_get_freq returning(-6) IO error

rig_get_freq(2103): freqMainA=0, modeMainA=, widthMainA=0
rig_get_freq(2103): freqMainB=0, modeMainB=, widthMainB=0
rig_get_cache(259):  vfo=VFOA, current_vfo=VFOA
rig_get_cache(435): vfo=VFOA, freq=0, mode=, width=0
rig_set_cache_freq(122):  vfo=VFOA, current_vfo=VFOA
 1:rig_get_freq: elapsed=6382ms
IO error
IO error
 while getting current frequency

Timestamp: 2023-07-20T18:08:54.617Z

mdblack98, 2023-07-20:
Please contact me off list
***@***.***
Mike W9MDB

 
 

   
   - Reduced power to 5 W.

KB3MDD, 2023-07-21:
Nope!

they were exactly how you said.

What settings FT-891 need? I would like start over. Never even once I managed to CAT control my RIG.

I am using WSJTX latest version and DigiRig-FT-8XX

mdblack98, 2023-07-21:
Please contact me directly and we can hook up using AnyDesk https://anydesk.com/en/downloads/windows?dv=win_exe
Your email address is not on qrz.com so don't know how to contact you directly.
Mike W9MDB

KB3MDD, 2023-07-21:
my email is prasad.saravate at the rate gmail dot com

mdblack98, 2023-07-21:
Was using the wrong COM port.
FT-891 using DigiRig, RTS keying to DigiRig, Enhanced port for CAT, USB mode as we could not get USB-D to receive at all.

closed 2023-07-21 by mdblack98