Hamlib/Hamlib #1340: C++ Rig Init fails

eriktunsch, 2023-07-19:
Hey, 

when I am using the rigclass.cc, during the init-process (Rig myRig{ RIG_MODEL_IC7300 };), the process crashes. When debugging the libhamlib-4.dll has a access violation in the funtion rig_init()

Hope somebody can help.

Erik

mdblack98, 2023-07-19:
Have you got some minimal code to demonstrate this?

It  works here.

Mike W9MDB

mdblack98, 2023-07-19:
> I am using the example c++ code:
>
> /*
>
>     * Hamlib sample C++ program
> */
>
> #include
> #include <hamlib/rigclass.h>
>
> int main(int argc, char* argv[])
> {

Linux or Windows?

It works here on Linux.

g++ -g -o testcpp testcpp.cc -lhamlib++

gdb rigtest
(gdb) run

Mike W9MDB

try {
    myRig.setConf("rig_pathname", "COM8");
    myRig.open();
    myRig.setFreq(MHz(144));
    std::cout << myRig.getLevelI(RIG_LEVEL_STRENGTH) << "dB" << std::endl;
    std::cout << "Modes for freq 14.332: " << myRig.RngRxModes(MHz(14.332)) << std::endl;
    myRig.close();
}
catch (const RigException& Ex) {
    Ex.print();
    return 1;
}

return 0;

}

with the rigclass.cc

eriktunsch, 2023-07-19:
I am running windows with msvc compiler.

mdblack98, 2023-07-19:
Do you have a project you can share?

Have you tried stepping though the debugger?

Mike W9MDB

eriktunsch, 2023-07-22:
I found out, it needed to be compiled by gcc.

Thank You!

closed 2023-07-22 by eriktunsch