Hamlib/Hamlib #1343: Connexion with Radio with a RTS interface always activate PTT for a second

magellan-13016, 2023-07-23:
Hi, 
I own a Yaesu FT-847 with the new DIGIMODE interface. To connect my computer to this new interface, I have to use this command : 

rigctld -m 1001 -r /dev/ttyUSB.YAESU-DIGIMODE -t 4532 -s 57600 -P RTS --set-conf=stop_bits=2,serial_handshake=None

the problem is that every time the connexion is done, the transceiver switch to PTT for a second. 

Currently I'm makink a software to check the PTT status and for that, I have to use this command every second:

rigctl -m 1001 -r /dev/ttyUSB.YAESU-DIGIMODE -P RTS t

But every second my radio activate the PTT. How can I avoid this ?

Thanks for your help.

David.

mdblack98, 2023-07-23:
Linux toggles the RTS and DTR ports when opened and may also do it when the port is close...I'm not sure about the close.
If you just keep rigctld running
rigctld -m 1001 -r /dev/ttyUSB.YAESU-DIGIMODE -t 4532 -s 57600 -P RTS --set-conf=stop_bits=2,serial_handshake=None
You should be able to do this:
rigctl -m 2 t
Mike W9MDB

magellan-13016, 2023-07-23:
Many thanks mdblack98 ! It works !

Thanks for having responding so quickly !

Best regards, David.

closed 2023-07-23 by mdblack98

finlstrm, 2023-10-27:
I'm using a DigiRig with my FT-847. The above solution does work, however it still chatters the PTT on startup, not good enough for me. I found larsks developed this https://github.com/larsks/force_rts (yesterday actually) which overrides the serial open() to force the RTS state. Works great on Debian Stable 12.2, no more erroneous PTT.

Using force_rts I'm able to use the following command no problem `rigctld --model=1001 --rig-file=/dev/ttyUSB0 -s 57600 --listen-addr=127.0.0.1 --port=5203 --ptt-type=RTS` and everything seems to work as expected.

magellan-13016, 2023-10-28:
Many thanks Michael ! I will try this ! I'm rebuilding my website and I will give this information.
www.f4bpp.com

Best regards. David.