Hamlib/Hamlib #1345: rigctld FlexRadio 6xxx (2036)  not working

wa1gon, 2023-07-25:
I am getting an error when trying to use rigctld:

PS C:\Users\wa1go> rigctld -m 2036 -s 19200 -r COM4 -t 4532 -vvv
Opened rig model 2036, '6xxx'
rig_open: error = Max# of rigctld client services=32
rig_check_rig_caps: p1=0000000060211CE0, p2=0000000060218258, rig_model=0000000060211CE0, macro_name=0000000060218258
rig_init: rig_model=FlexRadio 6xxx
rig_init: rig has VFO_A
rig_init: rig has VFO_B
kenwood_init called, version 20230318/20220306.0
kenwood_init: if_len = 37
kenwood.c(806):kenwood_init returning2(0)
main: twiddle=0, uplink=0, twiddle_rit=0
 1:rig.c(817):rig_open entered
rig_settings_get_path: path=.hamlib_settings
rig_settings_load_all: settings_file (.hamlib_settings): No such file or directory
rig_open: cwd=C:\Users\wa1go
rig_open: C:\Users\wa1go/hamlib_settings does not exist
rig_open: async_data_enable=1, async_data_supported=0
network_init: WSAStartup OK
rig_open: rs->comm_state==0?=0
 1:rig.c(1023):rig_open returning(-1) Invalid parameter

Invalid parameter
 COM4 No such file or directory
handle_socket: rig_open reopened retcode=-1
handle_socket: i/o error
handle_socket: rig_close retcode=-1
handle_socket: rig_open retcode=-1, opened=0

**However, if I use Kenwood TS-2000** 

PS C:\Users\wa1go> rigctld -m 2014 -s 19200 -r COM4 -t 4532 -vvv
kenwood_transaction: Unknown command or rig busy 'FB'
read_string_generic(): Timed out 0.309 seconds after 0 chars, direct=1
kenwood_transaction: read_string retval < 0, retval = -5, retry_read=1, rs->rigport.retry=1
Opened rig model 2014, 'TS-2000'

All the commands I tried worked.

wa1gon, 2023-07-25:
Oh, I forgot the version info:

hamlib Latest:
OS: Windows 11
SmartSDR: v3.4.24

mdblack98, 2023-07-25:
This should work -- The Flex is a Network-based interface.
rigctld -m 2036 -vvv

It connectsby default to 127.0.0.1:4532
If the flex is on another machine or port like port 6000
rigctld -m 2036 -r 192.168.1.1:6000 -vvv
Mike W9MDB

wa1gon, 2023-07-25:
Ok,  I got it working.  Thank you!

Why is the default flex port the same as the rigctld listening port?  Clearly, that would never work.

Thank you very much for all your and the hamlib team work!!!!

closed 2023-07-25 by wa1gon

mdblack98, 2023-07-25:
You would have ra sask flex. Hamlib has had that port for 20+ years.Sent from my iPhone

Closed #1345 as completed.