Hamlib/Hamlib #1352: Rig port_type in case of SERIAL/NETWORK Rig

foldynl, 2023-08-03:
Is there any way  how to distinguish whether a given rig is `SERIAL`-only, `NETWORK`-Only or `SERIAL/NETWORK` ? 

An example is this

```
    .model_name = "PiHPSDR",
    .mfg_name =  "OpenHPSDR",
    .version =  BACKEND_VER ".1",
    .copyright =  "LGPL",
    .status =  RIG_STATUS_STABLE,
    .rig_type =  RIG_TYPE_TRANSCEIVER,
    .ptt_type =  RIG_PTT_RIG,
    .dcd_type =  RIG_DCD_RIG,
    .port_type =  RIG_PORT_SERIAL,
```
The rig can be connected via `SERIAL` port and `NETWORK` but .port_type  can contain only one type - in this case SERIAL port.

mdblack98, 2023-08-03:
Not currently -- why do you need to know?  What are you trying to accomplish?

The only difference (usually) is that you use an address:port instead of a com port.

Mike W9MDB

foldynl, 2023-08-03:
I am [QLog ](https://github.com/foldynl/qlog) maintainer and  I would need to know what type the rig is - if serial or network or both - because I use a different Setting Dialog layout for each type.

So far I've been doing it according to port_type, but that's unreliable. That's why I'm looking for how to distinguish this. It seems to me that it is not user-friendly to display the serial port settings when the rig is only network-attached. In the case of the serial and network port, I would like to be able to inform that both types can be used.

mdblack98, 2023-08-03:
I would recommend you let the user choose which interface they want.

In Hamlib there is no difference between serial and network serial.

You just give an IP:PORT instead of COMX.

More and more things are adding serial-over-network so letting the user choose which GUI they want is the most future-compatible solution.

You could just as well open up the COM port box to allow 192.168.1.1:13013 for example which works.  That's the way WSJT-X does it.  The other settings are ignored.
You could recognize an IP quad and disable the serial port options.

Mike W9MDB

foldynl, 2023-08-03:
Ok. It can be a solution to let the user choose but how I can internally validate that the selected connection is supported. for example an selection of ic7300 and network connection type does not make sense if rigctld is not running there.

Ok.Your answer is clear for me - no api is available for this.

mikaelnousiainen, 2023-08-03:
Just an idea: What if you just added a separate model (with all the same caps), but one that has port type of network? Yes, a bit wasteful, but follows the abstraction of models in Hamlib and is backwards compatible. Obviously the names should indicate somehow a network vs serial based model.

mdblack98, 2023-08-03:
We also have serial-over-ethernet that can work with any rig.
So the real solution is let the user choose.
If one wants to make the GUI prettier than make it a checkbox or selection.
WSJT-X doesn't worry about making it prettier -- just let's anything be typed into the COM port box.

closed 2023-08-03 by mdblack98