Hamlib/Hamlib #1359: Hamlib 4.6 connection to IC9700

hb9gfx, 2023-08-08:
After installing HamLib 4.6 together with JTDX improved the connection from hamlib - IC-9700 does not wokr anymore.
It need 2 or more trys to get a connect to the TRX.

mdblack98, 2023-08-08:
Does that mean you are seeing an error message?

What version of hamlib?

I just installed  this one and it works fine out-of-the-box with an IC-9700.

https://phoenixnap.dl.sourceforge.net/project/jtdx-improved/jtdx_2.2.159/jtdx-2.2.159-win64_improved.exe

I then installed the latest  Hamlib DLL from https://n0nb.users.sourceforge.net/ and it still works

What does your Radio config page look like?

I have

Baud 115200
Data Bits Default
Stop Bits Default
Handshake Default
DTR/RTS blank
PTT CAT
Mode Data/Pkt
Split Rig

Mike W9MDB

mdblack98, 2023-08-08:
Also please try WSJT-X improved and see if you get the same error.

At least we can debug that one better.

Mike W9MDB

hb9gfx, 2023-08-08:
Hi did this already.

 

Same issue.

 

73 de stefan

hb9gfx, 2023-08-08:
It is Hamlib 4.6 as Uwe DG2YCB told me 

 

https://phoenixnap.dl.sourceforge.net/project/jtdx-improved/jtdx_2.2.159/jtdx-2.2.159-win64_improved.exe

mdblack98, 2023-08-08:
For WSJT-X (and improved versions)

Please place this file as described below
https://www.dropbox.com/s/t52ngcalsgnpm8m/wsjtx_log_config.ini?dl=0

    C:\Users\[username]\AppData\Local\WSJT-X
    The WSJT-X_Rigcontrol.log file will be in the same location

For Linux put it in  
    ~/.config
    The WSJT-X_Rigcontrol.log file will be here:
    ~/.local/share/WSJT-X

For MacOS put it in
    /Users/[username]/Library/Preferences
 
Restart WSJT-X and duplicate the problem.
Shut down WSJT-X

Then send me the WSJT-X_RigControl.log file
Mike W9MDB

https://phoenixnap.dl.sourceforge.net/project/jtdx-improved/jtdx_2.2.159/jtdx-2.2.159-win64_improved.exe

mdblack98, 2023-08-18:
No response so assume all is well...closing ticket.

closed 2023-08-18 by mdblack98