Hamlib/Hamlib #1366: Enhancement: Implement stop_morse for Elecraft K4

ny4i, 2023-08-18:
I am not quite ready yet to create a patch for hamlib but looking at the code, implementing the kenwood_stop_morse but the K4 version would appear to work to stop CW on a K4 (and a K3 if that is not yet implemented). I'm still figuring out how the Kenwood function and the K3/K4 functions work together.

The K3 command KY <ASCII 04> stops the CW immediately on both the K3 and the K4.

ny4i, 2023-08-21:
stop_morse does not seem to be working. I sent bABCDEFGHIJK and then right away sent ,\stop_morse VFOA but the CW did not stop. I then did a test where I did sent ,bABCDEFGHIJK then the next command I sent was ,b@ (where @ is the command in the K4 Programmers Reference that stops CW immediately. That worked.

It appears from the trace that hamlib sends KY1; to stop the morse. That is not a documented way to stop the morse. You would send KY |; or KY @; On the K3, it is \0x04 so KY<space><hex 04>;

[stop_morse-1.txt](https://github.com/Hamlib/Hamlib/files/12399448/stop_morse-1.txt)

ny4i, 2023-08-21:
I checked the code and there is just one thing missing. You need a space between the KY and the @ or hex 04. That should do it though. I tried to put that that in the above text but I may not have been clear.

mdblack98, 2023-09-30:
Is this working OK now?

ny4i, 2023-09-30:
Yes. Thank you

closed 2023-09-30 by mdblack98