Hamlib/Hamlib #1370: GUI for Hamlib

mdblack98, 2023-08-20:
Test .Net MAUI to see if we can create a portable GUI for Hamlib.

N0NB, 2023-08-21:
Is there a GPL/LGPL compatible version of .NET now?

mdblack98, 2023-08-22:
MIT License
https://github.com/dotnet/dotnet

We wouldn't be distributing .Net though anyways -- user would need to install it.

Mike W9MDB

N0NB, 2023-08-23:
Sounds like an interesting project.  I was unaware MS had released .NET under MIT.

AuroraRAS, 2023-10-01:
How about using Qt framework?

mdblack98, 2023-10-01:
Qt has broken all sorts of things and I'm looking more to the future with C# and dotnet abilities.

mdblack98, 2023-10-01:
I opened up my test code -- this is my learning project so may not be the final code repository
You need to add two multicast options with rigctld to test it
rigctld -m 3073 -r /dev/ttyUSB0 -s 115200 --vfo -Z -vvvvv --multicast-addr=224.0.0.1 --multicast-port=4532

Theoretically should be able to do this in MyAppdotnet run
Would have to install dotnet though on any platform to test.  
Mike W9MDB
Message ID: ***@***.***>

mdblack98, 2023-10-01:
Oops...here's the code
https://github.com/mdblack98/MyApp

N0NB, 2025-07-31:
Perhaps more useful for anyone interested would be to port [grig](https://github.com/fillods/grig) to GTK3 or GTK4 with libadwaita, or to use it as inspiration.