Hamlib/Hamlib #1373: include/hamlib/config.h in the tarball breaks readline support

df7cb, 2023-08-26:
Hi,

#947 shuffled the config.h header around so "HAVE_PTHREAD is available if needed".

This causes a chain of events:
* hamlib/config.h was added to nobase_include_HEADERS in include/Makefile.am
* the `make dist` tarball now contains a static `include/hamlib/config.h`
* when doing an out-of-tree (VPATH) build, `build/include/hamlib.config.h` is generated
* the `-I` flags on the compiler command line list the top source include directory before any out-of-tree build directory
* since the tarball config.h is preferred over the config.h generated by ./configure, none of the non-default flags in there is visible to .c files
* `test/.libs/rigctl` and `test/.libs/rotctl` aren't using libreadline anymore (note that the compiler command line still has `-lreadline`, but the `#ifdef` in the file doesn't trigger)
```
/bin/bash ../libtool  --tag=CC   --mode=link gcc -Wall  -I../src -I../security -g -O2 -ffile-prefix-map=/<<PKGBUILDDIR>>=. -fstack-protector-strong -Wformat -Werror=format-security   -Wl,-z,relro -Wl,-z,now -o rigctld rigctld-rigctld.o rigctld-rigctl_parse.o rigctld-dumpcaps.o   ../src/libhamlib.la ../lib/libmisc.la -ldl -lreadline
```
* the Debian libhamlib-utils package doesn't even depend on libreadline anymore

TBH I don't see why 9aade8de3f52 had to add config.h to nobase_include_HEADERS - the net patch there seems to be
```
-#ifdef _PTHREAD_H
+#ifdef HAVE_PTHREAD
```
and that should have worked without forcing config.h to be included in `make install` or `make dist`. Is that file of any use on end-user systems?

I think the fix is simply reverting the include/Makefile.am part:

```
diff --git a/include/Makefile.am b/include/Makefile.am
index 324c082f4..bba8cdc13 100644
--- a/include/Makefile.am
+++ b/include/Makefile.am
@@ -3,4 +3,4 @@ noinst_HEADERS = bandplan.h num_stdio.h
 nobase_include_HEADERS = hamlib/rig.h hamlib/riglist.h hamlib/rig_dll.h \
 		hamlib/rotator.h hamlib/rotlist.h hamlib/rigclass.h \
 		hamlib/rotclass.h hamlib/amplifier.h hamlib/amplist.h \
-		hamlib/ampclass.h hamlib/config.h hamlib/multicast.h
+		hamlib/ampclass.h hamlib/multicast.h
```

Does that make sense? If yes, I can throw it in as pull request.

gvanem, 2023-12-09:
Why is `HAVE_PTHREAD` still around? A lot of stuff does not compile w/o it.
And on Windows (with Pthreads-W32), stuff like `if (async_data_handler_priv->thread_id != 0)`
does not compile; `error: invalid operands to binary expression ('pthread_t' (aka '__ptw32_handle_t') and 'int'))`

Have to use macros to cast around that:
```c
  #if defined(_WIN32)
    #define PTHREAD_ID(pt_var)          (*(unsigned long*)&(pt_var))
    #define PTHREAD_ID_CLEAR(pt_var)    memset (&(pt_var), '\0', sizeof(pt_var))
  #else
    #define PTHREAD_ID(pt_var)          pt_var
    #define PTHREAD_ID_CLEAR(pt_var)    PTHREAD_ID(pt_var) = 0
  #endif
```

mdblack98, 2023-12-09:
What doesn't compile without it?

gvanem, 2023-12-09:
> What doesn't compile without it?

E.g. in `src/rig.c`, `async_data_handler_start()` is only defined inside a `HAVE_PTHREAD` section. A terrible mess.
Please try it.

gvanem, 2023-12-09:
And w/o a `HAVE_PTHREAD`, a link of `Hamlib*.dll` fails with these unresolved external symbols:
```
async_data_handler_start
async_data_handler_stop
morse_data_handler_start
morse_data_handler_stop
network_multicast_publisher_start
network_multicast_publisher_stop
network_multicast_receiver_start
network_multicast_receiver_stop
network_publish_rig_spectrum_data
network_publish_rig_transceive_data
pthread_create
pthread_exit
pthread_join
pthread_mutex_lock
pthread_mutex_unlock
rig_poll_routine_start
rig_poll_routine_stop
```

mdblack98, 2023-12-09:
What platform is this on?

gvanem, 2023-12-09:
Windows.

mdblack98, 2023-12-09:
This is using Visual C or such?
Can you send me a project that does not compile correctly?
Mike W9MDB

gvanem, 2023-12-09:
> This is using Visual C or such?

No, off-course not. As you should be aware of, MSVC is unable to compile Hamlib due to all the GCC-centric stuff.
I was using clang-cl.

> Can you send me a project that does not compile correctly?

Project? To me that implied some kind of `.vcxproj` files. There are none AFAIK. I used a home-made [GNU-makefile](https://gist.github.com/gvanem/f54eefad9e84954d390c8bd939ec34ae).

mdblack98, 2023-12-09:
So this is MacOS? -- cross compiling or such?

gvanem, 2023-12-09:
I give up.

mdblack98, 2023-12-09:
I need to be able to duplicate the problem in order to get a decent fix.Just tell me how to duplicate the problem.  
Do you have a clang example?

gvanem, 2023-12-09:
> .Just tell me how to duplicate the problem.  

For starters, try to remove `-DHAVE_PTHREAD` from your built.

closed 2023-12-30 by mdblack98