Hamlib/Hamlib #1380: TS-890S/990S change IF/MD command needed

mdblack98, 2023-09-12:
From Kenwood...

The "IF" and "MD" commands are implemented in both the TS-890S and TS-990S.

This is for compatibility reasons, such as the Sky command operations.

However, the parameters of these commands do not support new modes (such as PSK or DATA), or cross-mode split operation.

Therefore, the "PC CONTROL COMMAND Reference Guide" booklets of the TS-890S and TS-990S do not disclose "IF" or "MD", and recommend "SF" instead of "IF" and "OM" instead of "MD".

GeoBaltz, 2024-11-06:
Some updates - 
For the current Kenwood HF product line (TS-590SG/TS-890S/TS-990S)...
The 590 and 890 show the SF command - the 990 does not.  590 support varies with F/W version.
The 590 shows the MD command - the 890 and 990 do not document it, but it works on the 890(all modes).
The 590 also documents IF - on the 890 it works, but some missing data.
The 890 and 990 show OM.

As far as Hamlib is concerned, the MD command is no longer used on the 890 AFAICT.  I think it is gone for the 990, but not sure.
The 890 uses the SF command for most ops, but is still dependent on IF for others.
AFAICT OM is only used on the 990.

Is there any doc for the 990 SF?

Note that this issue is almost a dupe of #899.

mdblack98, 2024-11-06:
What is a 990SF?  I can't find any information on it.

Most of this is a duplicate of the other issue -- seems to me it should be combined with specifics on what firmware version is needed to support things on these commands.

Mike W9MDB

GeoBaltz, 2024-11-06:
The 990 SF command - it's not in the TS-990S PC CONTROL COMMAND Reference Guide.

mdblack98, 2024-11-06:
I just queried Kenwood about this since we have information on the other issue that it is theoretically supported.
Do you want to implement this for the 890 keeping the 990S as an option?

GeoBaltz, 2024-11-06:
With the documentation it should be pretty straightforward to bend the 890 SF code to work with the 990(I hope).
As far as getting rid of the IF dependency, there are about 15 calls to kenwood_get_if() that need to be investigated - I'll start looking, keeping both models in mind.
The OM command opens up a whole new can of worms(see issue #1291).  The 890 and 990 are very similar in function, but Kenwood's terminology confuses the issue between Main/Sub in the 990 and both A/B and right/left in the 890. Maybe in the future...

GeoBaltz, 2024-11-07:
And here I thought this would be easy...

Did the first few kenwood_get_if() calls.  The RIT/XIT stuff is much simpler than before.

But how do you find out if the rig is XMITting?  I don't see any way to just read the RX/TX state except P8 of the IF command.  There has to be something I'm missing, but what?