Hamlib/Hamlib #1383: Kenwood, Auto Information (AI), push of frequency change

akrpic77, 2023-09-15:
Hello, first, let me thank you for a nice piece of software. 

I use Kenwood transceivers, and looking at the "PC CONTROL COMMAND" manual I see an "AI", an Auto Information feature, which reports on changes of the frequency and other parameters (tested).

I connect to radio using TCP/IP via rigctld, and I have to poll it for frequency... or am I missing something?

How can I convince rigctld to push me changes of the radio (currently interested in frequency only). I searched the code for "AI*" commands and they are mentioned, though I couldn't figure out the if the push mechanism is there.

mdblack98, 2023-09-15:
What rig do you have?  The AI function is not very stable from all my testing.There are a few Yaesu rigs that work well but most of the rigs I've tested don't work well.
What are you trying to do that you need frequency pushed instead of polled?
Mike W9MDB

akrpic77, 2023-09-16:
I am using TS-590 and the AI feature so far  works solid. I am using UDP to Serial [bridge](https://github.com/patrickelectric/bridges).

I plan to control multiple rigs and and have as little latency as possible when changing bands. Only way to detect band change is to read frequency. I was way down to polling at 250ms interval, but doing that unnecessarily puts strain on the system.

akrpic77, 2023-09-23:
Maybe the role of Hamlib would be (already is?) to work around these quirks in hardware and unify experience across different types of rigs.

Currently I send “AI2;FA;” (auto information ON, read frequency) commands every second. On missing response for three seconds I declare rig unreachable.

This strategy so far works as expected.

We can close this issue as there is no broken functionality in the Hamlib regarding this, I just see Hamlib in a perfect position to provide such functionality.

mdblack98, 2023-09-23:
Currently working on multicast for Hamlib.  Partially working at this point but polling needs to be added.
This will send out immediate  messages when certain items change like freq, mode, ptt and perhaps others.
You can follow this issue to see updates....
https://github.com/Hamlib/Hamlib/issues/695

closed 2023-09-23 by mdblack98