Hamlib/Hamlib #1387: rigctld does not work with network port other than the default.

srs4511351, 2023-09-23:
I am using a Raspberry Pi 4 with CubicSDR and gqrx.
With hamlib version 4.5.5, I can use CubicSDR with Hamlib NET rigctl using rigctld:
`rigctld -v -m 1 -T 127.0.0.1 -t 7356`

In order to resolve an issue with gqrx giving an get_powerstat error, I installed the latest using git clone.
rigctld Hamlib 4.6~git 2023-09-20T22:35:29Z SHA=6644af 64-bit

Now, CubicSDR does not work with port 7356 using the same command. It does work with the default port 4532.
It is apparent that rigctld is ignoring the -t 7356 switch.

Just for fun, I tried model 2. CubicSDR timed out.
`rigctld -v -m 2 -T 127.0.0.1 -t 7356`
```
[I pressed ^C]
rig_open: error = rig_init: rig_model=Hamlib NET rigctl 20230802.0
rig_init: rx_range_list1 is empty, using rx_range_list2
rig_init: rig does not have tx_range!!
netrigctl_init version 20230802.0
set_conf: called
main: twiddle=0, uplink=0, twiddle_rit=0
rig.c(888):rig_open entered
rig_settings_get_path: path=/home/pi/.config/hamlib_settings
rig_settings_load_all: settings_file (/home/pi/.config/hamlib_settings): No such file or directory
rig_open: cwd=/home/pi
rig_open: /home/pi/.config//home/pi/hamlib_settings does not exist
rig_open: async_data_enable=0, async_data_supported=0
rig_open: using network address 127.0.0.1:4532
network_open: hoststr=127.0.0.1, portstr=4532
connect to 127.0.0.1:4532 failed, (trying next interface): Network error 111: Connection refused
network_open: failed to connect to 127.0.0.1:4532
rig_open: rs->comm_state==0?=0
rig.c(1095):rig_open returning2(-6) IO error

IO error
 (null) Connection refused 
```
Note that the messages all refer to the default port, not the one I specified.

Using an alternate syntax, I get errors:

`rigctld -m 1 -r 127.0.0.1:7356 -vvv`
```
connect to 127.0.0.1:7356 failed, (trying next interface): Network error 111: Connection refused
network_open: failed to connect to 127.0.0.1:7356
Opened rig model 1, 'Dummy'
```
[Here, I connected CubicSDR. When it failed and timed out, these messages appeared.]

```
rig_open: error =   3:rig.c(2779):rig_passband_normal entered
   3:rig_passband_normal: return filter#15, width=15000
  3:rig.c(2800):rig_passband_normal returning(15000) 
 2:dummy.c(307):dummy_init returning(0) 
set_conf: called
main: twiddle=0, uplink=0, twiddle_rit=0
rig.c(888):rig_open entered
rig_settings_get_path: path=/home/pi/.config/hamlib_settings
rig_settings_load_all: settings_file (/home/pi/.config/hamlib_settings): No such file or directory
rig_open: cwd=/home/pi
rig_open: /home/pi/.config//home/pi/hamlib_settings does not exist
rig_open: async_data_enable=0, async_data_supported=0
rig_open: using network address 127.0.0.1:7356
network_open: hoststr=127.0.0.1, portstr=7356
connect to 127.0.0.1:7356 failed, (trying next interface): Network error 111: Connection refused
network_open: failed to connect to 127.0.0.1:7356
rig_open: rs->comm_state==0?=0
rig.c(1095):rig_open returning2(-6) IO error

IO error
 127.0.0.1:7356 Connection refused 
connect to 127.0.0.1:7356 failed, (trying next interface): Network error 111: Connection refused
network_open: failed to connect to 127.0.0.1:7356
handle_socket: rig_open reopened retcode=-6
handle_socket: i/o error
handle_socket: rig_close retcode=-1
connect to 127.0.0.1:7356 failed, (trying next interface): Network error 111: Connection refused
network_open: failed to connect to 127.0.0.1:7356
handle_socket: rig_open retcode=-6, opened=0
handle_socket: rig_close retcode=-1
connect to 127.0.0.1:7356 failed, (trying next interface): Network error 111: Connection refused
network_open: failed to connect to 127.0.0.1:7356
handle_socket: rig_open retcode=-6, opened=0
handle_socket: rig_close retcode=-1
connect to 127.0.0.1:7356 failed, (trying next interface): Network error 111: Connection refused
network_open: failed to connect to 127.0.0.1:7356
handle_socket: rig_open retcode=-6, opened=0
handle_socket: rig_close retcode=-1
connect to 127.0.0.1:7356 failed, (trying next interface): Network error 111: Connection refused
network_open: failed to connect to 127.0.0.1:7356
handle_socket: rig_open retcode=-6, opened=0
```
Note that `rigctld -m 1 -r 127.0.0.1:4532 -vvv` worked (when I changed everything to use port 4532)

srs4511351, 2023-09-23:
gqrx does not need rigctld and does work with port 7356 on version Hamlib 4.6~git 2023-09-20T22:35:29Z SHA=6644af 64-bit

mdblack98, 2023-09-23:
So we can close this?
Mike W9MDB

srs4511351, 2023-09-23:
No, not until rigctld works with a user specified network port.
CubicSDR uses rigctld and this release breaks it.
I mentioned gqrx to point out that hamlib can use the user specified port, just not rigctld.

----Steve

mdblack98, 2023-09-24:
Let's confirm some behavior here.

#1 Versions
rigctld --version
rigctl --verion

#2 
rigctld -vvvvv -m 1 -T 127.0.0.1 -t 7356
You should see this as the last line
main: rigctld listening on port 7356
The run

rigctl -m 2 -vvvvv -r 127.0.0.1:7356

This should connect and allow querying freq, mode, etc.

#3 Please explain what  you are trying to do with CubicSDR and rigctld -- I don't see the relationship here.

srs4511351, 2023-09-25:
```
pi@raspberrypi:~ $ rigctld --version
rigctld Hamlib 4.6~git 2023-09-20T22:35:29Z SHA=6644af 64-bit
pi@raspberrypi:~ $ rigctl --version
rigctl Hamlib 4.6~git 2023-09-20T22:35:29Z SHA=6644af 64-bit
pi@raspberrypi:~ $ 
```
```
rigctl -m 2 -vvvvv -r 127.0.0.1:7356
Last line is
main: rigctld listening on port 7356
```
With CubicSDR control port set to 127.0.0.1:7356, it does not connect. It says `I/O error (Open failed?)`
rigctld later says `I/O error`
```
rigctl -m 2 -vvvvv -r 127.0.0.1:7356
The last few lines are
connect to 127.0.0.1:7356 failed, (trying next interface): Network error 111: Connection refused
network_open: failed to connect to 127.0.0.1:7356
rig_open: rs->comm_state==0?=0
rig.c(1095):rig_open returning2(-6) IO error
```
CubicSDR did not have hamlib enabled at that time. No other applications had hamlib enabled.
CubicSDR will not enable rig unless the connection is successful.

As for explaining why I use rigctld with Cubic SDR, It is a workaround for CubicSDR.
It will not connect directly to the application that is to control it, as that application will not enable rig unless there is something ready to connect to.
I have found that using the dummy device and rigctld will satisfy CubicSDR and it does talk to the application, which in this case is fldigi, but wsjt-x works the same.
I suspect that CubicSDR does not correctly implement the hamlib interface.

In any case, I think we need to find out why this setup only works with port 4532, where hamlib version 4.5.5 worked with my specified port.

----Steve

srs4511351, 2023-09-25:
I noticed something interesting when I tried this again later and I looked at a few more lines.
```
pi@raspberrypi:~ $ rigctld -vvvvv -m 1 -T 127.0.0.1 -t 7356

network.c(981):multicast address=0.0.0.0, port=4532
network.c(987): not starting multicast publisher
main: rigctld listening on port 7356
```
Although multicast was not started, it did specify port=**4532**

----Steve

mdblack98, 2023-09-25:
That port 4532 reference is not actually touching anything so should be irrelevant.

Ok...let's do this now....

rigctld -vvvvv -m 1 -T 127.0.0.1 -t 7356 >log.txt 2>&1

Then try to connect CubicSDR -- it fails -- ctrl-c the rigctld process and send me the log.txt file.

Mike W9MDB

mdblack98, 2023-09-25:
And where does one set the rigctld port in CubicSDR?  I don't see the ability to do that.

I also see another likely problem -- CubicSDR is using Hamlib V2.0 which is two versions behind and quite likely won't work with libhamlib-4

The debug log might alllow me to make it backwards compatible but no guarantees.

Mike W9MDB

srs4511351, 2023-09-25:
rigctld -vvvvv -m 1 -T 127.0.0.1 -t 7356 >log.txt 2>&1

Still fails.
[log.txt](https://github.com/Hamlib/Hamlib/files/12717901/log.txt)

I have CubicSDR v0.2.8 built on 6/8/2023
From the Rig Control menu, Control Port, enter 127.0.0.1:7356.

mdblack98, 2023-09-25:
This is fixed in a more current version of CubicSDR.  Not fixable in Hamlib.

closed 2023-09-25 by mdblack98

srs4511351, 2023-09-25:
With the latest CubicSDR installed today, it is working again.
I don't see anything in CubicSDR's list of commits since my last build that sounds like it would fix this, but it does seem to be fixed.
It is likely that this is due to compiling and installing CubicSDR after upgrading hamlib (both done today).

Thanks.

----Steve