Hamlib/Hamlib #1397: Question: Setting frequency of another VFO when split is not enabled

mikaelnousiainen, 2023-10-08:
Mainly a question to @mdblack98:

The way \set_split_freq works now is that it sets the frequency of the TX VFO. In case split is enabled (for the opposite for, e.g. for VFOB when RX is at VFOA), this works correctly and as expected.

However, when split is NOT enabled, the TX VFO is the same as RX VFO, so \set_split_freq changes the RX VFO freq -- effect is the same as with \set_freq.

My question: Is there any mechanism currently to set the frequency/mode of another VFO when split is NOT enabled? Many rigs support this, as you know.

mdblack98, 2023-10-08:
F VFOA 14074000F VFOB 14075000
That's with --vfo switch on or via the rig_set_freq call.
Mike W9MDB

mikaelnousiainen, 2023-10-08:
@mdblack98 Thanks. Right, forgot the VFO mode. But then every rigctld command _requires_ the VFO, right? So it requires changes to the client.

mdblack98, 2023-10-08:
Without the --vfo switch you have to swap VFOs yourself.  The --vfo switch allows Hamlib to use the FA/FB, MD, or 0x25/0x26 commands to do VFOB stuff much smoother.
WIthout --vfo
V VFOA
F 14074000
V VFOB
F 14075000
V VFOA

Not every command.  Just the ones that might be able to use a VFO

    { 'F',  "set_freq",         ACTION(set_freq),       ARG_IN1 | ARG_OUT1, "Frequency" },
    { 'f',  "get_freq",         ACTION(get_freq),       ARG_OUT, "Frequency", "VFO" },
    { 'F',  "set_freq",         ACTION(set_freq),       ARG_IN1, "Frequency" },
    { 'f',  "get_freq",         ACTION(get_freq),       ARG_OUT, "Frequency" },
    { 'M',  "set_mode",         ACTION(set_mode),       ARG_IN, "Mode", "Passband" },
    { 'm',  "get_mode",         ACTION(get_mode),       ARG_OUT, "Mode", "Passband" },
    { 'I',  "set_split_freq",   ACTION(set_split_freq), ARG_IN, "TX Frequency" },
    { 'i',  "get_split_freq",   ACTION(get_split_freq), ARG_OUT, "TX Frequency" },
    { 'X',  "set_split_mode",   ACTION(set_split_mode), ARG_IN, "TX Mode", "TX Passband" },
    { 'x',  "get_split_mode",   ACTION(get_split_mode), ARG_OUT, "TX Mode", "TX Passband" },
    { 'K',  "set_split_freq_mode",  ACTION(set_split_freq_mode), ARG_IN,    "TX Frequency", "TX Mode", "TX Passband" },
    { 'k',  "get_split_freq_mode",  ACTION(get_split_freq_mode), ARG_OUT,   "TX Frequency", "TX Mode", "TX Passband" },
    { 'S',  "set_split_vfo",    ACTION(set_split_vfo),  ARG_IN, "Split", "TX VFO" },
    { 's',  "get_split_vfo",    ACTION(get_split_vfo),  ARG_OUT, "Split", "TX VFO" },
    { 'N',  "set_ts",           ACTION(set_ts),         ARG_IN, "Tuning Step" },
    { 'n',  "get_ts",           ACTION(get_ts),         ARG_OUT, "Tuning Step" },
    { 'L',  "set_level",        ACTION(set_level),      ARG_IN, "Level", "Level Value" },
    { 'l',  "get_level",        ACTION(get_level),      ARG_IN1 | ARG_OUT2, "Level", "Level Value" },
    { 'U',  "set_func",         ACTION(set_func),       ARG_IN, "Func", "Func Status" },
    { 'u',  "get_func",         ACTION(get_func),       ARG_IN1 | ARG_OUT2, "Func", "Func Status" },
    { 'G',  "vfo_op",           ACTION(vfo_op),         ARG_IN, "Mem/VFO Op" },
    { 'g',  "scan",             ACTION(scan),           ARG_IN, "Scan Fct", "Scan Channel" },
    { 'R',  "set_rptr_shift",   ACTION(set_rptr_shift), ARG_IN, "Rptr Shift" },
    { 'r',  "get_rptr_shift",   ACTION(get_rptr_shift), ARG_OUT, "Rptr Shift" },
    { 'O',  "set_rptr_offs",    ACTION(set_rptr_offs),  ARG_IN, "Rptr Offset" },
    { 'o',  "get_rptr_offs",    ACTION(get_rptr_offs),  ARG_OUT, "Rptr Offset" },
    { 'C',  "set_ctcss_tone",   ACTION(set_ctcss_tone), ARG_IN, "CTCSS Tone" },
    { 'c',  "get_ctcss_tone",   ACTION(get_ctcss_tone), ARG_OUT, "CTCSS Tone" },
    { 'D',  "set_dcs_code",     ACTION(set_dcs_code),   ARG_IN, "DCS Code" },
    { 'd',  "get_dcs_code",     ACTION(get_dcs_code),   ARG_OUT, "DCS Code" },
    { 0x90, "set_ctcss_sql",    ACTION(set_ctcss_sql),  ARG_IN, "CTCSS Sql" },
    { 0x91, "get_ctcss_sql",    ACTION(get_ctcss_sql),  ARG_OUT, "CTCSS Sql" },
    { 0x92, "set_dcs_sql",      ACTION(set_dcs_sql),    ARG_IN, "DCS Sql" },
    { 0x93, "get_dcs_sql",      ACTION(get_dcs_sql),    ARG_OUT, "DCS Sql" },
    { 'T',  "set_ptt",          ACTION(set_ptt),        ARG_IN, "PTT" },
    { 't',  "get_ptt",          ACTION(get_ptt),        ARG_OUT, "PTT" },
    { 'E',  "set_mem",          ACTION(set_mem),        ARG_IN, "Memory#" },
    { 'e',  "get_mem",          ACTION(get_mem),        ARG_OUT, "Memory#" },
    { 'B',  "set_bank",         ACTION(set_bank),       ARG_IN, "Bank" },
    { 'J',  "set_rit",          ACTION(set_rit),        ARG_IN, "RIT" },
    { 'j',  "get_rit",          ACTION(get_rit),        ARG_OUT, "RIT" },
    { 'Z',  "set_xit",          ACTION(set_xit),        ARG_IN, "XIT" },
    { 'z',  "get_xit",          ACTION(get_xit),        ARG_OUT, "XIT" },

mikaelnousiainen, 2023-10-08:
Yes. I'm indeed after the "smooth" behavior without VFO swapping (Yaesu/Kenwood have the FA/FB commands too). Guess I'll just have to bite the bullet and go for the rigctld VFO mode. Thanks!

closed 2023-10-08 by mikaelnousiainen

mdblack98, 2023-10-08:
Also you can make it flexible though it is by-command.
chk_vfo returns 1 if --vfo is on or 0You can also turn on vfo mode on/off from a client with set_vfo_mode 1
Yyou can set a variable for the VFO  like this:
myvfo = "";if (chk_vfo)myvfo = " VFOA"; // notice the space at the frontsprinft(cmd, "F%s %d\n", myvfo, freq);

mikaelnousiainen, 2023-10-08:
Great, the `\set_vfo_opt` might save me here. Never paid attention to the VFO mode much, so didn't think any of this was possible :D