Hamlib/Hamlib #1399: Rigctl set_ant and get_ant may require VFO

mikaelnousiainen, 2023-10-10:
Both `set_ant` and `get_ant` may need VFO, because some more advanced rigs have targetable antenna selection.

Please see my comment in this commit: https://github.com/Hamlib/Hamlib/commit/64d6cd4af270bd2e94968eb6df5c34202b0763a4

mdblack98, 2023-10-10:
Which rigs would that be?
I would need to fix them too.

Mike W9MDB

Please see my comment in this commit: 64d6cd4

mdblack98, 2023-10-10:
Or better said...do you know of any other rigs that need this?

mikaelnousiainen, 2023-10-10:
At least FTDX5000 and FTDX9000 seem to allow this in addition to the newer FTDX101. Didn't check others yet, but just wanted to fix the issue in the commands to allow targetable antenna selection. Thanks!

closed 2023-10-10 by mikaelnousiainen